espressif/esp-dev-kits #3: Example code not running

pevsonic, 2020-06-23:
Hi,

I've just received a Kaluga board from DigiKey - it came pre-flashed with a simple LCD demo which appeared to show some images. However I've had problems trying to run the example code from this repository.

Environment :
- host : MacBook Pro running OSX 10.15.5
- esp-idf : release/v4.2
- esp-dev-kits : 34b90c43311b02781d134b6ea9f5bc11d810aeda
- Kaluga H/W version v1.2
- Both camera and cap touch board attached

I've built and flashed the example code as follows : 

`
. $HOME/esp/esp-idf/export.sh

cd ~/esp
git clone --recursive https://github.com/espressif/esp-dev-kits/
export ESPPORT=/dev/tty.usbserial-14201

# Try LCD example
cd ~/esp/esp-dev-kits/esp32-s2-kaluga-1/examples/lcd/
idf.py fullclean
idf.py set-target esp32s2
idf.py build
idf.py flash
idf.py monitor

# Try Camera example
cd ~/esp/esp-dev-kits/esp32-s2-kaluga-1/examples/camera/
idf.py fullclean
idf.py set-target esp32s2
idf.py build
idf.py flash
idf.py monitor

# Try Touch example
cd ~/esp/esp-dev-kits/esp32-s2-kaluga-1/examples/touch/
idf.py fullclean
idf.py set-target esp32s2
idf.py build
idf.py flash
idf.py monitor
`

All three examples build and flash, but none of them appear to work. LCD & camera turn on the LCD to a blank white screen and Touch just has a black screen.

Ive had a crack at activating the pushbuttons and capacitive buttons on each whilst the monitor is open but I don't see any debug output either.

I'd have expected these to be working and the LCD is working in hardware as the pre flashed demo worked - any ideas?

pevsonic, 2020-06-23:
I translated the README.md notes to English and have been experimenting a bit and found the issue with the LCD. 

When you run `idf.py menuconfig` and look at ``Example Configuration > lcd module type``,  the option for Autodetect does not work. Additionally, the obvious choice ` ST7789V (ESP32-S2-KALUGA_V1.1)` also does not work. However, if I set the option for ``ILI9341`` I get LCD output now. 

For reference, my LCD daughter card is labelled ESP-LyraP-LCD32 V1.1 which I understand to be the standard supplied board?

hanabic, 2020-06-24:
Have you check the switch state. The switch states of lcd example and touch example are different. Using lcd the switchs should be on and using touch should be off .

hanabic, 2020-06-24:
> I translated the README.md notes to English and have been experimenting a bit and found the issue with the LCD.
> 
> When you run `idf.py menuconfig` and look at `Example Configuration > lcd module type`, the option for Autodetect does not work. Additionally, the obvious choice ` ST7789V (ESP32-S2-KALUGA_V1.1)` also does not work. However, if I set the option for `ILI9341` I get LCD output now.
> 
> For reference, my LCD daughter card is labelled ESP-LyraP-LCD32 V1.1 which I understand to be the standard supplied board?
> 
> I also imagine that means that I need to make an equivalent change in the camera sample app for the LCD but it doesn't appear that there is the available option via menuconfig to do this there however?

Have you check the switch state? The switch states of lcd example and touch example are different. Using lcd the switchs should be on and using touch should be off .

hanabic, 2020-06-24:
> I translated the README.md notes to English and have been experimenting a bit and found the issue with the LCD.
>
> When you run `idf.py menuconfig` and look at `Example Configuration > lcd module [...]

> I translated the README.md notes to English and have been experimenting a bit and found the issue with the LCD.
>
> When you run `idf.py menuconfig` and look at `Example Configuration > lcd module [...]

The are two versions for the screen of  ESP-LyraP-LCD32,  ILI9341 or ST7789V. and the autodetect now have some problem on the new dev-kits.

AI-Thinker-wm, 2020-07-20:
> 你好
> 
> 我刚刚从DigiKey收到了Kaluga板-它预闪了一个简单的LCD演示，看起来可以显示一些图像。但是，在尝试从此存储库运行示例代码时遇到了问题。
> 
> 环境 ：
> 
> * 主机：运行OSX 10.15.5的MacBook Pro
> * esp-idf：版本/v4.2
> * esp-dev-kits：[34b90c4](https://github.com/espressif/esp-dev-kits/commit/34b90c43311b02781d134b6ea9f5bc11d810aeda)
> * Kaluga H / W版本v1.2
> * 相机和帽子触摸板均已连接
> 
> 我已经构建并刷新了示例代码，如下所示：
> 
> ```
> . $HOME/esp/esp-idf/export.sh
> 
> cd ~/esp
> git clone --recursive https://github.com/espressif/esp-dev-kits/
> export ESPPORT=/dev/tty.usbserial-14201
> 
> # Try LCD example
> cd ~/esp/esp-dev-kits/esp32-s2-kaluga-1/examples/lcd/
> idf.py fullclean
> idf.py set-target esp32s2
> idf.py build
> idf.py flash
> idf.py monitor
> 
> # Try Camera example
> cd ~/esp/esp-dev-kits/esp32-s2-kaluga-1/examples/camera/
> idf.py fullclean
> idf.py set-target esp32s2
> idf.py build
> idf.py flash
> idf.py monitor
> 
> # Try Touch example
> cd ~/esp/esp-dev-kits/esp32-s2-kaluga-1/examples/touch/
> idf.py fullclean
> idf.py set-target esp32s2
> idf.py build
> idf.py flash
> idf.py monitor
> ```
> 
> 这三个示例均已构建并刷新，但似乎都没有起作用。LCD和相机将LCD打开到空白的白屏，而Touch仅具有黑屏。
> 
> 在监视器处于打开状态时，Ive在激活每个按钮和电容式按钮方面存在缝隙，但我也看不到任何调试输出。
> 
> 我希望这些能够正常工作，并且液晶显示屏可以在预闪演示正常工作的情况下在硬件中正常工作-有什么想法吗？

AI-Thinker-wm, 2020-07-20:
> 你好
>
> 我刚刚从DigiKey收到了Kaluga板-它预闪了一个简单的LCD演示，看起来可以显示一些图像。但是，在尝试从此存储库运行示例代码时遇到了问题。
>
> 环境 ：
>
> * 主机：运行OSX 10.15.5的MacBook Pro
> * esp-idf：版本/v4.2
> * [...]

Hi，

AI-Thinker-wm, 2020-07-20:
Hi，Maybe you should modify the menuconfig according to LCD. My kit is: esp32-s2-kaluga-1 V1.2 + LCD32 V1.1. You can use KALUGA_v1.2 and ILI9341 to see the result, because I also encountered the same problem

danflys, 2020-12-21:
Ditto that,  brand new KALUGA-1_V1.3 unit had to change LCD Version to ILI9441 in 'idf.py menuconfig' utility to get the lcd example code to function.

lambrou, 2021-03-06:
I am having the same issue and have tried selecting both ILI9441 and ST7789V and I still just get a blank screen. The backlight comes on, but thats it. Can I confirm we are using the examples in \esp\examples\peripherals\spi_master\lcd

lucastucker1404, 2022-03-14:
Had the same problem too.

LCD used: ESP-LyraP-LCD32 V1.1

On the Espressif website it indicates that the driver for this model can either be an ST7789V or an ILI9341 (see here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-esp-lyrap-lcd32-v1.2.html). 

As said above, the example code provides an autodetect feature to automatically detect which driver the LCD is using. After flashing the project, I monitored the program's output and it indicated my LCD driver was ST7789V, which was incorrect. Therefore the autodetect feature was not working. I had to manually change the code and force it to recognize my LCD drivr as an ILI9341. After flashing again, the project ran perfectly :)

Note: I was also using a Kaluga V1.2, so don't foget to check the GPIO pin numbers if you also use the SPI Master lcd example. This example is destined for another kit so you'll also have to make this changes if using the same board as me.

Lzw655, 2023-05-12:
Sorry for not responding to your question in a timely manner!
Since this problem has lost its time, I will close it after a week. If you still need to solve this problem, please leave a comment or open a new issue, thank you!