espressif/esp-dev-kits #5: 编译了Camera工程烧录并运行，什么也看不到？

coollofty, 2020-06-30:
购买了S2-Kaluga-1开发板，到手后，设置Example Configuration中的：
Camera pad type & LCD pad type为v1.2
Camera version为OV2640
LCD version为ILI9341
编译，烧录该工程，启动，屏幕是亮了，但是没有像素，处于“黑”的状态
摄像头的灯也是亮的，所有日志：

`I (46) boot: ESP-IDF v4.2-dev-1856-g00148cd0c-dirty 2nd stage bootloader
I (46) boot: compile time 08:58:51
I (46) boot: chip revision: 0
I (50) qio_mode: Enabling default flash chip QIO
I (55) boot.esp32s2: SPI Speed      : 80MHz
I (60) boot.esp32s2: SPI Mode       : QIO
I (65) boot.esp32s2: SPI Flash Size : 4MB
I (70) boot: Enabling RNG early entropy source...
I (75) boot: Partition Table:
I (79) boot: ## Label            Usage          Type ST Offset   Length
I (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (101) boot:  2 factory          factory app      00 00 00010000 00100000
I (108) boot: End of partition table
I (113) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x06c40 ( 27712) map
I (127) esp_image: segment 1: paddr=0x00016c68 vaddr=0x3ffbe580 size=0x01e98 (  7832) load
I (132) esp_image: segment 2: paddr=0x00018b08 vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at E:/esp32/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (140) esp_image: segment 3: paddr=0x00018f14 vaddr=0x40024404 size=0x07104 ( 28932) load
I (155) esp_image: segment 4: paddr=0x00020020 vaddr=0x40080020 size=0x186ac (100012) map
0x40080020: _stext at ??:?

I (175) esp_image: segment 5: paddr=0x000386d4 vaddr=0x4002b508 size=0x03070 ( 12400) load
0x4002b508: get_block at E:/esp32/esp-idf/components/heap/multi_heap.c:120
 (inlined by) multi_heap_free_impl at E:/esp32/esp-idf/components/heap/multi_heap.c:531

I (184) boot: Loaded app from partition at offset 0x10000
I (184) boot: Disabling RNG early entropy source...
I (185) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (192) cpu_start: Pro cpu up.
I (196) cpu_start: Application information:
I (201) cpu_start: Project name:     esp32-s2-hmi
I (206) cpu_start: App version:      cb27d8f-dirty
I (212) cpu_start: Compile time:     Jun 30 2020 09:44:16
I (218) cpu_start: ELF file SHA256:  6ea20425819ca144...
I (224) cpu_start: ESP-IDF:          v4.2-dev-1856-g00148cd0c-dirty
I (231) cpu_start: Single core mode
I (235) heap_init: Initializing. RAM available for dynamic allocation:
I (242) heap_init: At 3FFC0C08 len 0003B3F8 (236 KiB): DRAM
I (248) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (254) cpu_start: Pro cpu start user code
I (314) spi_flash: detected chip: gd
I (314) spi_flash: flash io: qio
I (314) cpu_start: Starting scheduler on PRO CPU.
I (317) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (317) gpio: GPIO[11]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (327) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (337) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (347) lcd: lcd_buffer_size: 2048, lcd_dma_size: 1024, lcd_dma_node_cnt: 2

I (657) lcd: ILI19341 init...

I (857) lcd: lcd init ok

I (857) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (857) cam: cam_xclk_pin setup

I (857) cam: cam_buffer_size: 2048, cam_dma_size: 1024, cam_dma_node_cnt: 2, cam_total_cnt: 150

I (867) sccb: pin_sda 8 pin_scl 7

I (1057) main: sensor_id: 0x13

I (1057) main: OV2640 init start...
I (1107) OV2640: HID:9794

I (1147) main: camera init done`

我看了一下camera下的main.c，代码的意思应该是将摄像头的图像画到LCD上？但实际上什么也没有，这是为什么呢？

ChenZhengwei20190741, 2020-06-30:
拨码开关开了吗？

> 购买了S2-Kaluga-1开发板，到手后，设置Example Configuration中的：
> Camera pad type & LCD pad type为v1.2
> Camera version为OV2640
> LCD version为ILI9341
> 编译，烧录该工程，启动，屏幕是亮了，但是没有像素，处于“黑”的状态
> 摄像头的灯也是亮的，所有日志：
> 
> ```
> I (46) boot: ESP-IDF v4.2-dev-1856-g00148cd0c-dirty 2nd stage bootloader
> I (46) boot: compile time 08:58:51
> I (46) boot: chip revision: 0
> I (50) qio_mode: Enabling default flash chip QIO
> I (55) boot.esp32s2: SPI Speed      : 80MHz
> I (60) boot.esp32s2: SPI Mode       : QIO
> I (65) boot.esp32s2: SPI Flash Size : 4MB
> I (70) boot: Enabling RNG early entropy source...
> I (75) boot: Partition Table:
> I (79) boot: ## Label            Usage          Type ST Offset   Length
> I (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (101) boot:  2 factory          factory app      00 00 00010000 00100000
> I (108) boot: End of partition table
> I (113) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x06c40 ( 27712) map
> I (127) esp_image: segment 1: paddr=0x00016c68 vaddr=0x3ffbe580 size=0x01e98 (  7832) load
> I (132) esp_image: segment 2: paddr=0x00018b08 vaddr=0x40024000 size=0x00404 (  1028) load
> 0x40024000: _WindowOverflow4 at E:/esp32/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730
> 
> I (140) esp_image: segment 3: paddr=0x00018f14 vaddr=0x40024404 size=0x07104 ( 28932) load
> I (155) esp_image: segment 4: paddr=0x00020020 vaddr=0x40080020 size=0x186ac (100012) map
> 0x40080020: _stext at ??:?
> 
> I (175) esp_image: segment 5: paddr=0x000386d4 vaddr=0x4002b508 size=0x03070 ( 12400) load
> 0x4002b508: get_block at E:/esp32/esp-idf/components/heap/multi_heap.c:120
>  (inlined by) multi_heap_free_impl at E:/esp32/esp-idf/components/heap/multi_heap.c:531
> 
> I (184) boot: Loaded app from partition at offset 0x10000
> I (184) boot: Disabling RNG early entropy source...
> I (185) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
> I (192) cpu_start: Pro cpu up.
> I (196) cpu_start: Application information:
> I (201) cpu_start: Project name:     esp32-s2-hmi
> I (206) cpu_start: App version:      cb27d8f-dirty
> I (212) cpu_start: Compile time:     Jun 30 2020 09:44:16
> I (218) cpu_start: ELF file SHA256:  6ea20425819ca144...
> I (224) cpu_start: ESP-IDF:          v4.2-dev-1856-g00148cd0c-dirty
> I (231) cpu_start: Single core mode
> I (235) heap_init: Initializing. RAM available for dynamic allocation:
> I (242) heap_init: At 3FFC0C08 len 0003B3F8 (236 KiB): DRAM
> I (248) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
> I (254) cpu_start: Pro cpu start user code
> I (314) spi_flash: detected chip: gd
> I (314) spi_flash: flash io: qio
> I (314) cpu_start: Starting scheduler on PRO CPU.
> I (317) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
> I (317) gpio: GPIO[11]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
> I (327) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
> I (337) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
> I (347) lcd: lcd_buffer_size: 2048, lcd_dma_size: 1024, lcd_dma_node_cnt: 2
> 
> I (657) lcd: ILI19341 init...
> 
> I (857) lcd: lcd init ok
> 
> I (857) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
> I (857) cam: cam_xclk_pin setup
> 
> I (857) cam: cam_buffer_size: 2048, cam_dma_size: 1024, cam_dma_node_cnt: 2, cam_total_cnt: 150
> 
> I (867) sccb: pin_sda 8 pin_scl 7
> 
> I (1057) main: sensor_id: 0x13
> 
> I (1057) main: OV2640 init start...
> I (1107) OV2640: HID:9794
> 
> I (1147) main: camera init done
> ```
> 
> IDF 版本：v4.2
> dev-kits 版本：master最新
> 工具链：esptool.py v3.0-dev
> 
> 我看了一下camera下的main.c，代码的意思应该是将摄像头的图像画到LCD上？但实际上什么也没有，这是为什么呢？

是的， 将摄像头的图像画到LCD上。你看看你的拨码开关打开了没。

coollofty, 2020-06-30:
经常各种琢磨和深度，我用565模式（就是JPG不选的话），是已经可以显示了。但JPEG模式还是会报错，请看我另外一个Issue
另外，Flash很高概率失败的问题，我发现只需要拔掉摄像头，基本上就正常了

SunDUINO, 2020-07-08:
Hi 

How to compile examples for cameras in S2-Kaluga ?? I get the error:
`C:\esp-idf\examples\esp32s2\camera>idf.py build clean
Executing action: clean
Running ninja in directory c:\esp-idf\examples\esp32s2\camera\build
Executing "ninja clean"...
[1/1] Cleaning all built files...
Cleaning... 0 files.
Executing action: all (aliases: build)
Running ninja in directory c:\esp-idf\examples\esp32s2\camera\build
Executing "ninja all"...
ninja: error: 'C:/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a', needed by 'esp32-s2-hmi.elf', missing and no known rule to make it
ninja failed with exit code 1

C:\esp-idf\examples\esp32s2\camera>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\esp-idf\examples\esp32s2\camera\build
Executing "ninja all"...
ninja: error: 'C:/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a', needed by 'esp32-s2-hmi.elf', missing and no known rule to make it
ninja failed with exit code 1

C:\esp-idf\examples\esp32s2\camera>`

ChenZhengwei20190741, 2020-07-09:
> Hi
> 
> How to compile examples for cameras in S2-Kaluga ?? I get the error:
> 
> ```
> 
> 
> C:\esp-idf\examples\esp32s2\camera>idf.py build clean
> Executing action: clean
> Running ninja in directory c:\esp-idf\examples\esp32s2\camera\build
> Executing "ninja clean"...
> [1/1] Cleaning all built files...
> Cleaning... 0 files.
> Executing action: all (aliases: build)
> Running ninja in directory c:\esp-idf\examples\esp32s2\camera\build
> Executing "ninja all"...
> ninja: error: 'C:/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a', needed by 'esp32-s2-hmi.elf', missing and no known rule to make it
> ninja failed with exit code 1
> 
> C:\esp-idf\examples\esp32s2\camera>idf.py build
> Executing action: all (aliases: build)
> Running ninja in directory c:\esp-idf\examples\esp32s2\camera\build
> Executing "ninja all"...
> ninja: error: 'C:/esp-idf/components/esp_wifi/lib/esp32/libcoexist.a', needed by 'esp32-s2-hmi.elf', missing and no known rule to make it
> ninja failed with exit code 1
> 
> C:\esp-idf\examples\esp32s2\camera>
> ```
> 
> Im using IDF 4.0.1-dirty
> and my pathes:
> 
> ```
> Using Python in C:\Users\sunri\AppData\Local\Programs\Python\Python37\
> Python 3.7.3
> Using Git in C:\Program Files\Git\cmd\
> git version 2.21.0.windows.1
> Setting IDF_PATH: C:\esp-idf
> 
> Adding ESP-IDF tools to PATH...
>     C:\Users\sunri\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin
>     C:\Users\sunri\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin
>     C:\Users\sunri\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
>     C:\Users\sunri\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
>     C:\Users\sunri\.espressif\tools\cmake\3.16.4\bin
>     C:\Users\sunri\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200420\openocd-esp32\bin
>     C:\Users\sunri\.espressif\tools\ninja\1.10.0\
>     C:\Users\sunri\.espressif\tools\idf-exe\1.0.1\
>     C:\Users\sunri\.espressif\tools\ccache\3.7\
>     C:\Users\sunri\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
>     C:\Users\sunri\.espressif\python_env\idf4.0_py3.7_env\Scripts
>     C:\esp-idf\tools
> 
> Checking if Python packages are up to date...
> Python requirements from C:\esp-idf\requirements.txt are satisfied.
> 
> Done! You can now compile ESP-IDF projects.
> Go to the project directory and run:
> 
>   idf.py build
> 
> 
> C:\esp-idf>
> ```
you can checkout release/4.2， The toolchain for ESP32S2 was only available because of release /4.2

SunDUINO, 2020-07-09:
Hi
Thank you for the quick response, I have already updated ESP-iDF to 4.2, and checked the paths, and also exported the paths. I'm working on windows in Power Shell. I am currently getting the following errors: 

```
 C:/esp-idf/components/mbedtls/port/esp32s2/aes.c: In function 'esp_aes_process_dma':
 C:/esp-idf/components/mbedtls/port/esp32s2/aes.c:436:13: error: implicit declaration of function 'esp_ptr_external_ram'; did you mean 'esp_intr_set_in_iram'? [-Werror=implicit-function-declaration]
          if (esp_ptr_external_ram(input)) {
              ^~~~~~~~~~~~~~~~~~~~
              esp_intr_set_in_iram
C:/esp-idf/components/mbedtls/port/esp32s2/aes.c:447:14: error: implicit declaration of function 'esp_ptr_dma_ext_capable'; did you mean 'esp_aes_dma_wait_complete'? [-Werror=implicit-function-declaration]
         if (!esp_ptr_dma_ext_capable(input) && !esp_ptr_dma_capable(input)) {
               ^~~~~~~~~~~~~~~~~~~~~~~
               esp_aes_dma_wait_complete
 C:/esp-idf/components/mbedtls/port/esp32s2/aes.c:447:49: error: implicit declaration of function 'esp_ptr_dma_capable'; did you mean 'esp_intr_disable'? [-Werror=implicit-function-declaration]
          if (!esp_ptr_dma_ext_capable(input) && !esp_ptr_dma_capable(input)) {
                                                ^~~~~~~~~~~~~~~~~~~
                                                 esp_intr_disable
 cc1.exe: some warnings being treated as errors
 ninja: build stopped: subcommand failed.
 ninja failed with exit code 1
 
C:\esp-idf\examples\esp32s2\camera>
```

Thanks to the hints, I keep fighting. maybe it will succeed.

ChenZhengwei20190741, 2020-07-09:
> Hi
> Thank you for the quick response, I have already updated ESP-iDF to 4.2, and checked the paths, and also exported the paths. I'm working on windows in Power Shell. I am currently getting the following errors:
> 
> ```
> C:\esp-idf\examples\esp32s2\camera>idf.py set-target esp32s2
> Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
> Executing action: fullclean
> Build directory 'c:\esp-idf\examples\esp32s2\camera\build' not found. Nothing to clean.
> Executing action: set-target
> Set Target to: esp32s2, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
> Running cmake in directory c:\esp-idf\examples\esp32s2\camera\build
> Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DIDF_TARGET=esp32s2 -DIDF_TARGET=esp32s2 -DCCACHE_ENABLE=1 c:\esp-idf\examples\esp32s2\camera"...
> Warn about uninitialized values.
> -- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
> fatal: no submodule mapping found in .gitmodules for path 'components/esp_wifi/lib_esp32'
> -- ccache will be used for faster recompilation
> -- The C compiler identification is GNU 8.2.0
> -- The CXX compiler identification is GNU 8.2.0
> -- The ASM compiler identification is GNU
> -- Found assembler: C:/Users/sunri/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe
> -- Check for working C compiler: C:/Users/sunri/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe
> -- Check for working C compiler: C:/Users/sunri/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: C:/Users/sunri/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++.exe
> -- Check for working CXX compiler: C:/Users/sunri/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++.exe -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Building ESP-IDF components for target esp32s2
> Loading defaults file C:/esp-idf/examples/esp32s2/camera/sdkconfig.defaults...
> -- Found PythonInterp: C:/Users/sunri/.espressif/python_env/idf4.0_py3.7_env/Scripts/python.exe (found version "3.7.3")
> -- Could NOT find Perl (missing: PERL_EXECUTABLE)
> -- App "esp32-s2-hmi" version: v4.0.1-dirty
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.libgcc.ld
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-data.ld
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.spiflash.ld
> -- Adding linker script C:/esp-idf/examples/esp32s2/camera/build/esp-idf/esp32s2/esp32s2_out.ld
> -- Adding linker script C:/esp-idf/components/esp32s2/ld/esp32s2.project.ld.in
> -- Adding linker script C:/esp-idf/components/esp32s2/ld/esp32s2.peripherals.ld
> -- Components: app_trace app_update asio board bootloader bootloader_support bt cam cbor coap console cxx driver efuse es8311 esp-tls esp32s2 esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_tts esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap helix i2c_bus idf_test jpeg jsmn json lcd led_strip libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc sensors soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
> -- Component paths: C:/esp-idf/components/app_trace C:/esp-idf/components/app_update C:/esp-idf/components/asio C:/esp-idf/components/board C:/esp-idf/components/bootloader C:/esp-idf/components/bootloader_support C:/esp-idf/components/bt C:/esp-idf/components/cam C:/esp-idf/components/cbor C:/esp-idf/components/coap C:/esp-idf/components/console C:/esp-idf/components/cxx C:/esp-idf/components/driver C:/esp-idf/components/efuse C:/esp-idf/components/es8311 C:/esp-idf/components/esp-tls C:/esp-idf/components/esp32s2 C:/esp-idf/components/esp_common C:/esp-idf/components/esp_eth C:/esp-idf/components/esp_event C:/esp-idf/components/esp_gdbstub C:/esp-idf/components/esp_hid C:/esp-idf/components/esp_http_client C:/esp-idf/components/esp_http_server C:/esp-idf/components/esp_https_ota C:/esp-idf/components/esp_https_server C:/esp-idf/components/esp_ipc C:/esp-idf/components/esp_local_ctrl C:/esp-idf/components/esp_netif C:/esp-idf/components/esp_ringbuf C:/esp-idf/components/esp_rom C:/esp-idf/components/esp_serial_slave_link C:/esp-idf/components/esp_system C:/esp-idf/components/esp_timer C:/esp-idf/components/esp_tts C:/esp-idf/components/esp_websocket_client C:/esp-idf/components/esp_wifi C:/esp-idf/components/espcoredump C:/esp-idf/components/esptool_py C:/esp-idf/co [... cut ...]
>          if (!esp_ptr_dma_ext_capable(input) && !esp_ptr_dma_capable(input)) {
>                                                  ^~~~~~~~~~~~~~~~~~~
>                                                  esp_intr_disable
> cc1.exe: some warnings being treated as errors
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1
> 
> C:\esp-idf\examples\esp32s2\camera>
> ```
> 
> Thanks to the hints, I keep fighting. maybe it will succeed.

hi i think you should fullclean or `rm -rf build`

SunDUINO, 2020-07-09:
Thanks ChenZhengwei20190741

Unfortunately, neither fullclean nor deleting the build folder, re-uploading the entire example from git does not help, it cannot be compiled.

```
[675/969] Building C object esp-idf/es8311/CMakeFiles/__idf_es8311.dir/es8311.c.obj
FAILED: esp-idf/es8311/CMakeFiles/__idf_es8311.dir/es8311.c.obj
ccache C:\Users\sunri\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-gcc.exe  -Iconfig -I../../../../components/es8311/include -I../../../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/freertos/xtensa/include -I../../../../components/heap/include -I../../../../components/log/include -I../../../../components/lwip/include/apps -I../../../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/include -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/soc/src/esp32s2/. -I../../../../components/soc/src/esp32s2/include -I../../../../components/soc/include -I../../../../components/esp_rom/include -I../../../../components/esp_common/include -I../../../../components/esp_system/include -I../../../../components/xtensa/include -I../../../../components/xtensa/esp32s2/include -I../../../../components/esp32s2/include -I../../../../components/driver/include -I../../../../components/driver/esp32s2/include -I../../../../components/esp_ringbuf/include -I../../../../components/efuse/include -I../../../../components/efuse/esp32s2/include -I../../../../components/espcoredump/include -I../../../../components/esp_timer/include -I../../../../components/esp_ipc/include -I../../../../components/soc/soc/esp32s2/. -I../../../../components/soc/soc/esp32s2/include -I../../../../components/soc/soc/esp32s2/../include -I../../../../components/vfs/include -I../../../../components/esp_wifi/include -I../../../../components/esp_wifi/esp32s2/include -I../../../../components/esp_event/include -I../../../../components/esp_netif/include -I../../../../components/esp_eth/include -I../../../../components/tcpip_adapter/include -I../../../../components/app_trace/include -I../../../../components/i2c_bus/include -I../../../../components/board/include -mlongcalls   -fdiagnostics-color=always -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dev-1905-g625bd5eb1\" -DESP_PLATFORM -MD -MT esp-idf/es8311/CMakeFiles/__idf_es8311.dir/es8311.c.obj -MF esp-idf\es8311\CMakeFiles\__idf_es8311.dir\es8311.c.obj.d -o esp-idf/es8311/CMakeFiles/__idf_es8311.dir/es8311.c.obj   -c C:/esp-idf/components/es8311/es8311.c
In file included from c:\esp-idf\components\soc\soc\esp32s2\include\soc\io_mux_reg.h:17,
                 from c:\esp-idf\components\soc\soc\include\soc\gpio_periph.h:17,
                 from c:\esp-idf\components\driver\include\driver\gpio.h:22,
                 from C:/esp-idf/components/es8311/es8311.c:18:
C:/esp-idf/components/es8311/es8311.c: In function 'es8311_pa_power':
C:/esp-idf/components/es8311/es8311.c:247:32: error: 'GPIO_PA_EN' undeclared (first use in this function); did you mean 'GPIO_CLK_EN'?
     io_conf.pin_bit_mask = BIT(GPIO_PA_EN) | BIT(0);
                                ^~~~~~~~~~
c:\esp-idf\components\soc\soc\esp32s2\include\soc\soc.h:101:42: note: in definition of macro 'BIT'
 #define BIT(nr)                 (1UL << (nr))
                                          ^~
C:/esp-idf/components/es8311/es8311.c:247:32: note: each undeclared identifier is reported only once for each function it appears in
     io_conf.pin_bit_mask = BIT(GPIO_PA_EN) | BIT(0);
                                ^~~~~~~~~~
c:\esp-idf\components\soc\soc\esp32s2\include\soc\soc.h:101:42: note: in definition of macro 'BIT'
 #define BIT(nr)                 (1UL << (nr))
                                          ^~
C:/esp-idf/components/es8311/es8311.c: In function 'es8311_init':
C:/esp-idf/components/es8311/es8311.c:40:29: error: 'I2S_MCLK' undeclared (first use in this function); did you mean 'LCD_CLK'?
 #define INVERT_MCLK         I2S_MCLK /*!< do not invert */
                             ^~~~~~~~
C:/esp-idf/components/es8311/es8311.c:418:9: note: in expansion of macro 'INVERT_MCLK'
     if (INVERT_MCLK) {
         ^~~~~~~~~~~
C:/esp-idf/components/es8311/es8311.c:41:29: error: 'I2S_SCLK' undeclared (first use in this function); did you mean 'LCD_CLK'?
 #define INVERT_SCLK         I2S_SCLK
                             ^~~~~~~~
C:/esp-idf/components/es8311/es8311.c:431:9: note: in expansion of macro 'INVERT_SCLK'
     if (INVERT_SCLK) {
         ^~~~~~~~~~~
[676/969] Building C object esp-idf/protocomm/CMakeFiles/__idf_protocomm.dir/proto-c/sec1.pb-c.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\esp-idf\examples\esp32s2\camera>
```

always ends with an error. I think I'll let it go and wait maybe it will go sometime :(

ChenZhengwei20190741, 2020-07-09:
> Thanks ChenZhengwei20190741
>
> Unfortunately, neither fullclean nor deleting the build folder, re-uploading the entire example from git does not help, it cannot be compiled.
>
> ```
> [675/969] Building C [...]

maybe you should pull esp-dev-kit ?

SunDUINO, 2020-07-09:
I have downloaded esp-dev-kit and files thrown into the folder folderets and examples, but apart from that, ESP32-S2 fails to compile properly any exampla even Hello_world ends with errors. I have installed everything again a few times and will not work anymore.

Maybe it's Win10 fault ...

ChenZhengwei20190741, 2020-07-10:
Have you changed the code or updated the project to the latest `master`?

ChenZhengwei20190741, 2020-07-10:
> I have downloaded esp-dev-kit and files thrown into the folder folderets and examples, but apart from that, ESP32-S2 fails to compile properly any exampla even Hello_world ends with errors. I have installed everything again a few times and will not work anymore.
> 
> Maybe it's Win10 fault ...

Have you changed the code or updated the project to the latest MASTER

ChenZhengwei20190741, 2020-07-13:
> I have downloaded esp-dev-kit and files thrown into the folder folderets and examples, but apart from that, ESP32-S2 fails to compile properly any exampla even Hello_world ends with errors. I have installed everything again a few times and will not work anymore.
> 
> Maybe it's Win10 fault ...

I think you should `git status` to see what's different,or 
you should refer to The Windows environment setup.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html

closed 2023-05-12 by Lzw655