espressif/esp-dev-kits #6: ESP32-S2在Flash时经常失败，或成功后不能启动

coollofty, 2020-06-30:
S2-Kulaga-1开发板，Flash时经常失败，输出如下：

E:\esp32\esp-dev-kits\esp32-s2-kaluga-1\examples\camera>idf.py flash
Executing action: flash
Choosing default port b'COM13' (use '-p PORT' option to set a specific serial port)
Running ninja in directory e:\esp32\esp-dev-kits\esp32-s2-kaluga-1\examples\camera\build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D E:\esp32\esp-idf\components\esptoo... E:/esp32/esp-idf/components/esptool_py/run_esptool.cmake"
esptool.py --chip esp32s2 -p COM13 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 esp32-s2-hmi.bin
esptool.py v3.0-dev
Serial port COM13
Connecting....
Chip is ESP32-S2
Features: WiFi
Crystal is 40MHz
MAC: 7c:df:a1:00:79:5a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 1540.1 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
CMake Error at run_cmd.cmake:14 (message):
  esptool.py failed
Call Stack (most recent call first):
  run_esptool.cmake:21 (include)

FAILED: CMakeFiles/flash
cmd.exe /C "cd /D E:\esp32\esp-idf\components\esptool_py && C:\Users\Administrator\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="E:/esp32/esp-idf" -D ESPTOOLPY="C:\Users\Administrator\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe E:/esp32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32s2" -D ESPTOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="E:/esp32/esp-dev-kits/esp32-s2-kaluga-1/examples/camera/build" -P E:/esp32/esp-idf/components/esptool_py/run_esptool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

此时，如果在Flash之前手按Boot再按Reset，Flash成功的概率至少高1倍，否则，Flash 5次至少失败4次，按住Boot 5次可能只失败2~次。

另外，就是Flash完了之后，无法启动，monitor输出如下：

Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x1914
load:0x40050000,len:0x18a8
load:0x40054000,len:0x210c
Checksum failure. Calculated 0xd1 stored 0xd3
ets_main.c 386
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x7 (TG0WDT_SYS_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4000f79a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x1914
load:0x40050000,len:0x18a8
load:0x40054000,len:0x210c
Checksum failure. Calculated 0xbb stored 0xd3
ets_main.c 386

解决的方案就是再Flash 1次，就能正常运行。如果还是不行，再Flash 1次，肯定就可以正常运行。无须做任何修改，只需要不停的Flash，2~3次，就总有1次可以正常运行。

请问以上两个问题是何原因呢？

ChenZhengwei20190741, 2020-06-30:
> S2-Kulaga-1开发板，Flash时经常失败，输出如下：
> 
> ```
> E:\esp32\esp-dev-kits\esp32-s2-kaluga-1\examples\camera>idf.py flash
> Executing action: flash
> Choosing default port b'COM13' (use '-p PORT' option to set a specific serial port)
> Running ninja in directory e:\esp32\esp-dev-kits\esp32-s2-kaluga-1\examples\camera\build
> Executing "ninja flash"...
> [1/4] Performing build step for 'bootloader'
> ninja: no work to do.
> [1/2] cmd.exe /C "cd /D E:\esp32\esp-idf\components\esptoo... E:/esp32/esp-idf/components/esptool_py/run_esptool.cmake"
> esptool.py --chip esp32s2 -p COM13 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 esp32-s2-hmi.bin
> esptool.py v3.0-dev
> Serial port COM13
> Connecting....
> Chip is ESP32-S2
> Features: WiFi
> Crystal is 40MHz
> MAC: 7c:df:a1:00:79:5a
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 460800
> Changed.
> Configuring flash size...
> Compressed 3072 bytes to 103...
> Writing at 0x00008000... (100 %)
> Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 1540.1 kbit/s)...
> 
> A fatal error occurred: Timed out waiting for packet header
> CMake Error at run_cmd.cmake:14 (message):
>   esptool.py failed
> Call Stack (most recent call first):
>   run_esptool.cmake:21 (include)
> 
> 
> FAILED: CMakeFiles/flash
> cmd.exe /C "cd /D E:\esp32\esp-idf\components\esptool_py && C:\Users\Administrator\.espressif\tools\cmake\3.16.4\bin\cmake.exe -D IDF_PATH="E:/esp32/esp-idf" -D ESPTOOLPY="C:\Users\Administrator\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe E:/esp32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32s2" -D ESPTOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="E:/esp32/esp-dev-kits/esp32-s2-kaluga-1/examples/camera/build" -P E:/esp32/esp-idf/components/esptool_py/run_esptool.cmake"
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1
> ```
> 
> 此时，如果在Flash之前手按Boot再按Reset，Flash成功的概率至少高1倍，否则，Flash 5次至少失败4次，按住Boot 5次可能只失败2~次。
> 
> 另外，就是Flash完了之后，无法启动，monitor输出如下：
> 
> ```
> Build:Oct 25 2019
> rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3ffe8100,len:0x4
> load:0x3ffe8104,len:0x1914
> load:0x40050000,len:0x18a8
> load:0x40054000,len:0x210c
> Checksum failure. Calculated 0xd1 stored 0xd3
> ets_main.c 386
> ESP-ROM:esp32s2-rc4-20191025
> Build:Oct 25 2019
> rst:0x7 (TG0WDT_SYS_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
> Saved PC:0x4000f79a
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3ffe8100,len:0x4
> load:0x3ffe8104,len:0x1914
> load:0x40050000,len:0x18a8
> load:0x40054000,len:0x210c
> Checksum failure. Calculated 0xbb stored 0xd3
> ets_main.c 386
> ```
> 
> 解决的方案就是再Flash 1次，就能正常运行。如果还是不行，再Flash 1次，肯定就可以正常运行。无须做任何修改，只需要不停的Flash，2~3次，就总有1次可以正常运行。
> 
> IDF 版本：v4.2
> dev-kits 版本：master最新
> 工具链：esptool.py v3.0-dev
> 
> 请问以上两个问题是何原因呢？

请问，您手上的 kaluga 硬件是哪个版本的？

coollofty, 2020-06-30:
前天在淘宝上买的，底板的丝印显示是 v1.2 版

ChenZhengwei20190741, 2020-06-30:
你先尝试一下，不插入audio板进行烧录试试。

closed 2023-05-12 by Lzw655