espressif/esp-dev-kits #7: camera工程选择了JPEG模式之后，报错了

coollofty, 2020-06-30:
报错信息如下，持续的输出：

`E (6077) jpeg: Image decoder: jd_prepare failed (6)`

工程是：esp-dev-kits\esp32-s2-kaluga-1\examples\camera
IDF版本：v4.2
esp-dev-kits版本：master最新
开发板：esp32-s2-kaluga-1

如果不选择jpeg模式就能正常显示了，那么请问这个错误是什么原因？

closed 2020-07-20 by coollofty

ddddavidmartin, 2020-08-31:
Hi @coollofty,

how did you fix this issue? I'm currently having the same problem. The example doesn't work when jpg mode is enabled.

If I print the `cam_buf` contents in jpg mode they are always the same:
```
I (1581) main: fc d8 fc e0 00 10 48 44 48 44 00 00 00 00 00 00 
I (1581) main: 00 00 00 00 fc d8 00 40 00 0c 08 08 08 08 08 0c 
I (1591) main: 08 08 08 0c 0c 0c 0c 10 1c 14 10 10 10 10 24 18 
I (1601) main: 1c 14 1c 2c 24 2c 2c 28 24 28 28 30 34 44 38 30 
I (1601) main: 30 40 34 28 28 3c 50 3c 40 44 48 4c 4c 4c 2c 38 
I (1611) main: 54 58 54 48 58 44 4c 4c 48 fc d8 00 40 00 0c 0c
....
```

Thanks!

coollofty, 2020-09-01:
据官方说可能是摄像头拔码开关的问题，我这边由于自己画了板子，没有使用官方的开发板了确实也就没有了上面的问题

jack685, 2021-03-04:
how did you fix this problem

ddddavidmartin, 2021-03-04:
> how did you fix this problem

I didn't find a solution. I haven't worked with the hardware again since then and don't see myself doing so in the near future.

coollofty's comment from google translate:
> According to the official statement, it may be the problem of the camera's code switch. Since I drew the board by myself, I did not use the official development board, and the above problem would indeed be eliminated.

Looks like they worked with different hardware.

jack685, 2021-03-04:
this project report failed
I (569) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (579) gpio: GPIO[11]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (589) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (599) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (609) lcd: lcd_buffer_size: 2048, lcd_dma_size: 1024, lcd_dma_node_cnt: 2

I (909) lcd: ST7789 init...

I (909) lcd: lcd init ok

I (909) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (909) cam: cam_xclk_pin setup

I (909) cam: cam_buffer_size: 2048, cam_dma_size: 1024, cam_dma_node_cnt: 2, cam_total_cnt: 150

I (919) cam: frame1_buffer_en

I (929) cam: frame2_buffer_en

I (929) sccb: pin_sda 8 pin_scl 7

I (1179) main: sensor_id: 0x18

I (1179) main: OV2640 init start...
I (1229) OV2640: HID:9794

I (1269) main: camera init done

E (1409) jpeg: Image decoder: jd_decode failed (6)
E (1489) jpeg: Image decoder: jd_decode failed (6)
E (1599) jpeg: Image decoder: jd_decode failed (6)
E (1699) jpeg: Image decoder: jd_decode failed (6)