espressif/esp-dev-kits #12: 有一些摄像头采取出来的MJPEG数据出现问题，

coollofty, 2020-07-25:
[image]

ChenZhengwei20190741, 2020-07-27:
你是直接在上电过程中替换ov2640的吗?还是重启

coollofty, 2020-07-27:
> 1.你是直接在上电过程中替换ov2640的吗?还是掉电后替换ov2640再重新跑一下程序?
> 2.替换的 camera 管脚是否匹配.
> 3.有无尝试替换为原来的camera.是否输出和原来一样.

1、断电后替换，然后再上电来跑
2、都是OV2640，理论上应该匹配？这个我也无法得知，因为都是在淘宝上买的OV2640摄像头，并且销售方说是支持ESP32的
3、换回原来的摄像头，输出当然是正常的

ChenZhengwei20190741, 2020-07-27:
> 2、都是OV2640，理论上应该匹配？这个我也无法得知，因为都是在淘宝上买的OV2640摄像头，并且销售方说是支持ESP32的

不是的,esp32/esp32s2都是ov2640的, 只是不同厂家的管脚分配有可能不一样,建议跟他们要一下原理图看看管脚分配和kaluga上是否一样.

coollofty, 2020-07-27:
> > 2、都是OV2640，理论上应该匹配？这个我也无法得知，因为都是在淘宝上买的OV2640摄像头，并且销售方说是支持ESP32的
> 
> 不是的,esp32/esp32s2都是ov2640的, 只是不同厂家的管脚分配有可能不一样,建议跟他们要一下原理图看看管脚分配和kaluga上是否一样.

Thanks, 我再研究研究

coollofty, 2020-08-06:
这个摄像头表现正常，其pin图在这里：https://img.alicdn.com/imgextra/i2/795477101/O1CN01U6BcFr22KJ3grkCMM_!!795477101.png

而这个摄像头就是我主贴里说的情况，我对比了一下两个摄像头的pin，似乎没有任何区别：https://img.alicdn.com/imgextra/i3/461692664/O1CN011VY9ZJKebB0SR2U_!!461692664.png

ChenZhengwei20190741, 2020-08-06:
> 这个摄像头表现正常，其pin图在这里：https://img.alicdn.com/imgextra/i2/795477101/O1CN01U6BcFr22KJ3grkCMM_!!795477101.png
> 而这个摄像头就是我主贴里说的情况，我对比了一下两个摄像头的pin，似乎没有任何区别：https://img.alicdn.com/imgextra/i3/461692664/O1CN011VY9ZJKebB0SR2U_!!461692664.png

1、尝试 RGB565 输出到显示屏，看看有没有问题？排查是否RGB输出开始就数据错乱。
2、第一步没问题就说明可能是摄像头的 JPEG 压缩存在问题。
3、第一步如果有问题，用逻辑分析仪和示波器看看数据管脚，截个图给我。或者你自己看看有没有哪些数据线存在异常？

coollofty, 2020-08-06:
> 1、尝试 RGB565 输出到显示屏，看看有没有问题？排查是否RGB输出开始就数据错乱。
> 2、第一步没问题就说明可能是摄像头的 JPEG 压缩存在问题。
> 3、第一步如果有问题，用逻辑分析仪和示波器看看数据管脚，截个图给我。或者你自己看看有没有哪些数据线存在异常？

如您所料，用开发板的LCD看RGB565是正常的，但是编码出JPG数据之后出错了。这个要如何处理？

coollofty, 2020-08-07:
> 1、尝试 RGB565 输出到显示屏，看看有没有问题？排查是否RGB输出开始就数据错乱。
> 2、第一步没问题就说明可能是摄像头的 JPEG 压缩存在问题。
> 3、第一步如果有问题，用逻辑分析仪和示波器看看数据管脚，截个图给我。或者你自己看看有没有哪些数据线存在异常？

请问，这个是一个Bug吗？是否计划在下一个版本中进行更新？我现在做的产品需要用到这个功能，现在产品无法继续开发了……

ChenZhengwei20190741, 2020-08-28:
> 请问，这个是一个Bug吗？是否计划在下一个版本中进行更新？我现在做的产品需要用到这个功能，现在产品无法继续开发了……

hi，最近我最近测试发现有这个问题，大概率是你JTAG的拨码开关打开导致的。建议关掉试试。

closed 2020-09-18 by coollofty