espressif/esp-dev-kits #16: 停止Camera转到Audio会出现I2S错误

coollofty, 2020-09-18:
我使用如下代码停止了Camera：
cam_deinit();
i2s_driver_uninstall(I2S_NUM_0);
i2c_driver_delete(I2C_NUM_0);

然后再调用（代码来自web_tts工程的app_main.c）：
ESP_ERROR_CHECK(i2c_bus_init());
ESP_ERROR_CHECK(audio_init());

就会出错

```
I (26540) DRV8311: ES8311 in Slave mode

I (26810) DRV8311: ES8311 in I2S Format

I (27010) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (27010) DRV8311: SET: volume:255

E (27020) I2S: Register I2S Interrupt error
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400a261c  PS      : 0x00060330  A0      : 0x8008553b  A1      : 0x3ffd3be0
0x400a261c: i2s_set_pin at E:/esp32/esp-idf/components/driver/i2s.c:767

A2      : 0x00000000  A3      : 0x3ffd3c38  A4      : 0x3f00dac4  A5      : 0x00000000
A6      : 0x3ffcdb00  A7      : 0x3ffcdb00  A8      : 0x800a3024  A9      : 0x3ffd3b80
A10     : 0x00000003  A11     : 0x3f00c3ce  A12     : 0x3f00c932  A13     : 0x0000698c
A14     : 0x3f00c3ce  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000002c  LBEG    : 0x3f00c932  LEND    : 0x0000698c  LCOUNT  : 0x40026421
0x40026421: _xt_user_exc at E:/esp32/esp-idf/components/freertos/xtensa/xtensa_vectors.S:627
```

nuchi, 2021-04-23:
I got exactly the same error in `i2s_set_pin`, same LoadProhibited, same EXCVADDR. I didn't figure out why, but when I moved the task from core 0 to core 1, the error went away.

Lzw655, 2023-05-12:
Sorry for not responding to your question in a timely manner!
Since this problem has lost its time, I will close it after a week. If you still need to solve this problem, please leave a comment or open a new issue, thank you!

closed 2023-05-26 by Lzw655