espressif/esp-dev-kits #23: Wont flash

kayleejacob1, 2020-12-18:
When i connect the board with the 2 USB i get get two port one port wont do anything.  The other port connects but never will up load.  I tried turn the Jtag switch's on did nothing so switch them all off.

i am using the latest's IDF blink example

esptool.py v2.9-dev
Serial port COM27
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command C:\Users\Home\.espressif\python_env\idf4.1_py3.9_env\Scripts\python.exe c:\Users\Home\esp-idf\esp-idf\components\esptool_py\esptool\esptool.py -p COM27 -b 115200 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 blink.bin" terminated with exit code: 1.

Allyn69, 2021-01-20:
BOOTLOADER

kayleejacob1, 2021-02-19:
The configuration was wrong

closed 2021-02-19 by kayleejacob1