espressif/esp-dev-kits #24: Kaluga v1.3 Microphone Example?

andycarle, 2021-01-27:
Greetings!

I was wondering if any example code is available that uses the microphone on the Kaluga v1.3 ESP-LyraT-8311A v1.3 board. Or, alternately, if there is a full data sheet with register map and descriptions available for the ES8311.

I have successfully retrieved some audio samples from the microphone over I2S. But the samples are very quiet and my attempts to set the microphone's gain have been unsuccessful so far. I also have not had much luck adjusting the ADC sample rate.

Any guidance would be appreciated.

Thank you,
&nbsp;- Andy Carle

tore-espressif, 2021-02-18:
Hello Andy,

Thank you for your patience.

As you can see, ES8311's manufacturer Everest-Semi does not make their full-datasheets public, so I had to make sure with them whether we are allowed to share this information with you.

You can find latest datasheet and user-guide attached.
[ES8311 DS.pdf](https://github.com/espressif/esp-dev-kits/files/6002751/ES8311.DS.pdf)
[ES8311 user Guide.pdf](https://github.com/espressif/esp-dev-kits/files/6002754/ES8311.user.Guide.pdf)

I'll start working on a better ES8311 example tomorrow.

andycarle, 2021-02-23:
Hello @tore-espressif,

Thank you for the update! The documents from Everest-Semi should help us get going here.

I'll look forward to an ES8311 microphone example coming in the future!

closed 2023-05-12 by Lzw655

ESP-Mars, 2023-05-23:
- You can find the latest ES7210 datasheet here http://www.everest-semi.com/pdf/ES7210%20PB.pdf

- For the ES7120 user guide. We have been informed by Everest-Sami, the manufacturer of ES7210, that the user guide is currently being revised and will be made publicly available once the updates are complete. Subsequently, we will update the guide and provide it to you.