espressif/esp-dev-kits #27: ESP32-S2-kaluga-1/camera example report :E (1420) jpeg: Image decoder: jd_prepare failed (6)

jack685, 2021-03-15:
board: ESP32-s2-kaluga-1 ,  I enable jpeg mode ,but report err. what is the reason?
I (185) boot: Loaded app from partition at offset 0x10000
I (185) boot: Disabling RNG early entropy source...
I (187) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (194) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
I (201) spiram: Found 16MBit SPI RAM device
I (206) spiram: SPI RAM mode: sram 80m
I (211) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (218) cpu_start: Pro cpu up.
I (221) cpu_start: Application information:
I (226) cpu_start: Project name:     esp32-s2-hmi
I (232) cpu_start: App version:      cb426b3-dirty
I (237) cpu_start: Compile time:     Mar 15 2021 11:58:10
I (243) cpu_start: ELF file SHA256:  e3d0704fff0661e4...
I (249) cpu_start: ESP-IDF:          v4.2-332-ge7435a75f-dirty
I (256) cpu_start: Single core mode
I (478) spiram: SPI SRAM memory test OK
I (479) heap_init: Initializing. RAM available for dynamic allocation:
I (479) heap_init: At 3FFC0FB8 len 0003B048 (236 KiB): DRAM
I (485) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (491) cpu_start: Pro cpu start user code
I (496) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (504) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (566) spi_flash: detected chip: generic
I (566) spi_flash: flash io: qio
I (566) cpu_start: Starting scheduler on PRO CPU.
I (570) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (570) gpio: GPIO[11]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (580) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (590) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (600) lcd: lcd_buffer_size: 2048, lcd_dma_size: 1024, lcd_dma_node_cnt: 2

I (910) lcd: ST7789 init...

I (910) lcd: lcd init ok

I (910) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (910) cam: cam_xclk_pin setup

I (910) cam: cam_buffer_size: 2048, cam_dma_size: 1024, cam_dma_node_cnt: 2, cam_total_cnt: 150

I (920) cam: frame1_buffer_en

I (930) cam: frame2_buffer_en

I (930) sccb: pin_sda 8 pin_scl 7

I (1180) main: sensor_id: 0x18

I (1180) main: OV2640 init start...
I (1230) OV2640: HID:9794

I (1270) main: camera init done

E (1420) jpeg: Image decoder: jd_prepare failed (6)
E (1520) jpeg: Image decoder: jd_prepare failed (6)
E (1620) jpeg: Image decoder: jd_prepare failed (6)
E (1730) jpeg: Image decoder: jd_prepare failed (6)
E (1830) jpeg: Image decoder: jd_prepare failed (6)
E (1940) jpeg: Image decoder: jd_prepare failed (6)
E (2040) jpeg: Image decoder: jd_prepare failed (6)
E (2150) jpeg: Image decoder: jd_prepare failed (6)
E (2250) jpeg: Image decoder: jd_prepare failed (6)
E (2360) jpeg: Image decoder: jd_prepare failed (6)
E (2460) jpeg: Image decoder: jd_prepare failed (6)
E (2570) jpeg: Image decoder: jd_prepare failed (6)
E (2670) jpeg: Image decoder: jd_prepare failed (6)
E (2780) jpeg: Image decoder: jd_prepare failed (6)
E (2880) jpeg: Image decoder: jd_prepare failed (6)
E (2990) jpeg: Image decoder: jd_prepare failed (6)
E (3090) jpeg: Image decoder: jd_prepare failed (6)
E (3200) jpeg: Image decoder: jd_prepare failed (6)
E (3300) jpeg: Image decoder: jd_prepare failed (6)
E (3410) jpeg: Image decoder: jd_prepare failed (6)
E (3510) jpeg: Image decoder: jd_prepare failed (6)
E (3620) jpeg: Image decoder: jd_prepare failed (6)
E (3720) jpeg: Image decoder: jd_prepare failed (6)
E (3830) jpeg: Image decoder: jd_prepare failed (6)
E (3940) jpeg: Image decoder: jd_prepare failed (6)
E (4040) jpeg: Image decoder: jd_prepare failed (6)
E (4150) jpeg: Image decoder: jd_prepare failed (6)
E (4250) jpeg: Image decoder: jd_prepare failed (6)
E (4360) jpeg: Image decoder: jd_prepare failed (6)
E (4460) jpeg: Image decoder: jd_prepare failed (6)
E (4570) jpeg: Image decoder: jd_prepare failed (6)
E (4670) jpeg: Image decoder: jd_prepare failed (6)
E (4780) jpeg: Image decoder: jd_prepare failed (6)
E (4880) jpeg: Image decoder: jd_prepare failed (6)
E (4990) jpeg: Image decoder: jd_prepare failed (6)
E (5090) jpeg: Image decoder: jd_prepare failed (6)
E (5200) jpeg: Image decoder: jd_prepare failed (6)
E (5300) jpeg: Image decoder: jd_prepare failed (6)

jack685, 2021-03-15:
esp-idf commit e7435a75f54372e9620b4dbf40e0cce584301c4f (HEAD)

leeebo, 2021-03-15:
@jack685  `jd_prepare failed (6)` means  `JDR_FMT1, /* 6: Data format error (may be damaged data) */`

Seems trying to decode a broken jpeg file.

jack685, 2021-03-15:
@leeebo How to solve this problem？ My hardware is  ESP32-S2-kaluga-1 with OV2640， when i
 enable jpeg mode and report error log.

Lzw655, 2023-05-12:
Sorry for not responding to your question in a timely manner!
Since this problem has lost its time, I will close it after a week. If you still need to solve this problem, please leave a comment or open a new issue, thank you!

closed 2023-05-26 by Lzw655