espressif/esp-dev-kits #29: ESP32-S2-HMI-Devkit-1 Battery NOT included

FBEZ, 2021-05-27:
In the ESP32-S2-HMI-Devkit it's mentioned that  the kit is equipped with the 1950mAh battery but it's not
https://github.com/espressif/esp-dev-kits/blame/master/esp32-s2-hmi-devkit-1/README.md#L18

Shouldn't this line be replaced with "The 1950mAh battery is not included"?

Real-YuZhe, 2021-06-15:
Due to strict regulation on battery export, we are shipping ESP32-S2-HMI-DevKit-1 without battery outside the Chinese mainland. You can buy the Li-ion battery for iphone5, no capacity limitation.

pinkeen, 2021-08-03:
This is great to know. Just a clarification needed - this is the 5G battery to be exact, right?

sukesh-ak, 2021-11-26:
What about the battery connector? Any spec?

closed 2023-05-12 by Lzw655