espressif/esp-dev-kits #31: ESP32-S2-Kaluga-1: Camera example no longer works with V1.2 boards

jepler, 2021-06-02:
I have an ESP32-S2-Kaluga-1 V1.2 with ESP-LyraP-CAM V1.0 and ESP-LyraP-LCD32 V1.1.

At 282a35657a5b32eba656d76096e058965e9ab3da the camera demos sort-of work, though jpeg mode must be disabled (#27) and the image has frequent artifacts (https://github.com/espressif/esp-dev-kits/issues/27#issuecomment-836840782)

At a335f4990cfcbd1b4b4e7c321f18a684aa8556d8 I cannot get this hardware combination to work at all.

Typical debug-uart output:
```
I (569) cpu_start: Starting scheduler on PRO CPU.
I (572) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (572) gpio: GPIO[11]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (582) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (592) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (602) lcd: lcd_buffer_size: 2048, lcd_dma_size: 1024, lcd_dma_node_cnt: 2

I (912) lcd: ST7789 init...

I (912) lcd: lcd init ok

I (912) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (912) cam_hal: cam init ok
I (912) sccb: pin_sda 8 pin_scl 7
I (952) camera: Detected camera at address=0x30
I (1052) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
E (1332) sccb: SCCB_Write Failed addr:0x30, reg:0x07, data:0xc0, ret:-1
I (1332) camera: Detected OV2640 camera
I (1332) cam_hal: buffer_size: 153600, half_buffer_size: 76800, node_buffer_size: 3840, node_cnt: 40, total_cnt: 2

I (1342) cam_hal: Frame[0]: Offset: 16, Addr: 0x3FD80080
I (1342) cam_hal: Frame[1]: Offset: 12, Addr: 0x3FDA5890
I (1352) cam_hal: cam config ok
E (1492) sccb: SCCB_Write Failed addr:0x30, reg:0x6d, data:0x00, ret:-1
E (1492) camera: Failed to set frame size
E (1492) main: Camera Init Failed

```

alic-maker, 2021-06-15:
I just test ok with `ESP32-S2-Kaluga-1 V1.3` because I can't find the board V1.2 for testing. It should also work well with `ESP32-S2-Kaluga-1 V1.2`. 
From the log point of view, it's just that SCCB communication failed. Maybe you will succeed if you try several more times.

sundtek, 2021-08-13:
The camera is not working for the v1.3 devkit

I (894) camera: Detected camera at address=0x30
I (994) camera: Camera PID=0x00 VER=0x00 MIDL=0x3e MIDH=0x00
E (994) camera: Detected camera not supported.
E (994) camera: Camera probe failed with error 0x20004
E (994) main: Camera Init Failed

closed 2023-05-12 by Lzw655