espressif/esp-dev-kits #33: esp32-s2-hmi-devkit-1 all examples build failed

swchow888, 2021-06-09:
All examples build failed with the following error:

../main/main.c:23:10: fatal error: lvgl_port.h: No such file or directory
 #include "lvgl_port.h"
          ^~~~~~~~~~~~~
compilation terminated.
[933/1097] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/sensor.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Real-YuZhe, 2021-06-15:
Hi, did you compile codes under each folder of `examples` folder?

DCSBL, 2021-06-23:
Did you ran `git submodule update --init` (or something similar)? The LVGL needs to be fetched.

swchow888, 2021-06-29:
@DCSBL Thank you very much!  
Finally I can compile the "hello_world" example. Hope there are better documentations and instructions.

DCSBL, 2021-06-29:
@swchow888 well. You can take a look at the readme. There you also find a link to the official ‘getting started’ documentation :)

https://github.com/espressif/esp-dev-kits/tree/master/esp32-s2-hmi-devkit-1#build-projects

swchow888, 2021-06-29:
I built and flashed the 'sensors' in examples. Now the esp32-s2-hmi-devkit-1cannot be turned off. Any idea?

hzslyz, 2021-06-30:
Try to flash power demo.

To make dev board enter sleep mode. You can use the code in `ui_power.c` at function `btn_sleep_cb`.

swchow888, 2021-06-30:
@hzslyz Thanks a lot!
The peripheral  interface connectors are very tiny, what connectors should I use?

hzslyz, 2021-06-30:
The cable is included in the box.

[image]

swchow888, 2021-06-30:
I found them, thanks!
The "smart_panel" cannot be compiled. 
error: 'font_bar_symbol' undeclared

hqs1, 2021-06-30:
**The same problem is encountered, it seems that git can not pull down**

`-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.`

hzslyz, 2021-07-01:
> **The same problem is encountered, it seems that git can not pull down**
>
> `-- Detecting CXX compiler ABI info -- Detecting CXX compiler ABI info - done -- Detecting CXX compile features -- Detecting [...]

Please clone the repo instead of downloading the zip file. And also update sub-modules.

swchow888, 2021-07-01:
@hzslyz Thanks a lot for helping. Now I can compile the examples.

sluurpie, 2021-12-09:
I am trying to build and flash the smart-panel example. But running into errors.
Sorry i am new to this and need help. Much appreciated if someone can point me in the right direction

I get the following errors when trying to build.

[923/1003] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj

../main/main.c:26:10: fatal error: bsp_lcd.h: No such file or directory
 #include "bsp_lcd.h"

[925/1003] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/ws2812.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/ws2812.c.obj

../main/include/ws2812.h:9:10: fatal error: bsp_board.h: No such file or directory
 #include "bsp_board.h"

[926/1003] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/sys_check.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/sys_check.c.obj

/main/app/include/sys_check.h:29:10: fatal error: audio_hal.h: No such file or directory
 #include "audio_hal.h"

[927/1003] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/qcloud.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/qcloud.c.obj

../main/include/qcloud.h:15:10: fatal error: esp_qcloud_iothub.h: No such file or directory
 #include "esp_qcloud_iothub.h"

[940/1003] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/device.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/device.c.obj

../main/include/device.h:32:10: fatal error: bsp_ext_io.h: No such file or directory
 #include "bsp_ext_io.h"

compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Real-YuZhe, 2021-12-09:
Try to clone the repo and update the submodules, then build the demo in the factory_demo folder

sluurpie, 2021-12-09:
I cloned the repo, but am unsure how to update the submodules

Real-YuZhe, 2021-12-09:
`git submodule update --init --recursive --force`.

It is recommended to clone a brand new repo and run this command

sluurpie, 2021-12-09:
Thanks that helped me out.

But all i get is this on the scree
[image]
n

Real-YuZhe, 2021-12-09:
You might have changed the orientation of the LCD. Note that you need to modify the LCD resolution and the rotation direction in the LCD configuration

sluurpie, 2021-12-10:
I just used the set code, could you help point me in the right direction how to adjust?

Lzw655, 2023-05-12:
Sorry for not responding to your question in a timely manner!
Since this problem has lost its time, I will close it after a week. If you still need to solve this problem, please leave a comment or open a new issue, thank you!

closed 2023-05-26 by Lzw655