espressif/esp-dev-kits #36: ESP32S2使用HMI例程驱动ST7789数据位DB0出现延迟

DylanMingan, 2021-07-18:
**环境相关**
vscode插件，IDFV4.3版本
使用**esp32-s2-hmi-devkit-1**例程
硬件是自己画的ESP32S2开发板，lcd采用了ST7789v，16位并行8080协议
已经测试了硬件引脚都正常
参考官方的HMI代码
修改代码以下一些部分
```
scr_interface_create(SCREEN_IFACE_8080, &i2s_lcd_cfg, &iface_drv_i2s);
scr_find_driver(SCREEN_CONTROLLER_ST7789, &lcd);
```
使用逻辑分析仪抓取发现**数据位DB0**的跳变要延迟1us多，其他信号线的跳变都准时，这样最后一位不对就导致了发送的数据不对，是什么原因？
[image]

TDA-2030, 2021-07-19:
我用起来没发现这问题，降低频率或换个硬件试一下呢？

DylanMingan, 2021-07-19:
> 我用起来没发现这问题，降低频率或换个硬件试一下呢？

我换了个官方的开发板，发现确实没有延迟，可以确定是我自己硬件的问题，这里附上lcd的画线图，高亮那个线是DB0，使用的芯片是ESP32S2H4R2
[image]

TDA-2030, 2021-07-19:
看起来与其他线没什么区别，可能DB0使用的IO有点特殊或者焊接的问题？

DylanMingan, 2021-07-19:
> 看起来与其他线没什么区别，可能DB0使用的IO有点特殊或者焊接的问题？

个人感觉PCB画法应该影响不大，可能没等长？DB0用的是GPIO5
[image]
此外可能就是焊接的问题？附上自制的板子图
[image]

TDA-2030, 2021-07-19:
IO5也没听说有什么特殊的，感觉是硬件的锅，换个板子试一下

DylanMingan, 2021-07-19:
> IO5也没听说有什么特殊的，感觉是硬件的锅，换个板子试一下

好的，已经确定是硬件问题，我重新画板打板把！感谢指导！

closed 2021-07-19 by DylanMingan