espressif/esp-dev-kits #38: bsp_lcd: Screen name : [RM68120] | width : [800] | height : [480]�[0m assertion "( xTimeIncrement > 0U )" failed: file "/home/jin/.platformio/packages/framework-espidf/components/freertos/tasks.c", line 1416, function: vTaskDelayUntil

modi12jin, 2021-08-08:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x18d8
load:0x4004c000,len:0x9e8
load:0x40050000,len:0x2b70
entry 0x4004c1ec
[0;32mI (21) boot: ESP-IDF 4.3.0 2nd stage bootloader[0m
[0;32mI (21) boot: compile time 14:09:04[0m
[0;32mI (21) boot: chip revision: 0[0m
[0;32mI (24) boot.esp32s2: SPI Speed      : 80MHz[0m
[0;32mI (29) boot.esp32s2: SPI Mode       : DIO[0m
[0;32mI (33) boot.esp32s2: SPI Flash Size : 4MB[0m
[0;32mI (38) boot: Enabling RNG early entropy source...[0m
[0;32mI (43) boot: Partition Table:[0m
[0;32mI (47) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (54) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (69) boot:  2 factory          factory app      00 00 00010000 003cf000[0m
[0;32mI (77) boot: End of partition table[0m
[0;32mI (81) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=3ce24h (249380) map[0m
[0;32mI (143) esp_image: segment 1: paddr=0004ce4c vaddr=3ffbf690 size=02ad4h ( 10964) load[0m
[0;32mI (146) esp_image: segment 2: paddr=0004f928 vaddr=40022000 size=006f0h (  1776) load[0m
[0;32mI (149) esp_image: segment 3: paddr=00050020 vaddr=40080020 size=34964h (215396) map[0m
[0;32mI (203) esp_image: segment 4: paddr=0008498c vaddr=400226f0 size=0cf9ch ( 53148) load[0m
[0;32mI (218) esp_image: segment 5: paddr=00091930 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (225) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (226) boot: Disabling RNG early entropy source...[0m
[0;32mI (238) cacfx��������Սѥ���cache 	: size 8KB, 4Ways, cache line size 32Byte[0m
[0;32mI (238) cpu_start: Pro cpu up.[0m
[0;32mI (250) cpu_start: Pro cpu start user code[0m
[0;32mI (250) cpu_start: cpu freq: 160000000[0m
[0;32mI (250) cpu_start: Application information:[0m
[0;32mI (254) cpu_start: Project name:     ESP32-S2_ILI9488-8080-8BIT_LVGL[0m
[0;32mI (261) cpu_start: App version:      1[0m
[0;32mI (266) cpu_start: Compile time:     Aug  8 2021 15:59:58[0m
[0;32mI (272) cpu_start: ELF file SHA256:  2da47feb5e9ac017...[0m
[0;32mI (278) cpu_start: ESP-IDF:          4.3.0[0m
[0;32mI (283) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (290) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM[0m
[0;32mI (296) heap_init: At 3FFC3380 len 00038C80 (227 KiB): DRAM[0m
[0;32mI (302) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM[0m
[0;32mI (309) spi_flash: detected chip: generic[0m
[0;32mI (313) spi_flash: flash io: dio[0m
[0;32mI (318) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (323) i2c_bus: i2c0 bus inited[0m
[0;32mI (373) ESP32S2_I2S_LCD: lcd_buffer_size: 32000, lcd_dma_size: 4000, lcd_dma_node_cnt: 8[0m
[0;32mI (373) ESP32S2_I2S_LCD: lcd init ok[0m
[0;32mI (593) lcd rm68120: MADCTL=0x60[0m
[0;32mI (593) bsp_lcd: Screen name : [RM68120] | width : [800] | height : [480][0m
assertion "( xTimeIncrement > 0U )" failed: file "/home/jin/.platformio/packages/framework-espidf/components/freertos/tasks.c", line 1416, function: vTaskDelayUntil

abort() was called at PC 0x40019fbe on core 0

Backtrace:0x400283eb:0x3ffe5a60 0x40028c95:0x3ffe5a80 0x4002e482:0x3ffe5aa0 0x40019fbe:0x3ffe5b10 0x4001a42b:0x3ffe5b30 0x4002a051:0x3ffe5b60 0x40098c92:0x3ffe5b80 0x4002aefd:0x3ffe5ba0

ELF file SHA256: 2da47feb5e9ac017

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40025910
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x18d8
load:0x4004c000,len:0x9e8
[... 88 lines cut ...]
[0;32mI (252) cpu_start: cpu freq: 160000000[0m
[0;32mI (252) cpu_start: Application information:[0m
[0;32mI (257) cpu_start: Project name:     ESP32-S2_ILI9488-8080-8BIT_LVGL[0m
[0;32mI (264) cpu_start: App version:      1[0m
[0;32mI (268) cpu_start: Compile time:     Aug  8 2021 15:59:58[0m
[0;32mI (274) cpu_start: ELF file SHA256:  2da47feb5e9ac017...[0m
[0;32mI (280) cpu_start: ESP-IDF:          4.3.0[0m
[0;32mI (285) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (292) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM[0m
[0;32mI (298) heap_init: At 3FFC3380 len 00038C80 (227 KiB): DRAM[0m
[0;32mI (305) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM[0m
[0;32mI (311) spi_flash: detected chip: generic[0m
[0;32mI (316) spi_flash: flash io: dio[0m
[0;32mI (320) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (326) i2c_bus: i2c0 bus inited[0m
[0;32mI (376) ESP32S2_I2S_LCD: lcd_buffer_size: 32000, lcd_dma_size: 4000, lcd_dma_node_cnt: 8[0m
[0;32mI (376) ESP32S2_I2S_LCD: lcd init ok[0m
[0;32mI (596) lcd rm68120: MADCTL=0x60[0m
[0;32mI (596) bsp_lcd: Screen name : [RM68120] | width : [800] | height : [480][0m
assertion "( xTimeIncrement > 0U )" failed: file "/home/jin/.platformio/packages/framework-espidf/components/freertos/tasks.c", line 1416, function: vTaskDelayUntil

abort() was called at PC 0x40019fbe on core 0

Backtrace:0x400283eb:0x3ffe5a60 0x40028c95:0x3ffe5a80 0x4002e482:0x3ffe5aa0 0x40019fbe:0x3ffe5b10 0x4001a42b:0x3ffe5b30 0x4002a051:0x3ffe5b60 0x40098c92:0x3ffe5b80 0x4002aefd:0x3ffe5ba0
```

Real-YuZhe, 2021-09-08:
Try to change FreeRTOS tick rate to 1000 Hz

closed 2021-09-08 by modi12jin

weilian1977, 2023-01-26:
关注一下

Lzw655, 2023-01-29:
> 关注一下

请问你有什么问题吗？