espressif/esp-dev-kits #40: Cmake Error occurs in esp32-s2-hmi-devkit-1 examples while building

Wildpapa, 2021-08-24:
After cloned the esp-dev-kits, I tried to build the hello_world in  esp32-s2-hmi-devkit-1. But dosent work. The Error while building as follows:
"CMake Error at C:/esp-idf/tools/cmake/component.cmake:274 (message):
  SRC_DIRS entry 'lvgl/src/lv_core' does not exist.
Call Stack (most recent call first):
  C:/esp-idf/tools/cmake/component.cmake:435 (__component_add_sources)
  C:/esp-dev-kits/esp32-s2-hmi-devkit-1/components/lvgl/CMakeLists.txt:1 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "C:/esp-dev-kits/esp32-s2-hmi-devkit-1/examples/get-started/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1"

need assist!!!  Thanks in advance!!!

zhongluqiang, 2021-08-25:
I guess you forgot to update git sub modules, try `git submodule init` and `git submodule update` under `esp-dev-kits` directory.

Lzw655, 2023-05-12:
很抱歉没有及时回答你的问题！
由于这个问题已经失去时效性了，我将在一周后关闭它。如果你还需要解决这个问题，请留言或新开一个 issue，谢谢！

closed 2023-05-26 by Lzw655