espressif/esp-dev-kits #41: Build of esp32-s2-kaluga-1/examples/lcd fails in VS Code ESP-IDF Extension - No such file or directory  #include "soc/system_reg.h"

georgik, 2021-09-08:
How to simulate:
- clone esp-dev-kits
- open esp32-s2-kaluga-1/examples/lcd
- CTRL+Shift+P, select ESP-IDF: Build, flash...
- the build fails:
```
esp-dev-kits/esp32-s2-kaluga-1/components/lcd/lcd.c:22:10: fatal error: soc/system_reg.h: No such file or directory
 #include "soc/system_reg.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

Root cause: 
- VS Code ESP-IDF Extension does not select esp32s2 target as default. The build from command line ESP-IDF works and esp32s2 is correctly set

Workaround:
- CTRL+Shift+P, select ESP-IDF: Set Espressif device target, select ESP32-S2
- run build

Expected behavior: VS Code ESP-IDF Extension picks the target from the project, so that user do not need to set the target manually

FYI: @brianignacio5

brianignacio5, 2021-09-09:
Currently the extension has a configuration setting `idf.adapterTargetName` with `esp32` as default value which set IDF_TARGET environment variable for the build task.

I'll make a fix to get the IDF_TARGET from sdkconfig on selected workspace folder change.

georgik, 2021-09-14:
Tested with new version of VS Code. The chip is selected correctly and it's also visible in the bottom bar. Thank you.

closed 2021-09-14 by georgik