espressif/esp-dev-kits #47: [question] ESP32-S3-USB-OTG

chegewara, 2021-12-12:
As we can see espressif is selling now on alliexpress `ESP32-S3-USB-OTG`
https://www.aliexpress.com/item/1005003580502244.html
In product description is link to this repository, but it is not working:
https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-usb-otg

Any idea when it will be updated?

Thanks

leeebo, 2021-12-13:
Sorry, due to the preparation of online documentation. The esp32-s3-usb-otg guide is transferred to RST format and removed to https://github.com/espressif/esp-dev-kits/blob/master/docs/en/esp32s3/esp32-s3-usb-otg/user_guide.rst

We will fix the above URL as soon as possible.

chegewara, 2021-12-13:
Thanks, it is good docu. I think the issue may be closed.

leeebo, 2021-12-13:
@chegewara Please keep the issue open, until the URL fixed.

closed 2022-01-09 by chegewara