espressif/esp-dev-kits #58: ESP32-S2-HMI-DevKit-1 with LVGL version 8.x => Not working (IDF-Version 4.3) (IDFGH-8413)

VigneshAkshaya, 2022-09-29:


Lzw655, 2022-09-29:
@VigneshAkshaya Please attach your new code which porting to v8.x.

VigneshAkshaya, 2022-09-29:
Please find the attachment
[SampleProgramLVGL8.2.zip](https://github.com/espressif/esp-dev-kits/files/9674426/SampleProgramLVGL8.2.zip)

Lzw655, 2022-10-08:
@VigneshAkshaya I have reviewed your project, and I found there're some uncorrect configurations. You can modify them as follow:
* SampleProgramLVGL8.2/components/lvgl/lv_conf.h:
  * Set `LV_MEM_CUSTOM` to 1
  * Set `LV_MEMCPY_MEMSET_STD` to 1
[image]
* SampleProgramLVGL8.2/components/lvgl/lv_port/lvgl_port.c:
  * Modify `disp_drv.hor_res` and `disp_drv.hor_res`
[image]
  * Modify parameters of `bsp_lcd_flush`
[image]

VigneshAkshaya, 2022-10-13:
@Lzw655 
I have tried with the above configurations and it is working fine for labels, but I dont know why it is not working for other widgets.

•	Button was created and added to the existing “Hello World “ text – Only “Button” text label is getting displayed.
•	Event was assigned to the button to follow up the button event – It’s not clickable and given event also hasn’t happened.
•	And , widgets such as bar , slider , switch were also tested and only text is getting displayed for all .

Changes I have made,
	main\hello_world_main.c -> btn1 is created and its corresponding event handler .
[image]

Hardware Observation as below

[image]

Lzw655, 2022-10-14:
@VigneshAkshaya I checked the file "components\lvgl\lv_conf.h" in your project again, and I found the configuration "LV_USE_THEME_DEFAULT" is set to 0 which should be 1. This caused LVGL's widgets to have no default appearance.

Lzw655, 2022-10-24:
@VigneshAkshaya Is your problem solved? If so, please close this issue.

VigneshAkshaya, 2022-10-28:
Hey @Lzw655 , sorry for the delay, I was unavailable for whole past 2 weeks.

•	“LV_USE_THEME_DEFAULT”  was set it to 1.Dsiplay is changed and there was no response on clicking the button
•	I2C Bus and LCD Touch Init are also happening properly . Verified through ESP ERROR CHECK and print statements for all the init functions.
•	LV_EVENT_CLICKED given to the event handler is not working whereas other default events are happening – verified by printing event codes.

[image]

Lzw655, 2023-05-12:
Sorry for not responding to your question in a timely manner!
Since this problem has lost its time, I will close it after a week. If you still need to solve this problem, please leave a comment or open a new issue, thank you!

closed 2023-05-26 by Lzw655