espressif/esp-dev-kits #60: ESP32-S3-LCD-EV-BOARD Development Board 对idf版本和环境要求？

weilian1977, 2022-10-17:
Linux 22.04+4.4.2可以成功，Linux 22.04+master老是提示错误，对Linux、Cmake、python版本有什么要求？

Lzw655, 2022-10-17:
@weilian1977 麻烦提供一下错误信息，方便定位到问题

weilian1977, 2022-10-17:
weilian@weilian-OptiPlex-7080:~/esp-idf$ ./install.sh esp32s3
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.6
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Current system platform: linux-amd64
Updating /home/weilian/.espressif/idf-env.json
Selected targets are: esp32s3
Installing tools: xtensa-esp-elf-gdb, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@11.2_20220823 (already installed)
Skipping xtensa-esp32s3-elf@esp-2022r1-11.2.0 (already installed)
Skipping riscv32-esp-elf@esp-2022r1-11.2.0 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20220706 (already installed)
Skipping esp-rom-elfs@20220823 (already installed)
Installing Python environment and packages
Python 3.10.6
pip 22.2.2 from /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/pip (python 3.10)
Upgrading pip and setuptools...
Requirement already satisfied: pip in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (22.2.2)
Collecting pip
  Downloading pip-22.3-py3-none-any.whl (2.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 2.8 MB/s eta 0:00:00
Requirement already satisfied: setuptools in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (65.4.1)
Collecting setuptools
  Downloading setuptools-65.5.0-py3-none-any.whl (1.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 624.2 kB/s eta 0:00:00
Installing collected packages: setuptools, pip
  Attempting uninstall: setuptools
    Found existing installation: setuptools 65.4.1
    Uninstalling setuptools-65.4.1:
      Successfully uninstalled setuptools-65.4.1
  Attempting uninstall: pip
    Found existing installation: pip 22.2.2
    Uninstalling pip-22.2.2:
      Successfully uninstalled pip-22.2.2
Successfully installed pip-22.3 setuptools-65.5.0
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.1.txt
Destination: /home/weilian/.espressif/espidf.constraints.v5.1.txt.tmp
WARNING: Download failure: <urlopen error [Errno -3] Temporary failure in name resolution>
WARNING: Failed to download https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.1.txt to /home/weilian/.espressif/espidf.constraints.v5.1.txt.tmp
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.1.txt
Destination: /home/weilian/.espressif/espidf.constraints.v5.1.txt.tmp
Done
Installing Python packages
 Constraint file: /home/weilian/.espressif/espidf.constraints.v5.1.txt
 Requirement files:
  - /home/weilian/esp-idf/tools/requirements/requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring windows-curses: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 3)) (65.5.0)
Requirement already satisfied: click in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 4)) (8.1.3)
Requirement already satisfied: pyserial in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 5)) (3.5)
Requirement already satisfied: cryptography in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 6)) (38.0.2)
Requirement already satisfied: pyparsing in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 7)) (3.0.9)
Requirement already satisfied: pyelftools in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 8)) (0.29)
Requirement already satisfied: idf-component-manager in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 9)) (1.1.4)
Requirement already satisfied: esp-coredump in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 10)) (1.4)
Collecting esp-coredump
  Downloading https://dl.espressif.com/pypi/esp-coredump/esp_coredump-1.4.1-py3-none-any.whl (33 kB)
Requirement already satisfied: esptool in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 11)) (4.3)
Requirement already satisfied: kconfiglib in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 14)) (14.1.0)
Requirement already satisfied: freertos_gdb in /home/weilian/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages (from -r /home/weilian/esp-idf/tools/requirements/requirements.core.txt (line 18)) (1.0.1)
[... cut ...]
Call Stack (most recent call first):
  /home/weilian/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/weilian/esp-idf/examples/get-started/blink/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/weilian/esp-idf/examples/get-started/blink/build/log/idf_py_stderr_output_9733 and /home/weilian/esp-idf/examples/get-started/blink/build/log/idf_py_stdout_output_9733

Lzw655, 2022-10-17:
@weilian1977 这个报错应该是因为 master 的 example 里采用了[包管理器](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html?highlight=component%20manager)来下载依赖组件，但是你的设备网络无法访问到我们的服务器，所以导致网络连接错误。

你可以试试不使用包管理器，比如编译 examples/get-started/hello_world 这个用例，应该是不会报错的

weilian1977, 2022-10-17:
4.4.2的报错信息

weilian@weilian-OptiPlex-7080:~/esp/esp442/esp-idf$ ./install.sh
Detecting the Python interpreter
Checking "python" ...
Python 3.10.6
"python" has been detected
Installing ESP-IDF tools
Current system platform: linux-amd64
Traceback (most recent call last):
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1860, in <module>
    main(sys.argv[1:])
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1856, in main
    action_func(args)
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1423, in action_install
    targets = clean_targets(args.targets)
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1085, in clean_targets
    export_targets_to_idf_env_json(targets_from_tools_json)
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1053, in export_targets_to_idf_env_json
    targets = list(set(targets + get_user_defined_targets()))
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1099, in get_user_defined_targets
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'
weilian@weilian-OptiPlex-7080:~/esp/esp442/esp-idf$ sudo apt-get install git wget flex bison gperf python3 python3-pip python3-setuptools cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0
[sudo] weilian 的密码： 
正在读取软件包列表... 完成
正在分析软件包的依赖关系树... 完成
正在读取状态信息... 完成                 
bison 已经是最新版 (2:3.8.2+dfsg-1build1)。
flex 已经是最新版 (2.6.4-8build2)。
libffi-dev 已经是最新版 (3.4.2-4)。
python3-setuptools 已经是最新版 (59.6.0-1.2)。
wget 已经是最新版 (1.21.2-2ubuntu1)。
ccache 已经是最新版 (4.5.1-1)。
dfu-util 已经是最新版 (0.9-1)。
gperf 已经是最新版 (3.1-1build1)。
ninja-build 已经是最新版 (1.10.1-1)。
python3-pip 已经是最新版 (22.0.2+dfsg-1)。
cmake 已经是最新版 (3.22.1-1ubuntu1.22.04.1)。
git 已经是最新版 (1:2.34.1-1ubuntu1.4)。
libssl-dev 已经是最新版 (3.0.2-0ubuntu1.6)。
libusb-1.0-0 已经是最新版 (2:1.0.25-1ubuntu2)。
python3 已经是最新版 (3.10.6-1~22.04)。
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 8 个软件包未被升级。
weilian@weilian-OptiPlex-7080:~/esp/esp442/esp-idf$ ./install.sh
Detecting the Python interpreter
Checking "python" ...
Python 3.10.6
"python" has been detected
Installing ESP-IDF tools
Current system platform: linux-amd64
Traceback (most recent call last):
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1860, in <module>
    main(sys.argv[1:])
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1856, in main
    action_func(args)
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1423, in action_install
    targets = clean_targets(args.targets)
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1085, in clean_targets
    export_targets_to_idf_env_json(targets_from_tools_json)
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1053, in export_targets_to_idf_env_json
    targets = list(set(targets + get_user_defined_targets()))
  File "/home/weilian/esp/esp442/esp-idf/tools/idf_tools.py", line 1099, in get_user_defined_targets
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'
weilian@weilian-OptiPlex-7080:~/esp/esp442/esp-idf$

Lzw655, 2022-10-17:
@weilian1977 目前 IDF 的 master 在切换环境时存在 bug，需要删除用户目录下 .espressif/idf-env.json 文件之后再执行其他操作

weilian1977, 2022-10-17:
> @weilian1977 目前 IDF 的 master 在切换环境时存在 bug，需要删除用户目录下 .espressif/idf-env.json 文件之后再执行其他操作

可以理解为配置了IDF   master之后必须删除用户目录下 .espressif/idf-env.json 文件之后才能使用其他版本的IDF？

Lzw655, 2022-10-17:
不一定，跟你的系统配置、软件环境有关，我这里切换版本就没有遇到这种问题

weilian1977, 2022-10-17:
删除之后4.4.2基本正常了，但是编译esp32-s3-lcd-ev-board/examples/lvgl_demo报错如下，是不是这个工程只能在master下编译？

weilian@weilian-OptiPlex-7080:~/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos$ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos"...
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
CMake Error at /home/weilian/esp/esp442/esp-idf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'esp_adc'.
Call Stack (most recent call first):
  /home/weilian/esp/esp442/esp-idf/tools/cmake/build.cmake:227 (__build_resolve_and_add_req)
  /home/weilian/esp/esp442/esp-idf/tools/cmake/build.cmake:228 (__build_expand_requirements)
  /home/weilian/esp/esp442/esp-idf/tools/cmake/build.cmake:509 (__build_expand_requirements)
  /home/weilian/esp/esp442/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:11 (project)

-- Configuring incomplete, errors occurred!
See also "/home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

weilian1977, 2022-10-17:
> 不一定，跟你的系统配置、软件环境有关，我这里切换版本就没有遇到这种问题

您这面是什么系统配置?软件环境？

Lzw655, 2022-10-17:
我是用的 Manjaro Linux 系统，然后通过 anaconda 来管理 python 环境

Lzw655, 2022-10-17:
> 删除之后4.4.2基本正常了，但是编译esp32-s3-lcd-ev-board/examples/lvgl_demo报错如下，是不是这个工程只能在master下编译？
>
> weilian@weilian-OptiPlex-7080:~/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos$ idf.py menuconfig [...]

esp32-s3-lcd-ev-board/examples 里面的工程需要 IDF master 的版本在 release/v5.0 及以上，并且需要打上 PSRAM 120M 的补丁，详细说明见 https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-lcd-ev-board/examples/factory#idf-patch

weilian1977, 2022-10-17:
关闭4.4.2终端，删除 .espressif/idf-env.json 文件 ，信息如下
weilian@weilian-OptiPlex-7080:~/esp-idf$ . export.sh
Setting IDF_PATH to '/home/weilian/esp-idf'
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.6
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
ERROR: tool riscv32-esp-elf-gdb has no installed versions. Please run '/usr/bin/python3 /home/weilian/esp-idf/tools/idf_tools.py install' to install it.
ERROR: tool xtensa-esp32-elf has no installed versions. Please run '/usr/bin/python3 /home/weilian/esp-idf/tools/idf_tools.py install' to install it.
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/usr/bin/python3 /home/weilian/esp-idf/tools/idf_tools.py install' to install it.
按照提示进行安装
weilian@weilian-OptiPlex-7080:~/esp-idf$ /usr/bin/python3 /home/weilian/esp-idf/tools/idf_tools.py install
Current system platform: linux-amd64
Updating /home/weilian/.espressif/idf-env.json
Selected targets are: esp32c6, esp32c3, esp32c2, esp32h2, esp32, esp32s2, esp32s3
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@11.2_20220823 (already installed)
Installing riscv32-esp-elf-gdb@11.2_20220823
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v11.2_20220823/riscv32-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz
Destination: /home/weilian/.espressif/dist/riscv32-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz.tmp
Done
Extracting /home/weilian/.espressif/dist/riscv32-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz to /home/weilian/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823
Installing xtensa-esp32-elf@esp-2022r1-11.2.0
Downloading https://github.com/espressif/crosstool-NG/releases/download/esp-2022r1/xtensa-esp32-elf-gcc11_2_0-esp-2022r1-linux-amd64.tar.xz
Destination: /home/weilian/.espressif/dist/xtensa-esp32-elf-gcc11_2_0-esp-2022r1-linux-amd64.tar.xz.tmp
Done
Extracting /home/weilian/.espressif/dist/xtensa-esp32-elf-gcc11_2_0-esp-2022r1-linux-amd64.tar.xz to /home/weilian/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0
Installing xtensa-esp32s2-elf@esp-2022r1-11.2.0
Downloading https://github.com/espressif/crosstool-NG/releases/download/esp-2022r1/xtensa-esp32s2-elf-gcc11_2_0-esp-2022r1-linux-amd64.tar.xz
Destination: /home/weilian/.espressif/dist/xtensa-esp32s2-elf-gcc11_2_0-esp-2022r1-linux-amd64.tar.xz.tmp
Done
Extracting /home/weilian/.espressif/dist/xtensa-esp32s2-elf-gcc11_2_0-esp-2022r1-linux-amd64.tar.xz to /home/weilian/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0
Skipping xtensa-esp32s3-elf@esp-2022r1-11.2.0 (already installed)
Skipping riscv32-esp-elf@esp-2022r1-11.2.0 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20220706 (already installed)
Skipping esp-rom-elfs@20220823 (already installed)

weilian@weilian-OptiPlex-7080:~/esp-idf$ . export.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.6
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Constraint file: /home/weilian/.espressif/espidf.constraints.v5.1.txt
Requirement files:
 - /home/weilian/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/weilian/.espressif/python_env/idf5.1_py3.10_env/bin/python
Python requirements are satisfied.
Added the following directories to PATH:
  /home/weilian/esp-idf/components/esptool_py/esptool
  /home/weilian/esp-idf/components/espcoredump
  /home/weilian/esp-idf/components/partition_table
  /home/weilian/esp-idf/components/app_update
  /home/weilian/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin
  /home/weilian/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin
  /home/weilian/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin
  /home/weilian/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin
  /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin
  /home/weilian/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin
  /home/weilian/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /home/weilian/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin
  /home/weilian/.espressif/python_env/idf5.1_py3.10_env/bin
  /home/weilian/esp-idf/tools

[... 34 lines cut ...]
-- Check for working C compiler: /home/weilian/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/weilian/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/weilian/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/weilian/esp-idf/examples/get-started/blink/sdkconfig
[... cut ...]
  /home/weilian/esp-idf/tools/cmake/build.cmake:398 (include)
  /home/weilian/esp-idf/tools/cmake/build.cmake:617 (__build_process_project_includes)
  /home/weilian/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "/home/weilian/esp-idf/examples/get-started/blink/build/CMakeFiles/CMakeOutput.log".
See also "/home/weilian/esp-idf/examples/get-started/blink/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

weilian1977, 2022-10-17:
> > 删除之后4.4.2基本正常了，但是编译esp32-s3-lcd-ev-board/examples/lvgl_demo报错如下，是不是这个工程只能在master下编译？
> > weilian@weilian-OptiPlex-7080:~/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos$ idf.py menuconfig Executing action: menuconfig Running cmake in directory /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos"... -- Found Git: /usr/bin/git (found version "2.34.1") -- The C compiler identification is GNU 8.4.0 -- The CXX compiler identification is GNU 8.4.0 -- The ASM compiler identification is GNU -- Found assembler: /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- Check for working C compiler: /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- Check for working C compiler: /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- works -- Detecting C compiler ABI info -- Detecting C compiler ABI info - done -- Detecting C compile features -- Detecting C compile features - done -- Check for working CXX compiler: /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -- Check for working CXX compiler: /home/weilian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -- works -- Detecting CXX compiler ABI info -- Detecting CXX compiler ABI info - done -- Detecting CXX compile features -- Detecting CXX compile features - done -- Building ESP-IDF components for target esp32s3 CMake Error at /home/weilian/esp/esp442/esp-idf/tools/cmake/build.cmake:201 (message): Failed to resolve component 'esp_adc'. Call Stack (most recent call first): /home/weilian/esp/esp442/esp-idf/tools/cmake/build.cmake:227 (__build_resolve_and_add_req) /home/weilian/esp/esp442/esp-idf/tools/cmake/build.cmake:228 (__build_expand_requirements) /home/weilian/esp/esp442/esp-idf/tools/cmake/build.cmake:509 (__build_expand_requirements) /home/weilian/esp/esp442/esp-idf/tools/cmake/project.cmake:384 (idf_build_process) CMakeLists.txt:11 (project)
> > -- Configuring incomplete, errors occurred! See also "/home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/CMakeFiles/CMakeOutput.log". cmake failed with exit code 1
> 
> esp32-s3-lcd-ev-board/examples 里面的工程需要 IDF 的版本在 release/v5.0 及以上，并且需要打上 PSRAM 120M 的补丁，详细说明见 https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-lcd-ev-board/examples/factory#idf-patch

> 现在的问题是就配置不出master的编译环境，还到不了打补丁这里

Lzw655, 2022-10-17:
切换 commit 或者 branch 后需要确保 git 仓库干净，用 git status 查看有没有 untrack file 之类的

weilian1977, 2022-10-17:
> 我把4.4.2和master放在了不同的文件夹下

weilian1977, 2022-10-17:
> 
没有运行

Lzw655, 2022-10-17:
先运行一下吧，然后重新 ./install.sh 一下

weilian1977, 2022-10-17:
我的理解，git代码后想用不同版本的代码直接git checkout 指定版本后重新 ./install.sh？不需要不同的版本放在不同的文件夹下？

weilian1977, 2022-10-17:
重新git之后
weilian@weilian-OptiPlex-7080:~/esp-idf$ git checkout --recurse-submodules master
M	.editorconfig
M	.flake8
D	.github/ISSUE_TEMPLATE/01_build_install_bug.yml
D	.github/ISSUE_TEMPLATE/02_runtime_bug.yml
D	.github/ISSUE_TEMPLATE/03_feature_request.yml
D	.github/ISSUE_TEMPLATE/04_other_issue.yml
M	.github/ISSUE_TEMPLATE/config.yml
M	.github/workflows/pre_commit_check.yml
M	.gitlab-ci.yml
M	.gitlab/CODEOWNERS
M	.gitlab/ci/README.md
M	.gitlab/ci/assign-test.yml
M	.gitlab/ci/build.yml
D	.gitlab/ci/default-build-test-rules.yml
M	.gitlab/ci/dependencies/README.md
M	.gitlab/ci/dependencies/dependencies.yml
M	.gitlab/ci/deploy.yml
M	.gitlab/ci/docs.yml
M	.gitlab/ci/host-test.yml
M	.gitlab/ci/pre_check.yml
M	.gitlab/ci/rules.yml
M	.gitlab/ci/static-code-analysis.yml
M	.gitlab/ci/target-test.yml
M	.gitlab/ci/upload_cache.yml
M	.gitmodules
M	.pre-commit-config.yaml
M	Kconfig
M	components/README.md
M	components/app_trace/sys_view/Sample/Config/SEGGER_SYSVIEW_Config_FreeRTOS.c
M	components/app_trace/sys_view/Sample/OS/SEGGER_SYSVIEW_FreeRTOS.c
M	components/app_trace/sys_view/Sample/OS/SEGGER_SYSVIEW_FreeRTOS.h
M	components/app_trace/test/CMakeLists.txt
M	components/app_update/CMakeLists.txt
D	components/app_update/esp_ota_app_desc.c
M	components/app_update/esp_ota_ops.c
M	components/app_update/include/esp_ota_ops.h
M	components/app_update/test/CMakeLists.txt
M	components/app_update/test/test_ota_ops.c
M	components/bootloader/Kconfig.projbuild
M	components/bootloader/project_include.cmake
M	components/bootloader/subproject/CMakeLists.txt
D	components/bootloader/subproject/main/ld/esp32c6/bootloader.ld
D	components/bootloader/subproject/main/ld/esp32c6/bootloader.rom.ld
D	components/bootloader_support/.build-test-rules.yml
M	components/bootloader_support/CMakeLists.txt
M	components/bootloader_support/bootloader_flash/src/bootloader_flash.c
D	components/bootloader_support/bootloader_flash/src/bootloader_flash_config_esp32c6.c
M	components/bootloader_support/bootloader_flash/src/flash_qio_mode.c
M	components/bootloader_support/include/bootloader_common.h
M	components/bootloader_support/include/bootloader_memory_utils.h
M	components/bootloader_support/include/esp_app_format.h
M	components/bootloader_support/include/esp_image_format.h
M	components/bootloader_support/private_include/bootloader_signature.h
M	components/bootloader_support/src/bootloader_clock_init.c
M	components/bootloader_support/src/bootloader_common_loader.c
M	components/bootloader_support/src/bootloader_console.c
M	components/bootloader_support/src/bootloader_efuse.c
M	components/bootloader_support/src/bootloader_init.c
M	components/bootloader_support/src/bootloader_mem.c
M	components/bootloader_support/src/bootloader_panic.c
D	components/bootloader_support/src/bootloader_random_esp32c6.c
M	components/bootloader_support/src/bootloader_utility.c
D	components/bootloader_support/src/esp32c6/bootloader_esp32c6.c
D	components/bootloader_support/src/esp32c6/bootloader_sha.c
D	components/bootloader_support/src/esp32c6/bootloader_soc.c
D	components/bootloader_support/src/esp32c6/flash_encryption_secure_features.c
D	components/bootloader_support/src/esp32c6/secure_boot_secure_features.c
M	components/bootloader_support/src/esp_image_format.c
[... 2233 lines cut ...]
D	tools/test_apps/system/panic/test_panic_util/__init__.py
D	tools/test_apps/system/panic/test_panic_util/panic_dut.py
D	tools/test_build_system/MIGRATION.md
D	tools/test_build_system/README.md
D	tools/test_build_system/build_test_app/CMakeLists.txt
D	tools/test_build_system/build_test_app/README.md
D	tools/test_build_system/build_test_app/main/CMakeLists.txt
D	tools/test_build_system/build_test_app/main/build_test_app.c
D	tools/test_build_system/conftest.py
D	tools/test_build_system/pytest.ini
D	tools/test_build_system/test_build_system_helpers/__init__.py
D	tools/test_build_system/test_build_system_helpers/editing.py
D	tools/test_build_system/test_build_system_helpers/idf_utils.py
D	tools/test_build_system/test_build_system_helpers/snapshot.py
D	tools/test_build_system/test_common.py
D	tools/test_build_system/test_rebuild.py
D	tools/test_idf_py/error_output.yml
D	tools/test_idf_py/test_hints.py
D	tools/test_idf_size/expected_output.csv
D	tools/unit-test-app/components/test_utils/test/leak_test.c
D	tools/unit-test-app/configs/spi_flash_config_s3
已经位于 'master'
您的分支与上游分支 'origin/master' 一致。
weilian@weilian-OptiPlex-7080:~/esp-idf$ git checkout --recurse-submodules f3159
fatal: 不是 git 仓库：examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib/../../../../../../.git/modules/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib

weilian1977, 2022-10-17:
weilian@weilian-OptiPlex-7080:~$ git clone  --recursive https://github.com/espressif/esp-idf.git
正克隆到 'esp-idf'...
remote: Enumerating objects: 382836, done.
remote: Counting objects: 100% (3009/3009), done.
remote: Compressing objects: 100% (1483/1483), done.
remote: Total 382836 (delta 1651), reused 2785 (delta 1465), pack-reused 379827
接收对象中: 100% (382836/382836), 208.38 MiB | 6.57 MiB/s, 完成.
处理 delta 中: 100% (282159/282159), 完成.
子模组 'components/bootloader/subproject/components/micro-ecc/micro-ecc'（https://github.com/kmackay/micro-ecc.git）已对路径 'components/bootloader/subproject/components/micro-ecc/micro-ecc' 注册
子模组 'components/bt/controller/lib_esp32'（https://github.com/espressif/esp32-bt-lib.git）已对路径 'components/bt/controller/lib_esp32' 注册
子模组 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib'（https://github.com/espressif/esp32c2-bt-lib.git）已对路径 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib' 注册
子模组 'components/bt/controller/lib_esp32c3_family'（https://github.com/espressif/esp32c3-bt-lib.git）已对路径 'components/bt/controller/lib_esp32c3_family' 注册
子模组 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib'（https://github.com/espressif/esp32h2-bt-lib.git）已对路径 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib' 注册
子模组 'components/bt/host/nimble/nimble'（https://github.com/espressif/esp-nimble.git）已对路径 'components/bt/host/nimble/nimble' 注册
子模组 'components/cmock/CMock'（https://github.com/ThrowTheSwitch/CMock.git）已对路径 'components/cmock/CMock' 注册
子模组 'components/esp_phy/lib'（https://github.com/espressif/esp-phy-lib.git）已对路径 'components/esp_phy/lib' 注册
子模组 'components/esp_wifi/lib'（https://github.com/espressif/esp32-wifi-lib.git）已对路径 'components/esp_wifi/lib' 注册
子模组 'components/heap/tlsf'（https://github.com/espressif/tlsf.git）已对路径 'components/heap/tlsf' 注册
子模组 'components/ieee802154/lib'（https://github.com/espressif/esp-ieee802154-lib.git）已对路径 'components/ieee802154/lib' 注册
子模组 'components/json/cJSON'（https://github.com/DaveGamble/cJSON.git）已对路径 'components/json/cJSON' 注册
子模组 'components/lwip/lwip'（https://github.com/espressif/esp-lwip.git）已对路径 'components/lwip/lwip' 注册
子模组 'components/mbedtls/mbedtls'（https://github.com/espressif/mbedtls.git）已对路径 'components/mbedtls/mbedtls' 注册
子模组 'components/mqtt/esp-mqtt'（https://github.com/espressif/esp-mqtt.git）已对路径 'components/mqtt/esp-mqtt' 注册
子模组 'components/openthread/lib'（https://github.com/espressif/esp-thread-lib.git）已对路径 'components/openthread/lib' 注册
子模组 'components/openthread/openthread'（https://github.com/espressif/openthread.git）已对路径 'components/openthread/openthread' 注册
子模组 'components/protobuf-c/protobuf-c'（https://github.com/protobuf-c/protobuf-c.git）已对路径 'components/protobuf-c/protobuf-c' 注册
子模组 'components/spiffs/spiffs'（https://github.com/pellepl/spiffs.git）已对路径 'components/spiffs/spiffs' 注册
子模组 'components/tinyusb/tinyusb'（https://github.com/espressif/tinyusb.git）已对路径 'components/tinyusb/tinyusb' 注册
子模组 'components/unity/unity'（https://github.com/ThrowTheSwitch/Unity.git）已对路径 'components/unity/unity' 注册
正克隆到 '/home/weilian/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc'...
fatal: 无法访问 'https://github.com/kmackay/micro-ecc.git/'：Could not resolve host: github.com
fatal: 无法克隆 'https://github.com/kmackay/micro-ecc.git' 到子模组路径 '/home/weilian/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc'
克隆 'components/bootloader/subproject/components/micro-ecc/micro-ecc' 失败。按计划重试
正克隆到 '/home/weilian/esp-idf/components/bt/controller/lib_esp32'...
fatal: 无法访问 'https://github.com/espressif/esp32-bt-lib.git/'：Could not resolve host: github.com
fatal: 无法克隆 'https://github.com/espressif/esp32-bt-lib.git' 到子模组路径 '/home/weilian/esp-idf/components/bt/controller/lib_esp32'
克隆 'components/bt/controller/lib_esp32' 失败。按计划重试
正克隆到 '/home/weilian/esp-idf/components/bt/controller/lib_esp32c2/esp32c2-bt-lib'...
remote: Enumerating objects: 42, done.        
remote: Counting objects: 100% (42/42), done.        
remote: Compressing objects: 100% (30/30), done.        
remote: Total 42 (delta 21), reused 31 (delta 10), pack-reused 0        
接收对象中: 100% (42/42), 1.08 MiB | 1.51 MiB/s, 完成.
处理 delta 中: 100% (21/21), 完成.
正克隆到 '/home/weilian/esp-idf/components/bt/controller/lib_esp32c3_family'...
remote: Enumerating objects: 224, done.        
remote: Counting objects: 100% (224/224), done.        
remote: Compressing objects: 100% (140/140), done.        
remote: Total 224 (delta 103), reused 166 (delta 45), pack-reused 0        
接收对象中: 100% (224/224), 677.66 KiB | 1.05 MiB/s, 完成.
处理 delta 中: 100% (103/103), 完成.
正克隆到 '/home/weilian/esp-idf/components/bt/controller/lib_esp32h2/esp32h2-bt-lib'...
remote: Enumerating objects: 24, done.        
remote: Counting objects: 100% (24/24), done.        
remote: Compressing objects: 100% (17/17), done.        
remote: Total 24 (delta 4), reused 22 (delta 2), pack-reused 0        
接收对象中: 100% (24/24), 1.37 MiB | 1.68 MiB/s, 完成.
处理 delta 中: 100% (4/4), 完成.
正克隆到 '/home/weilian/esp-idf/components/bt/host/nimble/nimble'...
remote: Enumerating objects: 44489, done.        
remote: Counting objects: 100% (49/49), done.        
remote: Compressing objects: 100% (34/34), done.        
remote: Total 44489 (delta 24), reused 38 (delta 15), pack-reused 44440        
接收对象中: 100% (44489/44489), 13.10 MiB | 5.89 MiB/s, 完成.
处理 delta 中: 100% (27581/27581), 完成.
正克隆到 '/home/weilian/esp-idf/components/cmock/CMock'...
remote: Enumerating objects: 5764, done.        
[... cut ...]
weilian@weilian-OptiPlex-7080:~/esp-idf$ git checkout --recurse-submodules f3159
fatal: 不是 git 仓库：../../../../../../.git/modules/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib
fatal: 无法重置子模组索引
weilian@weilian-OptiPlex-7080:~/esp-idf$ sudo git checkout --recurse-submodules f3159
[sudo] weilian 的密码： 
fatal: 不是 git 仓库：examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib/../../../../../../.git/modules/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib

Lzw655, 2022-10-17:
> fatal: 无法访问 '[https://github.com/kmackay/micro-ecc.git/'：Could](https://github.com/kmackay/micro-ecc.git/'%EF%BC%9ACould) not resolve host: github.com
> fatal: 无法克隆 'https://github.com/kmackay/micro-ecc.git' 到子模组路径 '/home/weilian/esp-

好几个子模块 clone 失败了，再用 `git submodule update -f` 试下

weilian1977, 2022-10-18:
已经成功！非常感谢！

weilian1977, 2022-10-18:
fatal: 不是 git 仓库：examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib/../../../../../../.git/modules/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib

这个错误，git init 然后 git  submodules  update就不报错了！

Lzw655, 2022-10-18:
好的，学到了

weilian1977, 2022-10-18:
> 好几个子模块 clone 失败了，再用 `git submodule update -f` 试下

这个命令估计也管用，但是没有尝试

weilian1977, 2022-10-18:
ubuntu下 master版本可以直接通过ESP32S3 的USB口烧录程序么？

Lzw655, 2022-10-18:
可以

weilian1977, 2022-10-18:
具体怎么操作？通过插拔开发板对比设备变化，感觉开发板的虚拟串口应该是ttyACM0
weilian@weilian-OptiPlex-7080:~/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build
Executing "ninja all"...
[1/4] cd /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/esp-idf/esptool_py && /home/weilian/.espressif/python_env/idf5.1_py3.10_env/bin/python /home/weilian/esp-idf/components/pa...e app /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/partition_table/partition-table.bin /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/lvgl_demos.binlvgl_demos.bin binary size 0xf2290 bytes. Smallest app partition is 0x400000 bytes. 0x30dd70 bytes (76%) free.
[1/1] cd /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader/esp-idf/esptool_py && /home/weilian/.espressif/python_env/idf5.1_py3.10_env/bin/python /home/weilian/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader/bootloader.binBootloader binary size 0x5460 bytes. 0x2ba0 bytes (34%) free.

Project build complete. To flash, run this command:
/home/weilian/.espressif/python_env/idf5.1_py3.10_env/bin/python ../../../../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32s3  write_flash --flash_mode dout --flash_size 8MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/lvgl_demos.bin
or run 'idf.py -p (PORT) flash'
weilian@weilian-OptiPlex-7080:~/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos$ idf.py -p ttyACM0 -b 460800
weilian@weilian-OptiPlex-7080:~/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos$ sudo idf.py -p ttyACM0 -b 460800
[sudo] weilian 的密码： 
sudo: idf.py：找不到命令
weilian@weilian-OptiPlex-7080:~/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos$ idf.py -p ttyACM0 -b 460800
weilian@weilian-OptiPlex-7080:~/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos$ idf.py -p /dev/ttyACM0 -b 460800
weilian@weilian-OptiPlex-7080:~/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos$ /home/weilian/.espressif/python_env/idf5.1_py3.10_env/bin/python ../../../../esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyACM0 -b 460800 --before default_reset --after hard_reset --chip esp32s3  write_flash --flash_mode dout --flash_size 8MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/lvgl_demos.bin
esptool.py v4.3
Serial port /dev/ttyACM0

A fatal error occurred: Could not open /dev/ttyACM0, the port doesn't exist

应该怎么查看开发板的USB虚拟端口？

Lzw655, 2022-10-18:
USB 虚拟端口固定是 /dev/ttyACM0，用 linux 命令查看  `ls /dev/ttyACM0`

weilian1977, 2022-10-19:
使用命令ls /dev/ttyA*，查看插拔开发板前后的设备变化，可以确认是ttyACM0，但是报错信息是
A fatal error occurred: Could not open /dev/ttyACM0, the port doesn't exist

weilian1977, 2022-10-19:
1、增加权限或进入 Boot 模式之后烧录程序都不行，报错依然是A fatal error occurred: Could not open /dev/ttyACM0, the port doesn't exist；
2、进入python 
     import serial
     da=serial.Serial("/dev/ttyACM00",9600)
     也报这个错误；
3、这个错误是the port doesn't exist，查询了一下一般报错是Permission denied；
4、PuTTY设置好串口参数后点击Open要么提示音无法打开要么直接界面自动关闭；
4、menuconfig里面需要配置什么？看到C3芯片的好像要配置？

weilian1977, 2022-10-19:
sudo chmod a+rw /dev/ttyACM0 之后不在报错，看来还是权限问题

Lzw655, 2022-10-19:
好的

weilian1977, 2022-10-19:
目前已经可以编译烧录，还有几个问题
1、esp32-s3-lcd-ev-board/examples 里面的工程需要 IDF 的版本在 release/v5.0 及以上，并且需要打上 PSRAM 120M 补丁，我的理解主要是实现了octal PSRAM 的DDR 120Mhz，对于2M的Quad PSRAM 的SDR 120Mhz，是不是也必须 release/v5.0 及以上+PSRAM 120M 补丁？V4.4.2是否可以？
2、esp32-s3-lcd-ev-board/examples/lvgl_demos能否在menuconfig里面配置octal PSRAM为DDR或SDR、时钟频率120Mhz或80MHz？主要是为了测试不同工作参数下屏幕刷新速度的区别。

Lzw655, 2022-10-19:
1. 2M 的 Quad PSRAM 应该能够直接在 menuconfig 里设置到 120M，那个 120M 补丁只用于使得 Octal 能够设置到 120M （在 menuconfig 里能够选上）。工程里的补丁只能用在 v5.0，v4.4.2 的 RGB LCD 驱动没有 v5.0 完善，推荐使用 v5.0 开发 RGB LCD 应用。
2.  PSRAM 设置为 80M 默认对应 SDR，120M 默认对应 DDR。lvgl_demos 下的 readme 有在 120M 和 80M 下的帧率测试结果。

weilian1977, 2022-10-20:
1、master的版本是V5.1-dev，是否需要退回到V5.0？项目还需要Camera
2、8线 PSRAM能否软件设置为4线PSRAM SDR模式下工作？测试结果能否代表2M Quad PSRAM SDR？
3、lvgl_demos下的帧率测试Mode1、Mode2、Mode3 、 Mode4 参数在menuconfig里面可以设置么？

weilian1977, 2022-10-20:
2、8线 PSRAM能否软件设置为4线PSRAM SDR模式下工作？在menuconfig里设置为Quad PSRAM直接黑屏

Lzw655, 2022-10-20:
1. RGB LCD 方面不需要，Camera 需要去 esp-iot-solution 下咨询
2. 可以，能
3. 这几个 mode 对应 LVGL 的配置，Mode 3、4 可以在 menucofnig 下的 `Avoid tearing effect` 配置，Mode 1、2 通过修改  esp32-s3-lcd-ev-board/examples/lvgl_demos/main/lv_port.c 中 `LV_PORT_BUFFER_HEIGHT` 和 `LV_PORT_BUFFER_MALLOC` 注释代码配置

Lzw655, 2022-10-20:
> 2、8线 PSRAM能否软件设置为4线PSRAM SDR模式下工作？在menuconfig里设置为Quad PSRAM直接黑屏

请详细阐述报错信息

Lzw655, 2022-10-20:
@weilian1977 您现在的问题范围比较广，不适合在 issue 下进行回答，麻烦联系官方的 TS 团队支持，链接 https://www.espressif.com/en/contact-us/technical-inquiries

weilian1977, 2022-10-20:
> 

[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/sdkconfig
Loading defaults file /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/sdkconfig.defaults...
-- App "lvgl_demos" version: 9a9cb89-dirty
-- Adding linker script /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/weilian/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /home/weilian/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /home/weilian/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /home/weilian/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /home/weilian/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /home/weilian/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /home/weilian/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bsp bt bus cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser i2c_devices idf_test ieee802154 iot_button json log lvgl lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocol_examples_common protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tinyusb touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/weilian/esp-idf/components/app_trace /home/weilian/esp-idf/components/app_update /home/weilian/esp-idf/components/bootloader /home/weilian/esp-idf/components/bootloader_support /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/components/bsp /home/weilian/esp-idf/components/bt /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/components/bus /home/weilian/esp-idf/components/cmock /home/weilian/esp-idf/components/console /home/weilian/esp-idf/components/cxx /home/weilian/esp-idf/components/driver /home/weilian/esp-idf/components/efuse /home/weilian/esp-idf/components/esp-tls /home/weilian/esp-idf/components/esp_adc /home/weilian/esp-idf/components/esp_app_format /home/weilian/esp-idf/components/esp_common /home/weilian/esp-idf/components/esp_eth /home/weilian/esp-idf/components/esp_event /home/weilian/esp-idf/components/esp_gdbstub /home/weilian/esp-idf/components/esp_hid /home/weilian/esp-idf/components/esp_http_client /home/weilian/esp-idf/components/esp_http_server /home/weilian/esp-idf/components/esp_https_ota /home/weilian/esp-idf/components/esp_https_server /home/weilian/esp-idf/components/esp_hw_support /home/weilian/esp-idf/components/esp_lcd /home/weilian/esp-idf/components/esp_local_ctrl /home/weilian/esp-idf/components/esp_netif /home/weilian/esp-idf/components/esp_phy /home/weilian/esp-idf/components/esp_pm /home/weilian/esp-idf/components/esp_psram /home/weilian/esp-idf/components/esp_ringbuf /home/weilian/esp-idf/components/esp_rom /home/weilian/esp-idf/components/esp_system /home/weilian/esp-idf/components/esp_timer /home/weilian/esp-idf/components/esp_wifi /home/weilian/esp-idf/components/espcoredump /home/weilian/esp-idf/components/esptool_py /home/weilian/esp-idf/components/fatfs /home/weilian/esp-idf/components/freertos /home/weilian/esp-idf/components/hal /home/weilian/esp-idf/components/heap /home/weilian/esp-idf/components/http_parser /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/components/i2c_devices /home/weilian/esp-idf/components/idf_test /home/weilian/esp-idf/components/ieee802154 /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/components/iot_button /home/weilian/esp-idf/components/json /home/weilian/esp-idf/components/log /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/components/lvgl /home/weilian/esp-idf/components/lwip /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/main /home/weilian/esp-idf/components/mbedtls /home/weilian/esp-idf/components/mqtt /home/weilian/esp-idf/components/newlib /home/weilian/esp-idf/components/nvs_flash /home/weilian/esp-idf/components/openthread /home/weilian/esp-idf/components/partition_table /home/weilian/esp-idf/components/perfmon /home/weilian/esp-idf/components/protobuf-c /home/weilian/esp-idf/examples/common_components/protocol_examples_common /home/weilian/esp-idf/components/protocomm /home/weilian/esp-idf/components/pthread /home/weilian/esp-idf/components/sdmmc /home/weilian/esp-idf/components/soc /home/weilian/esp-idf/components/spi_flash /home/weilian/esp-idf/components/spiffs /home/weilian/esp-idf/components/tcp_transport /home/weilian/esp-idf/components/tinyusb /home/weilian/esp-idf/components/touch_element /home/weilian/esp-idf/components/ulp /home/weilian/esp-idf/components/unity /home/weilian/esp-idf/components/usb /home/weilian/esp-idf/components/vfs /home/weilian/esp-idf/components/wear_levelling /home/weilian/esp-idf/components/wifi_provisioning /home/weilian/esp-idf/components/wpa_supplicant /home/weilian/esp-idf/components/xtensa
-- Configuring done
-- Generating done
[... cut ...]
-- Build files have been written to: /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader
[1/1] cd /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader/esp-idf/esptool_py && /home/weilian/.espressif/python_env/idf5.1_py3.10_env/bin/python /home/weilian/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/weilian/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/bootloader/bootloader.bin
Bootloader binary size 0x5440 bytes. 0x2bc0 bytes (34%) free.
ninja: build stopped: subcommand failed.

weilian1977, 2022-10-20:
> @weilian1977 您现在的问题范围比较广，不适合在 issue 下进行回答，麻烦联系官方的 TS 团队支持，链接 https://www.espressif.com/en/contact-us/technical-inquiries

OK

Lzw655, 2022-10-24:
@weilian1977 麻烦关闭一下这个 issue 哈

closed 2022-10-27 by weilian1977