espressif/esp-dev-kits #61: Trouble installing USB Driver for esp32-s2-hmi-devkit-1

formus14, 2022-11-23:
Hello,

Im getting started with the esp32-s2-hmi-devkit-1, but unfortuntally the driver is neither installed by default neither accepting ftdi installation for it.

- I want to ask what dirver shall i installer in order to flash the dev kit

[image]

Thanks in advance

Lzw655, 2022-11-24:
Hi, you can check [this guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/establish-serial-connection.html#connect-esp32-s2-to-pc) to install the driver.

formus14, 2022-11-24:
Thanks @Lzw655 , CP210x: [CP210x USB to UART Bridge VCP Drivers](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers) driver was what i missed.
Works fine now!

closed 2022-11-24 by formus14