espressif/esp-dev-kits #65: lvgl  dropdown  组件无法 选中

Wvirgil123, 2023-01-29:
问题：使用lvgl 的dropdown 组件时， 无法选中 选项
https://user-images.githubusercontent.com/25890961/215316979-e36c5a54-5675-45c5-ba97-666d01418687.mp4

Lzw655, 2023-01-29:
这个 Demo 在我们的开发板上测试是正常的，lvgl 版本为 8.3.0，你能切换到这个版本试一下吗

Wvirgil123, 2023-01-31:
> 这个 Demo 在我们的开发板上测试是正常的，lvgl 版本为 8.3.0，你能切换到这个版本试一下吗

我这边切换到8.3.0 这个版本也是这样的。 不只是这个demo ， 连最简单显示一个dropdown 组件，也是无法选中的呢

Lzw655, 2023-01-31:
我这里是用的是包管理器里 8.3.0 的 [LVGL](https://components.espressif.com/components/lvgl/lvgl)，现在无法复现这个问题，你能用这个试一下吗

Wvirgil123, 2023-01-31:
> 我这里是用的是包管理器里 8.3.0 的 [LVGL](https://components.espressif.com/components/lvgl/lvgl)，现在无法复现这个问题，你能用这个试一下吗

我下载添加到程序中， 测试也是无法选中呢
[image]
****

Wvirgil123, 2023-01-31:
是否可能这里有特殊的配置呢
[image]

Lzw655, 2023-01-31:
那你将前两个选项取消，使用 LVGL 默认配置看下呢，参考[配置](https://github.com/espressif/esp-dev-kits/blob/master/esp32-s3-lcd-ev-board/examples/lvgl_demos/sdkconfig.defaults)

Wvirgil123, 2023-01-31:
> 

一样不行呢。 选择其他的会触发值变化事件，但是打印出来还是最后一个值。

[image]

Lzw655, 2023-01-31:
麻烦你把当前的工程打包（删掉 build），发我测试一下吧。

Wvirgil123, 2023-02-02:
> 麻烦你把当前的工程打包（删掉 build），发我测试一下吧。

不是使用贵司的板子呢， 您这边应该无法测试

Lzw655, 2023-02-02:
没关系，我这边只用你工程里的应用层程序和配置，不然无法复现问题来解决。

Wvirgil123, 2023-02-02:
> 没关系，我这边只用你工程里的应用层程序和配置，不然无法复现问题来解决。

谢谢， 工程在这个链接， 运行 esp32_fw/examples/lvgl_demos 这个demo

https://github.com/Wvirgil123/file/blob/main/esp32_fw.7z

Lzw655, 2023-02-02:
> > 没关系，我这边只用你工程里的应用层程序和配置，不然无法复现问题来解决。
> 
> 谢谢， 工程在这个链接， 运行 esp32_fw/examples/lvgl_demos 这个demo
> 
> https://github.com/Wvirgil123/file/blob/main/esp32_fw.7z

不客气，我尽快验证一下

Wvirgil123, 2023-02-03:
> > > 没关系，我这边只用你工程里的应用层程序和配置，不然无法复现问题来解决。
> > 
> > 
> > 谢谢， 工程在这个链接， 运行 esp32_fw/examples/lvgl_demos 这个demo
> > https://github.com/Wvirgil123/file/blob/main/esp32_fw.7z
> 
> 不客气，我尽快验证一下

您好，您这边情况是一样的吗

Lzw655, 2023-02-03:
> > > > 没关系，我这边只用你工程里的应用层程序和配置，不然无法复现问题来解决。
> > > 
> > > 
> > > 谢谢， 工程在这个链接， 运行 esp32_fw/examples/lvgl_demos 这个demo
> > > https://github.com/Wvirgil123/file/blob/main/esp32_fw.7z
> > 
> > 
> > 不客气，我尽快验证一下
> 
> 您好，您这边情况是一样的吗

抱歉，有点忙住了，我下午会试一下

Lzw655, 2023-02-03:
@Wvirgil123 实在抱歉，即使用你工程里的 LVGL 以及配置文件，我这里仍然无法复现这个问题，有没有尝试换块屏幕？

Wvirgil123, 2023-02-08:
> @Wvirgil123 实在抱歉，即使用你工程里的 LVGL 以及配置文件，我这里仍然无法复现这个问题，有没有尝试换块屏幕？

我这边大概找到了问题原因， 我使用的是 FT6336U 触摸IC , 在没有按压的情况， 读取的坐标是一个非常大的值， 将这些坐标输入到 lvgl 处理，就导致了一些异常。解决方法是： 修改lv_port.c 文件， 在没有按压的情况， 坐标位置保持上次按压的坐标。

[image]

Wvirgil123, 2023-02-08:
https://github.com/espressif/esp-dev-kits/blob/d338b3ac7a1722bdba4a6b64b60d8f447af284ce/esp32-s3-lcd-ev-board/examples/smart-panel/components/i2c_devices/touch_panel/ft5x06.c#L14

当没有按压时， x和y 并没有赋值， x和y 也没有初始化， 那给上层应用应该是一个随机值。这个处理应该是有问题的

Lzw655, 2023-02-08:
> https://github.com/espressif/esp-dev-kits/blob/d338b3ac7a1722bdba4a6b64b60d8f447af284ce/esp32-s3-lcd-ev-board/examples/smart-panel/components/i2c_devices/touch_panel/ft5x06.c#L142
> 
> 当没有按压时， x和y 并没有赋值， x和y 也没有初始化， 那给上层应用应该是一个随机值。这个处理应该是有问题的

是的，这个例程里面是一个旧版本的驱动，现在已经不再维护了，**ft5x06** 请使用[包管理器驱动](https://components.espressif.com/components/espressif/esp_lcd_touch_ft5x06)，使用参考[例程](https://github.com/espressif/esp-bsp/blob/805faecbfa61104c30d221e30bef12c98b3e86f2/esp32_s3_lcd_ev_board/sub_board/sub_board_480x480.c#L158-L183)。

Lzw655, 2023-02-21:
@Wvirgil123 你好，如果问题已解决，请关闭此 issue，谢谢

closed 2023-05-12 by Lzw655