espressif/esp-dev-kits #67: ESP32-S3-LCD-EV-BOARDD例程缺少文件？

1201yuge, 2023-04-02:
如题，打开usb_camera_lcd的main.c文件，里面的宏BSP_LCD_H_RES找不到定义的地方。
[image]

找了一下资料，好像是缺少了相关的bsp文件，但是我已经update了所有的subModule, 但是还是esp32-s3-lcd-ev-board文件夹下也没有看到类似components这样子的文件夹。
[image]

Lzw655, 2023-04-03:
@1201yuge 你好，这个 `BSP_LCD_H_RES` 宏所在文件位于[ BSP 组件](https://github.com/espressif/esp-bsp/blob/44c0707b854504643e229a9a9c5225c49979488e/esp32_s3_lcd_ev_board/include/bsp/esp32_s3_lcd_ev_board.h#L247)中，这个组件是通过包管理器工具自动拉取的，可以在[ yml 文件](https://github.com/espressif/esp-dev-kits/blob/d87d437a66e54c8b94a4bf71c96f0d436b9a6501/esp32-s3-lcd-ev-board/examples/usb_camera_lcd/main/idf_component.yml#L3)中看到依赖关系。

执行 `idf.py reconfigure` 或 `idf.py build` 会自动拉取该组件，可以在工程目录下看到自动生成了 `managed_components/espressif__esp32_s3_lcd_ev_board`。

1201yuge, 2023-04-03:
@Lzw655 请问这个包管理工具 指的是什么，具体怎么操作？我直接使用git clone这个项目后，再使用git submodule update成功后，也未看到这个 BSP组件。

closed 2023-04-03 by 1201yuge

Lzw655, 2023-04-03:
@1201yuge 关于包管理器的详细资料，请参考[文档](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/tools/idf-component-manager.html)。