espressif/esp-dev-kits #68: Typo in documentations for USB Host power options?

shreeve, 2023-04-06:
The documentation at:

     https://espressif-docs.readthedocs-hosted.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-usb-otg/user_guide.html

has the following:

[image]

Isn't this incorrect?

In order for me to use the `USB Host` (USB-A female port), I need to set:

[image]

Which is the opposite of what the documentation states.

Am I missing something here?

shreeve, 2023-04-06:
At a minimum, it seems like the text should be clearer.

leeebo, 2023-04-07:
Hi @shreeve, the description of the document is correct, although it is a little convoluted.

The symbols in schematic `USB_DEV D+ D-` means the Pins will link to an external USB device (but through Type-A female / Host port)

BTW. you can use the already encapsulated API [bsp esp32_s3_usb_otg](https://components.espressif.com/components/espressif/esp32_s3_usb_otg) to switch ports

shreeve, 2023-04-07:
I never even knew about the BSP mechanism... pretty cool

shreeve, 2023-04-07:
There is a nice set of API calls in the source code within the BSP code.

But, where do I find the API documentation for this BSP code?

Lzw655, 2023-04-07:
> 

There is no API documentation for the BSP code now. You can directly find API information from the [header file](https://github.com/espressif/esp-bsp/blob/master/esp32_s3_usb_otg/include/bsp/esp32_s3_usb_otg.h).

shreeve, 2023-04-07:
@Lzw655 - Ah, ok. Thank you!

closed 2023-04-07 by shreeve