espressif/esp-dev-kits #69: esp32-s3-usb-otg devkit, Use green and yellow LEDs not working

HeadHodge, 2023-04-11:
Hi,

I'm trying to use the onboard yellow and green LED lights.

Im using:

    // Turn On Green LED
    pinMode(GPIO_NUM_15, OUTPUT);
    digitalWrite(GPIO_NUM_15, HIGH);

    // Turn On Yellow LED
    pinMode(GPIO_NUM_16, OUTPUT);
    digitalWrite(GPIO_NUM_16, HIGH);

Neither LED, turns on. I'm using the GPIO_NUM stated in the esp32-s3-usb-otg devkit docs.

Please help
Thanks

fairytail655, 2023-04-12:
@HeadHodge Hi, are you developing with Arduino?

HeadHodge, 2023-04-12:
> Hi, are you developing with Arduino?

Yes (btw... the menu button works fine)

`void setup() {
    Serial.begin(115200);
    delay(4000);
    
    SysTools::addLog("%s", "lcdDisplayTest::Setup");

    // initialize the LED pin as an output:
    pinMode(GPIO_NUM_16, OUTPUT);
    digitalWrite(GPIO_NUM_16, HIGH);
    
    //gpio_set_direction(GPIO_NUM_10, GPIO_MODE_OUTPUT);
    //gpio_set_level(GPIO_NUM_10, 1);

    pinMode(14, INPUT_PULLUP);
}

void loop() {
  // read the state of the pushbutton value:
  SysTools::addLog("buttonState: %u", digitalRead(14));
  
  // check if the pushbutton is pressed. If it is, the buttonState is HIGH:
  if (buttonState == HIGH) {
    SysTools::addLog("%s", "button high");
    // turn LED on:
    //digitalWrite(ledPin, HIGH);
  } else {
    //SysTools::addLog("%s", "button low");
    // turn LED off:
    //digitalWrite(ledPin, LOW);
  }

  //delay(2000);
}`

fairytail655, 2023-04-12:
> Yes (btw... the menu button works fine)
>
> `void setup() { Serial.begin(115200); delay(4000);
>
> ```
> SysTools::addLog("%s", "lcdDisplayTest::Setup");
>
> // initialize the LED pin as an output: [...]

Ok, I will test it as soon as possible.

fairytail655, 2023-04-12:
@HeadHodge Well, I use your code and all leads turn on. Do you have a voltage gauge to measure the voltage? Maybe they are broken...

HeadHodge, 2023-04-12:
> Maybe they are broken...

Thanks for your time.

I forgot i had another new unused board. When I plug uart/usb cable into my laptop both leds start flashing and the espressif logo displays on the lcd!! The other board never did that. 😏 So seems like bad board.

Before I toss the board, do you know how i can program it to factory default?

Thanks

HeadHodge, 2023-04-13:
@fairytail655 > _**Well, I use your code and all leads turn on.**_
I did the same on the new board and it also works!!

But to do that, I had to flash my program over the factory default.

I would like to flash the factory demo bin back to the board. I think I found the bin file, but not sure what address to flash it to and not sure if there are co-binaries needed  to be flashed. Any insight would be very appreciated.

**factory_bin:**

https://github.com/espressif/esp-dev-kits/tree/d87d437a66e54c8b94a4bf71c96f0d436b9a6501/esp32-s3-usb-otg/examples/factory/factory_bin

fairytail655, 2023-04-13:
@HeadHodge Sure, you can flash the bin to address 0x00.

HeadHodge, 2023-04-13:
> Sure, you can flash the bin to address 0x00.

It works... THANKS!!!

Everything is good now!
Thanks Again

fairytail655, 2023-04-13:
Yeah, you're welcome. If there're no more problems, please close this issue. Thanks.

closed 2023-04-13 by HeadHodge