espressif/esp-dev-kits #70: 我已经成功编译并且下载程序，但是程序运行出现错误。esp32s3 n16r8

lbylan, 2023-04-23:
[0;32mI (1415) esp_psram: SPI SRAM memory test OK[0m
[0;32mI (1424) cpu_start: Pro cpu start user code[0m
[0;32mI (1424) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (1424) cpu_start: Application information:[0m
[0;32mI (1427) cpu_start: Project name:     86box_demo[0m
[0;32mI (1432) cpu_start: App version:      1[0m
[0;32mI (1437) cpu_start: Compile time:     Apr 23 2023 16:10:19[0m
[0;32mI (1443) cpu_start: ELF file SHA256:  4816ccf9259f2a80...[0m
[0;32mI (1449) cpu_start: ESP-IDF:          v5.0.1-dirty[0m
[0;32mI (1455) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (1459) cpu_start: Max chip rev:     v0.99 [0m
[0;32mI (1464) cpu_start: Chip rev:         v0.1[0m
[0;32mI (1469) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1477) heap_init: At 3FC9CFD8 len 0004C738 (305 KiB): D/IRAM[0m
[0;32mI (1483) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM[0m
[0;32mI (1490) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (1496) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM[0m
[0;32mI (1502) esp_psram: Adding pool of 2432K of PSRAM memory to heap allocator[0m
[0;32mI (1510) spi_flash: detected chip: gd[0m
[0;32mI (1514) spi_flash: flash io: qio[0m
[0;32mI (1519) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (1534) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;31mE (1543) tca9554: write_direction_reg(129): Write direction reg failed[0m
[0;31mE (1550) tca9554: reset(146): Write dir reg failed[0m
[0;31mE (1556) tca9554: esp_io_expander_new_i2c_tca9554(79): Reset failed[0m
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x420173d1
file: "./managed_components/espressif__esp32_s3_lcd_ev_board/esp32_s3_lcd_ev_board.c" line 72
func: bsp_io_expander_init
expression: esp_io_expander_new_i2c_tca9554((1), ((0x20)), &io_expander)

abort() was called at PC 0x40380607 on core 0

Backtrace: 0x403762be:0x3fcf35a0 0x40380611:0x3fcf35c0 0x40385ffa:0x3fcf35e0 0x40380607:0x3fcf3650 0x420173d1:0x3fcf3680 0x42017525:0x3fcf36a0 0x42008a76:0x3fcf36f0 0x4205bae1:0x3fcf3730

lbylan, 2023-04-23:
已解决

Lzw655, 2023-05-08:
@lbylan 你好，如果没有问题的话请关闭此 issue，谢谢！

closed 2023-05-12 by Lzw655