espressif/esp-dev-kits #71: Unable to get VBUS (+5V) on the ESP USB (USB Host) port on the ESP32-S3 DevKit C v1.0 board

shreeve, 2023-04-27:
On my ESP32-S3-USB-OTG board, I need to set GPIO 12, 13, 17, 18 in order to enable the USB Host port. Otherwise, that port will not be able to supply power the USB Devices that connect to it.

On my ESP32-S3 DevKitC v1.0 board (see diagram below), I don't see anything that would imply that I have to set those same pins. In fact, it looks like both micro USB ports are connected to +5V and GND and that they both have their own D+ and D- pins, which is all that I need.

Unfortunately, I can't see the +5V on the ESP USB port. Instead, I have to get share +5V/GND pins from an external USB A host port.

Is there a way for the board to supply the needed +5V (I assume the GND is already working)?

[image]

leeebo, 2023-05-08:
Hi @shreeve, For the `esp32-s3-devkitc` board the micro USB port can only act as 5V input, and can not as 5V output. Because there's a diode between the 5V and the VBUS (You can short the diode to get 5V output, but please note the power risks)

shreeve, 2023-05-12:
@leeebo - Thank you for the information!!!

closed 2023-05-12 by shreeve