espressif/esp-dev-kits #72: Where can I find the factory reset firmware for a ESP32-S3-USB-OTG board?

shreeve, 2023-04-27:
I accidentally blasted the firmware on my ESP32-S3-USB-OTG board, which had a very cool display on the little built in screen.

I've tried to download what looks like a factory reset `*.bin` file, but it only repeatedly shows a message about having an "Invalid Header", or something like that.

I also tried to rebuild the firmware, but it uses ESP-IDF v4.4 and many of the dependencies no longer are able to compile with the tools I have on my machine.

In summary, I'd like to "reset" my ESP32-S3-USB-OTG board to the factory state, but don't know how.

Any tips?

Lzw655, 2023-05-08:
Sorry for being late. Did you try to download this [bin](https://github.com/espressif/esp-dev-kits/blob/master/esp32-s3-usb-otg/examples/factory/factory_bin/usb_otg_fac_1_0_2.bin)? Or you can directly download the factory in [here](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://raw.githubusercontent.com/espressif/esp-dev-kits/master/launch.toml) online.

leeebo, 2023-05-08:
@shreeve We suggest using[ Launchpad](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://raw.githubusercontent.com/espressif/esp-dev-kits/master/launch.toml) to flash the factory bin directly. If you download [bin](https://github.com/espressif/esp-dev-kits/blob/master/esp32-s3-usb-otg/examples/factory/factory_bin/usb_otg_fac_1_0_2.bin) from the repo, please flash to address `0x00`

[image]

shreeve, 2023-05-09:
@leeebo - **THANK YOU!!!** This finally worked. I wasn't able to do it form the Launchpad, but I was able to run this command from `macOS`:

```
esptool.py --chip esp32s3 --baud 115200 --port /dev/cu.usbserial-10 write_flash -z 0 ~/Desktop/usb_otg_fac_1_0_2.bin
```

Here is what it looked like:

```
$ esptool.py --chip esp32s3 --baud 115200 --port /dev/cu.usbserial-10 write_flash -z 0 ~/Desktop/usb_otg_fac_1_0_2.bin

esptool.py v4.5.1
Serial port /dev/cu.usbserial-10
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:87:cc:4c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00354fff...
Compressed 3493888 bytes to 1042883...
Wrote 3493888 bytes (1042883 compressed) at 0x00000000 in 93.1 seconds (effective 300.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

shreeve, 2023-05-09:
Are there any plans to update the code so that it can be compiled with ESP-IDF v5.1?

The `.bin` file works great, but the source for the `factory` example can't be compiled with v5.1+ because of several broken dependencies.

Also, where is it that people go to learn how to use the component libraries? Is there some website somewhere where there is documentation or how to use the code? It seems to me like going from zero/nothing to a working app is a huge amount of work. Am I missing something?

Lzw655, 2023-05-10:
@shreeve Sorry, there is no documentation for the component libraries in esp32-s3-usb-otg. But there are rich [docs](https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32s3/index.html) for ESP-IDF SDK. And you can find more official component libraries [here](https://components.espressif.com/).

shreeve, 2023-05-12:
@Lzw655 - Thank you, very helpful!

closed 2023-05-12 by shreeve