espressif/esp-dev-kits #73: 我不知道ESP32 S3的驱动在哪里？而且支不支持ST7701S驱动？

ZTFXS, 2023-05-03:


Lzw655, 2023-05-08:
ESP32-S3 LCD 驱动介绍见[文档](https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32s3/api-reference/peripherals/lcd.html)，ST7701S RGB 接口驱动例程见附件，需注意该例程是基于 ESP32-S3R8 并开启了 PSRAM Octal 120M 特性，详情见 [README](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-lcd-ev-board/factory#idf-patch)。

[ST7701S_RGB_2_1_480_480.zip](https://github.com/espressif/esp-dev-kits/files/11416335/ST7701S_RGB_2_1_480_480.zip)

Lzw655, 2023-05-09:
@ZTFXS 你好，如果问题已解决的话，请关闭此 issue，谢谢！

ZTFXS, 2023-05-09:
@Lzw655 我驱动ST7701S使用3线SPI 9bit+RGB驱动液晶屏，不是I2C，没有用扩展器

Lzw655, 2023-05-09:
> @Lzw655 我驱动ST7701S使用3线SPI 9bit+RGB驱动液晶屏，不是I2C，没有用扩展器

你可以按照这个示例将 IO 扩展芯片的操作换成普通 IO 操作，我们推荐使用模拟 SPI 实现。

closed 2023-05-10 by ZTFXS