espressif/esp-dev-kits #74: ES7210 ADC full data sheet

rpatel3001, 2023-05-19:
Are you able to share a full datasheet and user guide for the ES7210 as you did for the ES8311 [here](https://github.com/espressif/esp-dev-kits/issues/24#issuecomment-781314125)?

ESP-Mars, 2023-05-23:
- You can find the latest ES7210 datasheet here http://www.everest-semi.com/pdf/ES7210%20PB.pdf

- For the ES7120 user guide. We have been informed by Everest-Sami, the manufacturer of ES7210, that the user guide is currently being revised and will be made publicly available once the updates are complete. Subsequently, we will update the guide and provide it to you.

ESP-Mars, 2023-05-24:
> Thank you, I look forward to the user guide. Is there a datasheet available that has the register map included?

The 'Configuration REGISTER DEFINITION' in the datasheet is an introduction to register.

rpatel3001, 2023-05-24:
Is there not a newer version of something like this: https://uploadcdn.oneyac.com/attachments/userfile/68/63/1629167664244_5904.pdf ?

Lzw655, 2023-05-26:
> Is there not a newer version of something like this: https://uploadcdn.oneyac.com/attachments/userfile/68/63/1629167664244_5904.pdf ?

Sorry, we don't have a newer version of this.

Lzw655, 2023-05-30:
@rpatel3001 Hi, if you don't have any questions, please close this issue. Thank you!

rpatel3001, 2023-05-30:
Please leave this issue open until the user guide is available

Lzw655, 2023-05-30:
> Please leave this issue open until the user guide is available

Ok.

M4n197, 2026-02-12:
Hello, it's now 2026 and I still can't find a datasheet for the ES7210 that includes the register description. Is there any news?