espressif/esp-dev-kits #75: 编译并上传lvgl_demos后，出现以下错误是什么原因

efffx, 2023-06-12:
E (843) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(119): i2c transaction failed
E (843) gt1151: read_product_id(207): I2C read failed
E (845) gt1151: esp_lcd_touch_new_i2c_gt1151(89): Read product id failed
E (853) gt1151: Initialization failed!

Lzw655, 2023-06-13:
@efffx 看 log 信息，这个 demo 是配置为 800x480 sub3 子板的，但是并未检测到该子板的触摸芯片（GT1151），请问是否在 ESP32-S3-LCD-EV-Board 开发板上正确连接该子板？

efffx, 2023-06-13:
> @efffx 看 log 信息，这个 demo 是配置为 800x480 sub3 子板的，但是并未检测到该子板的触摸芯片（GT1151），请问是否在 ESP32-S3-LCD-EV-Board 开发板上正确连接该子板？

我的子板是480*480的

Lzw655, 2023-06-13:
示例默认为 800*480 的配置，请参考示例的 [README ](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-lcd-ev-board/examples/lvgl_demos#configurations)修改配置

Lzw655, 2023-06-14:
@efffx 请问问题是否解决，如是麻烦关闭此 issue，谢谢。

efffx, 2023-06-14:
感谢解答

closed 2023-06-14 by efffx