espressif/esp-dev-kits #77: ESP32-S3-USB-OTG factory source code

TechRelief, 2023-07-16:
I see a bin file in the esp-dev-kits repo for the ESP32-S3-USB-OTG board, which presumably is a compiled image.  Where can I find the actual source code for this.
I am new to this board and it would be helpful to look at the code as a starting point on how to program this board.

Lzw655, 2023-07-17:
@TechRelief Hi, you can find the source code in folder *esp32-s3-usb-otg/examples/factory*.

Lzw655, 2023-07-27:
@TechRelief If your problem has been resolved, please close this issue. Thank you!

closed 2023-08-08 by Lzw655

TechRelief, 2023-08-19:
Did you mean: https://github.com/espressif/esp-dev-kits/blob/master/esp32-s3-usb-otg/examples/factory/main/app_main.c
If so you can close the issue.  Sorry about the dalay got Covid-19 again and am now recovered.

Lzw655, 2023-08-21:
@TechRelief Yeah. No worries. You can open a new issue if you have any other problems.