espressif/esp-dev-kits #78: 86box UI interface cannot voice wake-up control under factory tested firmware

maochonghao, 2023-07-20:
I saw at the official developer conference that the factory routines of the 86box development board can voice wake up and respond to the air conditioning command on the UI interface. The interface will jump to the air conditioning UI and open it, but my board's factory routines do not respond to voice after entering the UI, and can only use a pure voice interface. May I ask what the situation is?

espressif2022, 2023-07-21:
> I saw at the official developer conference that the factory routines of the 86box development board can voice wake up and respond to the air conditioning command on the UI interface. The interface [...]

We divide the sr function and interface into two demos;
If you want to use them together, you can `do sys_set->sr_enable = true`  in the app_main.

maochonghao, 2023-07-21:
> We divide the sr function and interface into two demos; If you want to use them together, you can `do sys_set->sr_enable = true` in the app_main.

Thank you very much. Additionally, during the testing process, I found that the 480 * 480 LCD development board I purchased officially was undergoing testing on 86_ Demo and lv_ During the demo process, IDF5.1 version was used, but it cannot run normally after compilation. The program will restart indefinitely. If these two demos run in IDF5.0? Compile directly 86_ Smart_ Pannel's demo can run normally.

espressif2022, 2023-07-21:
> Thank you very much. Additionally, during the testing process, I found that the 480 * 480 LCD development board I purchased officially was undergoing testing on 86_ Demo and lv_ During the demo [...]

can you provide some useful log ?

Lzw655, 2023-08-08:
Hi @maochonghao, if your problem has been resolved, please close this issue. Thank you!

closed 2023-08-21 by Lzw655