espressif/esp-dev-kits #79: Problems encountered compiling ST77903_QSPI_1_6_400_400 using idf ninja: error: 'D:/Espressif/frameworks/esp-idf-master/components/esp_wifi/lib/esp32s3/libcore.a', needed by 'st77903_qspi.elf',  missing and no known rule to make it

woshisunkejun, 2023-08-02:
d:\Espressif\frameworks\esp-idf-master\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
Running cmake in directory D:\Downloads\esp32\ST77903_QSPI_1_6_400_400\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=d:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 D:\Downloads\esp32\ST77903_QSPI_1_6_400_400"...
-- IDF_TARGET is not set, guessed 'esp32s3' from sdkconfig 'D:/Downloads/esp32/ST77903_QSPI_1_6_400_400/sdkconfig'
-- Found Git: D:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 2 dependencies:
[1/2] idf (5.2.0)
[2/2] lvgl/lvgl (8.3.8)
-- Project sdkconfig file D:/Downloads/esp32/ST77903_QSPI_1_6_400_400/sdkconfig
Loading defaults file D:/Downloads/esp32/ST77903_QSPI_1_6_400_400/sdkconfig.defaults...
Loading defaults file D:/Downloads/esp32/ST77903_QSPI_1_6_400_400/sdkconfig.defaults.esp32s3...
-- Compiler supported targets: xtensa-esp32s3-elf
Note: HPM is enabled for the flash, force the ROM bootloader into DOUT mode for stable boot on
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: d:/Espressif/python_env/idf5.2_py3.11_env/Scripts/python.exe (found version "3.11.2") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "st77903_qspi" version: 1
-- Adding linker script D:/Downloads/esp32/ST77903_QSPI_1_6_400_400/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-master/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script D:/Espressif/frameworks/esp-idf-master/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-master/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-master/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-master/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-master/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script D:/Espressif/frameworks/esp-idf-master/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_lcd_st77903_qspi esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lvgl__lvgl lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: D:/Espressif/frameworks/esp-idf-master/components/app_trace D:/Espressif/frameworks/esp-idf-master/components/app_update D:/Espressif/frameworks/esp-idf-master/components/bootloader D:/Espressif/frameworks/esp-idf-master/components/bootloader_support D:/Espressif/frameworks/esp-idf-master/components/bt D:/Espressif/frameworks/esp-idf-master/components/cmock D:/Espressif/frameworks/esp-idf-master/components/console D:/Espressif/frameworks/esp-idf-master/components/cxx D:/Espressif/frameworks/esp-idf-master/components/driver D:/Espressif/frameworks/esp-idf-master/components/efuse D:/Espressif/frameworks/esp-idf-master/components/esp-tls D:/Espressif/frameworks/esp-idf-master/components/esp_adc D:/Espressif/frameworks/esp-idf-master/components/esp_app_format D:/Espressif/frameworks/esp-idf-master/components/esp_bootloader_format D:/Espressif/frameworks/esp-idf-master/components/esp_coex D:/Espressif/frameworks/esp-idf-master/components/esp_common D:/Espressif/frameworks/esp-idf-master/components/esp_eth D:/Espressif/frameworks/esp-idf-master/components/esp_event D:/Espressif/frameworks/esp-idf-master/components/esp_gdbstub D:/Espressif/frameworks/esp-idf-master/components/esp_hid D:/Espressif/frameworks/esp-idf-master/components/esp_http_client D:/Espressif/frameworks/esp-idf-master/components/esp_http_server D:/Espressif/frameworks/esp-idf-master/components/esp_https_ota D:/Espressif/frameworks/esp-idf-master/components/esp_https_server D:/Espressif/frameworks/esp-idf-master/components/esp_hw_support D:/Espressif/frameworks/esp-idf-master/components/esp_lcd D:/Downloads/esp32/ST77903_QSPI_1_6_400_400/components/esp_lcd_st77903_qspi D:/Espressif/frameworks/esp-idf-master/components/esp_local_ctrl D:/Espressif/frameworks/esp-idf-master/components/esp_mm D:/Espressif/frameworks/esp-idf-master/components/esp_netif D:/Espressif/frameworks/esp-idf-master/components/esp_netif_stack D:/Espressif/frameworks/esp-idf-master/components/esp_partition D:/Espressif/frameworks/esp-idf-master/components/esp_phy D:/Espressif/frameworks/esp-idf-master/components/esp_pm D:/Espressif/frameworks/esp-idf-master/components/esp_psram D:/Espressif/frameworks/esp-idf-master/components/esp_ringbuf D:/Espressif/frameworks/esp-idf-master/components/esp_rom D:/Espressif/frameworks/esp-idf-master/components/esp_system D:/Espressif/frameworks/esp-idf-master/components/esp_timer D:/Espressif/frameworks/esp-idf-master/components/esp_wifi D:/Espressif/frameworks/esp-idf-master/components/espcoredump D:/Espressif/frameworks/esp-idf-master/components/esptool_py D:/Espressif/frameworks/esp-idf-master/components/fatfs D:/Espressif/frameworks/esp-idf-master/components/freertos D:/Espressif/frameworks/esp-idf-master/components/hal D:/Espressif/frameworks/esp-idf-master/components/heap D:/Espressif/frameworks/esp-idf-master/components/http_parser D:/Espressif/frameworks/esp-idf-master/components/idf_test D:/Espressif/frameworks/esp-idf-master/components/ieee802154 D:/Espressif/frameworks/esp-idf-master/components/json D:/Espressif/frameworks/esp-idf-master/components/log D:/Downloads/esp32/ST77903_QSPI_1_6_400_400/managed_components/lvgl__lvgl D:/Espressif/frameworks/esp-idf-master/components/lwip D:/Downloads/esp32/ST77903_QSPI_1_6_400_400/main D:/Espressif/frameworks/esp-idf-master/components/mbedtls D:/Espressif/frameworks/esp-idf-master/components/mqtt D:/Espressif/frameworks/esp-idf-master/components/newlib D:/Espressif/frameworks/esp-idf-master/components/nvs_flash D:/Espressif/frameworks/esp-idf-master/components/nvs_sec_provider D:/Espressif/frameworks/esp-idf-master/components/openthread D:/Espressif/frameworks/esp-idf-master/components/partition_table D:/Espressif/frameworks/esp-idf-master/components/perfmon D:/Espressif/frameworks/esp-idf-master/components/protobuf-c D:/Espressif/frameworks/esp-idf-master/components/protocomm D:/Espressif/frameworks/esp-idf-master/components/ [... cut ...]
-- Build files have been written to: D:/Downloads/esp32/ST77903_QSPI_1_6_400_400/build
Running ninja in directory D:\Downloads\esp32\ST77903_QSPI_1_6_400_400\build
Executing "ninja all"...
ninja: error: 'D:/Espressif/frameworks/esp-idf-master/components/esp_wifi/lib/esp32s3/libcore.a', needed by 'st77903_qspi.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the D:\Downloads\esp32\ST77903_QSPI_1_6_400_400\build\log\idf_py_stderr_output_47460 and D:\Downloads\esp32\ST77903_QSPI_1_6_400_400\build\log\idf_py_stdout_output_47460

Lzw655, 2023-08-02:
@woshisunkejun Please read the README in the project carefully to ensure that you have correctly configured ESP-IDF and successfully applied the patches.

woshisunkejun, 2023-08-03:
PS D:\Espressif\frameworks\esp-idf-master> git apply D:\Downloads\esp32\ST77903_QSPI_1_6_400_400\patch\spi_support_segment_mode_c133949da6.patch
error: patch failed: Kconfig:574
error: Kconfig: patch does not apply
error: patch failed: components/driver/Kconfig:122
error: components/driver/Kconfig: patch does not apply
error: components/driver/include/esp_private/spi_master_internal.h: already exists in working directory
error: patch failed: components/driver/spi/gpspi/spi_master.c:1
error: components/driver/spi/gpspi/spi_master.c: patch does not apply
error: patch failed: components/driver/spi/include/driver/spi_master.h:11
error: components/driver/spi/include/driver/spi_master.h: patch does not apply
error: patch failed: components/driver/test_apps/spi/master/main/CMakeLists.txt:6
error: components/driver/test_apps/spi/master/main/CMakeLists.txt: patch does not apply
error: components/driver/test_apps/spi/master/main/test_spi_master_sct.c: already exists in working directory
error: patch failed: components/hal/esp32c2/include/hal/spi_ll.h:227
error: components/hal/esp32c2/include/hal/spi_ll.h: patch does not apply
error: patch failed: components/hal/esp32c3/include/hal/spi_ll.h:227
error: components/hal/esp32c3/include/hal/spi_ll.h: patch does not apply
error: patch failed: components/hal/esp32c6/include/hal/spi_ll.h:231
error: components/hal/esp32c6/include/hal/spi_ll.h: patch does not apply
error: patch failed: components/hal/esp32h2/include/hal/spi_ll.h:233
error: components/hal/esp32h2/include/hal/spi_ll.h: patch does not apply
error: patch failed: components/hal/esp32s2/include/hal/spi_ll.h:18
error: components/hal/esp32s2/include/hal/spi_ll.h: patch does not apply
error: patch failed: components/hal/esp32s3/include/hal/spi_ll.h:229
error: components/hal/esp32s3/include/hal/spi_ll.h: patch does not apply
error: patch failed: components/hal/include/hal/spi_hal.h:32
error: components/hal/include/hal/spi_hal.h: patch does not apply
error: patch failed: components/hal/spi_hal_iram.c:203
error: components/hal/spi_hal_iram.c: patch does not apply
error: patch failed: components/soc/esp32c2/include/soc/Kconfig.soc_caps.in:423
error: components/soc/esp32c2/include/soc/Kconfig.soc_caps.in: patch does not apply
error: patch failed: components/soc/esp32c2/include/soc/soc_caps.h:215
error: components/soc/esp32c2/include/soc/soc_caps.h: patch does not apply
error: patch failed: components/soc/esp32c3/include/soc/Kconfig.soc_caps.in:639
error: components/soc/esp32c3/include/soc/Kconfig.soc_caps.in: patch does not apply
error: patch failed: components/soc/esp32c3/include/soc/soc_caps.h:298
error: components/soc/esp32c3/include/soc/soc_caps.h: patch does not apply
error: patch failed: components/soc/esp32c6/include/soc/Kconfig.soc_caps.in:867
error: components/soc/esp32c6/include/soc/Kconfig.soc_caps.in: patch does not apply
error: patch failed: components/soc/esp32c6/include/soc/soc_caps.h:368
error: components/soc/esp32c6/include/soc/soc_caps.h: patch does not apply
error: patch failed: components/soc/esp32h2/include/soc/Kconfig.soc_caps.in:839
error: components/soc/esp32h2/include/soc/Kconfig.soc_caps.in: patch does not apply
error: patch failed: components/soc/esp32h2/include/soc/soc_caps.h:360
error: components/soc/esp32h2/include/soc/soc_caps.h: patch does not apply
error: patch failed: components/soc/esp32s2/include/soc/Kconfig.soc_caps.in:651
error: components/soc/esp32s2/include/soc/Kconfig.soc_caps.in: patch does not apply
error: patch failed: components/soc/esp32s2/include/soc/soc_caps.h:284
error: components/soc/esp32s2/include/soc/soc_caps.h: patch does not apply
error: patch failed: components/soc/esp32s3/include/soc/Kconfig.soc_caps.in:767
error: components/soc/esp32s3/include/soc/Kconfig.soc_caps.in: patch does not apply
error: patch failed: components/soc/esp32s3/include/soc/soc_caps.h:308
error: components/soc/esp32s3/include/soc/soc_caps.h: patch does not apply
error: patch failed: components/soc/linux/include/soc/Kconfig.soc_caps.in:2
error: components/soc/linux/include/soc/Kconfig.soc_caps.in: patch does not apply
error: patch failed: components/soc/linux/include/soc/soc_caps.h:1
error: components/soc/linux/include/soc/soc_caps.h: patch does not apply
PS D:\Espressif\frameworks\esp-idf-master>

Lzw655, 2023-08-03:
Did you switch to the specific commit (c133949da6)?

woshisunkejun, 2023-08-03:
PS D:\Espressif\frameworks\esp-idf-master> git checkout --recurse-submodules c133949da6
HEAD is now at c133949da6 Merge branch 'bugfix/obss_scan_disasrm' into 'master'
M       Kconfig
M       components/driver/Kconfig
M       components/driver/spi/gpspi/spi_master.c
M       components/driver/spi/include/driver/spi_master.h
M       components/driver/test_apps/spi/master/main/CMakeLists.txt
M       components/hal/esp32c2/include/hal/spi_ll.h
M       components/hal/esp32c3/include/hal/spi_ll.h
M       components/hal/esp32c6/include/hal/spi_ll.h
M       components/hal/esp32h2/include/hal/spi_ll.h
M       components/hal/esp32s2/include/hal/spi_ll.h
M       components/hal/esp32s3/include/hal/spi_ll.h
M       components/hal/include/hal/spi_hal.h
M       components/hal/spi_hal_iram.c
M       components/soc/esp32c2/include/soc/Kconfig.soc_caps.in
M       components/soc/esp32c2/include/soc/soc_caps.h
M       components/soc/esp32c3/include/soc/Kconfig.soc_caps.in
M       components/soc/esp32c3/include/soc/soc_caps.h
M       components/soc/esp32c6/include/soc/Kconfig.soc_caps.in
M       components/soc/esp32c6/include/soc/soc_caps.h
M       components/soc/esp32h2/include/soc/Kconfig.soc_caps.in
M       components/soc/esp32h2/include/soc/soc_caps.h
M       components/soc/esp32s2/include/soc/Kconfig.soc_caps.in
M       components/soc/esp32s2/include/soc/soc_caps.h
M       components/soc/esp32s3/include/soc/Kconfig.soc_caps.in
M       components/soc/esp32s3/include/soc/soc_caps.h
M       components/soc/linux/include/soc/Kconfig.soc_caps.in
M       components/soc/linux/include/soc/soc_caps.h

woshisunkejun, 2023-08-03:
error: patch failed: components/hal/include/hal/spi_hal.h:32
error: components/hal/include/hal/spi_hal.h: patch does not apply

Lzw655, 2023-08-03:
[ST77903_QSPI_1_6_400_400.zip](https://github.com/espressif/esp-dev-kits/files/12249442/ST77903_QSPI_1_6_400_400.zip)

There are some bugs in the older driver. Here is a new project. For more details, please read the README in the project. (It depends on a different IDF version)

woshisunkejun, 2023-08-03:
Setting PYTHONNOUSERSITE, was not set
Using Python in d:\Espressif\python_env\idf5.2_py3.11_env\Scripts\
Python 3.11.2
Using Git in d:\Espressif\tools\idf-git\2.39.2\cmd\
git version 2.39.2.windows.1
Checking Python compatibility
Setting IDF_PATH: d:\Espressif\frameworks\esp-idf-master

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.26.0.
    d:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin
    d:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin
    d:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin
    d:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin
    d:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin
    d:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin
    d:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    d:\Espressif\tools\cmake\3.24.0\bin
    d:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin
    d:\Espressif\tools\ninja\1.10.2\
    d:\Espressif\tools\idf-exe\1.0.3\
    d:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
    d:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
    d:\Espressif\frameworks\esp-idf-master\tools

Checking if Python packages are up to date...
Constraint file: d:\Espressif\espidf.constraints.v5.2.txt
Requirement files:
 - d:\Espressif\frameworks\esp-idf-master\tools\requirements\requirements.core.txt
Python being checked: d:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe
d:\Espressif\frameworks\esp-idf-master\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Python requirements are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
Is this cmake version too high?

Lzw655, 2023-08-03:
No worries, you can compile now.

woshisunkejun, 2023-08-03:
Try again or not.
d:\Espressif\frameworks\esp-idf-master\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
Running ninja in directory D:\Downloads\esp32\ST77903_QSPI_1_6_400_400\build
Executing "ninja all"...
ninja: error: 'D:/Espressif/frameworks/esp-idf-master/components/esp_wifi/lib/esp32/libcore.a', needed by 'st77903_qspi.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the D:\Downloads\esp32\ST77903_QSPI_1_6_400_400\build\log\idf_py_stderr_output_33796 and D:\Downloads\esp32\ST77903_QSPI_1_6_400_400\build\log\idf_py_stdout_output_33796

Lzw655, 2023-08-03:
It seems like your IDF repository is incomplete. Try to use `git submodule update --init --recursive`

Lzw655, 2023-08-03:
The `libcore.a` library is too large to download with the slow network from GitHub. If so, you can try to use this [tool](https://gitee.com/EspressifSystems/esp-gitee-tools) in Gitee.

Lzw655, 2023-08-08:
Hi @woshisunkejun, if your problem has been resolved, please close this issue. Thank you!

closed 2023-08-21 by Lzw655