espressif/esp-dev-kits #80: 使用 ESP32-S3-LCD-EV_BOARD 开发板测试例程都无法正常显示

Rainbow-Cai, 2023-08-09:
对于 ESP32-S3-LCD-EV_BOARD 开发板对应的 examples ，我都做了测试，也修改了相应的参数到 480*480，因为没有摄像头，其中一个没测试，其他测试结果如下：

列程名称                      编译                                                   烧写                       运行
86box_demo                 OK                                                       OK                      无显示，输出log（86box_demo_log.txt）
86box smart panel        失败（86box_smart_panel.txt））
lvgl_demos                    OK（修改参数到子板2 480*480）      OK                     无显示，输出log(lvgl_demo_log.txt)      
smart_panel                  失败（修改了子板2参数）
usb_keyboard               OK  （修改参数）                                 OK                    无显示（USB 串口闪退）

没有一个成功的案列，希望能提供出厂的那个 soucecode 给我，那个是唯一能运行的。

espressif2022, 2023-08-09:
86box 看 log 是正常的
86box_smart_panel 编译失败，提示删除 espressif__esp-sr，按步骤操作就行
lvgl_demo 看着也是正常的
smart_panel 看是找 IIC 地址失败了，选择的 800*480 的配置文件

你的硬件是社么配置？
烧写我们出厂固件是否正常？

Hans-mail, 2023-08-09:
硬件是ESP32-S3-LCD-EV_BOARD 开发板 + 2号子板，屏幕是480*480，烧写出厂bin文件，启动正常

Hans-mail, 2023-08-09:
[image]

Lzw655, 2023-08-09:
@Rainbow-Cai smart_panel 仅适用于 800x480 子板，请仔细阅读 README，部分 example 有一定的硬件要求。

Rainbow-Cai, 2023-08-10:
Hi @Lzw655  出产固件里面，设置过后，如何重新设置参数？
但是有个问题就是，wifi 密码输入的时候，输入数字的时候，一按键就菜单就返回了，致使操作不成功，两次了都是这样。
是什么原因？

espressif2022, 2023-08-10:
> 

顶部下拉会出现设置栏，可以点击进入配置
wifi 密码输入完，按 √，完成输出

closed 2023-08-21 by Lzw655