espressif/esp-dev-kits #81: esp32c3 stuck in jtag mode

LucaCraft89, 2023-08-26:
i havbe an esp32 c3 i bought and after flashing a sketch to it and resetting afterwards it is only detected as usb jtag and not a serial (COM) device i tried holding boot while plugging in but no luck

Lzw655, 2023-08-28:
Hi @LucaCraft89, are you using the ESP32-C3-LCDkit board? If so, it is indeed only detected as a USB JTAG. Because the USB port is directly connected to the chip as below.

[image]
[image]

LucaCraft89, 2023-08-28:
Hi @Lzw655 i'm not and before i got the jtag behavior i was able to get the serial port

Lzw655, 2023-08-28:
Please provide the detailed development board model and version you are currently using.

LucaCraft89, 2023-08-28:
I got it on AliExpress [image]

Lzw655, 2023-08-28:
Well, it's not an official ESP board. Looks like this board directly connects the USB port with the ESP chip as same as the picture I showed before. If so, it should not be detected as a serial (COM) device anyhow.

Please check its scheme or provide it here and I'll check it.

LucaCraft89, 2023-08-28:
[image]

Lzw655, 2023-08-28:
According to this schematic, the USB port is directly connected to the ESP chip and it should always be detected as a USB JTAG device.

leeebo, 2023-08-29:
@LucaCraft89 Hi, for the USB-Serial-JTAG controller in ESP32-C3, which has two interfaces, one is a USB serial interface, and the other is a USB JTAG interface. Two new devices should be detected as below:

[image]

If you can not find the serial (COM) device, please re-install the driver using tools like Zadig, choose interface 0, then choose `usbser` or `USB CDC` driver

[image]

LucaCraft89, 2023-08-29:
Fact is i don't seee the second one

LucaCraft89, 2023-08-30:
Ohhh ok i'll try

Lzw655, 2023-09-21:
Hi @LucaCraft89, if your problem has been resolved, please close this issue. Thank you!

LucaCraft89, 2023-09-21:
Hi @Lzw655 i have not resolved the issue and still need help with it

Lzw655, 2023-09-21:
> Hi @Lzw655 i have not resolved the issue and still need help with it

Please describe the detailed issue and @leeebo will help you.

closed 2023-11-03 by Lzw655