espressif/esp-dev-kits #82: 开发板 esp32-s3-lcd-ev-board\examples\86box_smart_panel 中汉字不能显示

13825650991, 2023-09-01:
在开发板列程中有些汉字不能显示出来，为什么呢？

[image]

espressif2022, 2023-09-01:
这个字号的码表里面没有对应的字符，需要添加一下
https://lvgl.io/tools/fontconverter

13825650991, 2023-09-01:
那如何需要全部简体汉字，如何办？

 

 https://lvgl.io/tools/fontconverter

espressif2022, 2023-09-01:
1：要是还是用 map 表的话，就把汉字的范围填进去。

2：可以用这个字库，目前正式 demo github 还没出去，https://components.espressif.com/components/espressif/freetype

13825650991, 2023-09-01:
谢谢

Lzw655, 2023-09-21:
@13825650991 请问你的问题是否已经解决，如果是的话请关闭此 issue，谢谢!

13825650991, 2023-09-22:
通过生成字库文件，并申明，已经可以使用。

closed 2023-09-22 by 13825650991

13825650991, 2023-09-23:
上一个问题已经关闭，现在有一个新的问题，由于github 太难登录，所以只能把问题发在这里了。

title： 在项目中做编码转化的时候，用到c库中的iconv 函数，但是该函数总是返回 -1，errno = 9
代码如下：

int code_convert(char* from_charset,char *to_chartset,char* inbuf,size_t inlen,char *outbuf,size_t outlen)
{
    iconv_t cd;
    char **pin = &amp;inbuf;
    char **pout = &amp;outbuf;
    cd = iconv_open(to_chartset,from_charset);
    if(cd == 0)
    {
        ESP_LOGI("ESP_TOOS", "iconv_open =%d\n",cd);
        return -1;
    }
    memset(outbuf,0,outlen);
    if(iconv(cd,pin,&amp;inlen,pout,&amp;outlen) == -1)
    {
        switch(errno)
        {
        case EBADF: 		/* Bad file number */
        ESP_LOGI("ESP_TOOS", "EBADF\n");
        break;
        case E2BIG:
        ESP_LOGI("ESP_TOOS", "E2BiG\n");
        break;
        case EILSEQ:
         ESP_LOGI("ESP_TOOS","EILSEQ\n");
        break;
        case EINVAL:
         ESP_LOGI("ESP_TOOS","EINVAL\n");
        break;
        }
        ESP_LOGI("ESP_TOOS", "errno:%d,iconv =-1\n",errno);
        iconv_close(cd);
        return -1;
    }
    iconv_close(cd);
    *pout ='\0';
    return 0;
}

int g2u(char* inbuf,size_t inlen,char* outbuf,size_t outlen)
{
    return code_convert("GBK","UTF-8",inbuf, inlen,outbuf,outlen);
}

网上有反映是C库不完整所致，所以请帮忙测试，并生成完整c库，谢谢。