espressif/esp-dev-kits #86: i2s_common: i2s_channel_disable : the channel has not been enabled yet

llinzzi, 2023-11-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### Espressif SoC revision.

ESP32C3

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3-LCDKIT

### Power Supply used.

USB

### What is the expected behavior?

knob_panel的官方例子，预计应该正常播放音频。

### What is the actual behavior?

实际上，第一次会正常播放，第二次会报错。错误如下
i2s_common: i2s_channel_disable : the channel has not been enabled yet

### Steps to reproduce.

烧录，旋转。

LHYhorion, 2023-11-29:
> ### Answers checklist.
> * [x]  I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
> * [x]  I have [...]

目前已知这个报错，这个报错并不影响音频功能，后续会修复这个问题。

Lzw655, 2023-12-19:
@llinzzi 你好，感谢你的反馈，请问还有什么问题吗？如果没有，请关闭此 issue，或者我将在下周关闭它。

closed 2023-12-26 by Lzw655

llinzzi, 2024-01-19:
大约哪个版本会修复

Lzw655, 2024-01-20:
Hi @llinzzi, 这个是 esp_codec_dev 组件内的报错，我们已经跟组件作者反映，需要等待那边的修复。