espressif/esp-dev-kits #87: macOS is unable to start OpenOCD debugging

yin-qiyu, 2023-12-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

openocd has been installed correctly

 % openocd --version                   
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html

% openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!

[OpenOCD]
❌ Error: unable to open ftdi device with description '*', serial '*' at bus location '*'

lijunru-hub, 2023-12-05:
If you are using esp-usb-bridge for JTAG, please refer to https://github.com/espressif/esp-usb-bridge#jtag-bridge.

lijunru-hub, 2024-05-27:
If you have no further questions or issues, we will proceed to close this issue. If you need any more help, please feel free to reopen this issue or create a new one.

Thank you for your understanding.

closed 2024-08-02 by Lzw655