espressif/esp-dev-kits #89: espc3-lcd-kit free IO pins?

tomasApo, 2024-03-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I was hoping to use free pins on the ESP32-c3-lcd kit for IO 

but it looks like very single pin is being used. I tried with io18 and io19 to no success as I think these are used for USB data. 

Any ideas?

[image]
[image]

Lzw655, 2024-03-26:
Hi @tomasApo, may I ask what functionality you intend to achieve using these IOs? For example, are you just looking for simple digital output?

tomasApo, 2024-03-26:
@Lzw655 yes just looking to achieve some simple digital out like 0v and 3.3v

Lzw655, 2024-03-26:
Since `IO18` and `IO19` are used for USB port which need to be connected to PC, it's not recommended to use these two IOs.
But I think you can use the following IOs:
1. `IO4` (without using `IR_RX/IR_TX`)
2. `RXD0` and `TXD0` (without using UART0)
3. `IO8` (the RGB LED will flash...)