espressif/esp-dev-kits #94: runing the LVGL demo with a slight modification

ganindu7, 2024-04-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to run the [lvgl demo](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-lcd-ev-board/examples/lvgl_demos) with a slight modification tot he graphics asset. (just to understand how this works becase I am not familiar with ESP32 at all. 

all I did was simply modify the .png file `{............}/examples/lvgl_demos/components/lvgl__lvgl/demos/widgets/assets/avatar.png`

then I ran 

`idf.py fullclean`

and 

```
idf.py -p /dev/ttyUSB0 flash monitor
```
bit I keep getting the old `avarar.png` I guess that this is becase this is coming from a binary blob being used instead of my changes. 

my question is how do I get to the point where my changes get reflected downstream. 

is there a setting to force rebuild including my changes.

Lzw655, 2024-05-07:
Hi @ganindu7, did you copy the component `lvgl__lvgl` from the folder `{............}/examples/lvgl_demos/managed_components` to the folder `{............}/examples/lvgl_demos/components`? If so, please also delete the file `{............}/examples/lvgl_demos/dependencies.lock` and build again. Otherwise, the project will always use the `lvgl` in the folder `{............}/examples/lvgl_demos/managed_components`.

ganindu7, 2024-05-07:
Thanks will do and let you know

Sent from Gmail Mobile

ganindu7, 2024-05-07:
Hi I have deleted the lock but it still builds the same way (with the default demo not changing) 

I have added my folder structure from my workspace(with depeth limited to 3 for clarity) can you splot anything wrong here. 

```
ganindu@pc:~/Workspace/sandbox/expressif/workspace/lcd-dev-board/examples/lvgl_demos$ tree -d -L 3
.
├── build
│   ├── bootloader
│   │   ├── CMakeFiles
│   │   ├── config
│   │   └── esp-idf
│   ├── bootloader-prefix
│   │   ├── src
│   │   └── tmp
│   ├── CMakeFiles
│   │   ├── 3.22.1
│   │   ├── bootloader.dir
│   │   ├── CMakeTmp
│   │   ├── git-data
│   │   └── lvgl_demos_ganindu.elf.dir
│   ├── config
│   ├── esp-idf
│   │   ├── app_trace
│   │   ├── app_update
│   │   ├── bootloader
│   │   ├── bootloader_support
│   │   ├── bt
│   │   ├── CMakeFiles
│   │   ├── cmock
│   │   ├── console
│   │   ├── cxx
│   │   ├── driver
│   │   ├── efuse
│   │   ├── esp_adc
│   │   ├── esp_app_format
│   │   ├── esp_bootloader_format
│   │   ├── esp_coex
│   │   ├── esp_common
│   │   ├── espcoredump
│   │   ├── esp_driver_ana_cmpr
│   │   ├── esp_driver_cam
│   │   ├── esp_driver_dac
│   │   ├── esp_driver_gpio
│   │   ├── esp_driver_gptimer
│   │   ├── esp_driver_i2c
│   │   ├── esp_driver_i2s
│   │   ├── esp_driver_isp
│   │   ├── esp_driver_jpeg
│   │   ├── esp_driver_ledc
│   │   ├── esp_driver_mcpwm
│   │   ├── esp_driver_parlio
│   │   ├── esp_driver_pcnt
│   │   ├── esp_driver_rmt
│   │   ├── esp_driver_sdio
│   │   ├── esp_driver_sdm
│   │   ├── esp_driver_sdmmc
│   │   ├── esp_driver_sdspi
│   │   ├── esp_driver_spi
│   │   ├── esp_driver_tsens
│   │   ├── esp_driver_uart
│   │   ├── esp_driver_usb_serial_jtag
│   │   ├── esp_eth
│   │   ├── esp_event
│   │   ├── esp_gdbstub
│   │   ├── esp_hid
│   │   ├── esp_http_client
│   │   ├── esp_http_server
│   │   ├── esp_https_ota
│   │   ├── esp_https_server
[... 182 lines cut ...]
-- Adding linker script /home/ganindu/Workspace/sandbox/expressif/workspace/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/ganindu/Workspace/sandbox/expressif/workspace/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /home/ganindu/Workspace/sandbox/expressif/esp-idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /home/ganindu/Workspace/sandbox/expressif/esp-idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /home/ganindu/Workspace/sandbox/expressif/esp-idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /home/ganindu/Workspace/sandbox/expressif/esp-idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /home/ganindu/Workspace/sandbox/expressif/esp-idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /home/ganindu/Workspace/sandbox/expressif/esp-idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /home/ganindu/Workspace/sandbox/expressif/esp-idf/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- BUTTON: 3.2.0
-- ESP_LCD_GC9503: 3.0.1
-- ESP_LCD_PANEL_IO_ADDITIONS: 1.0.1
CMake Error at /home/ganindu/Workspace/sandbox/expressif/esp-idf/esp-idf/tools/cmake/component.cmake:314 (message):
  Include directory
  '/home/ganindu/Workspace/sandbox/expressif/workspace/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/managed_components/lvgl__lvgl/demos'
  is not a directory.
Call Stack (most recent call first):
  /home/ganindu/Workspace/sandbox/expressif/esp-idf/esp-idf/tools/cmake/component.cmake:482 (__component_add_include_dirs)
  main/CMakeLists.txt:10 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "/home/ganindu/Workspace/sandbox/expressif/workspace/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/ganindu/Workspace/sandbox/expressif/workspace/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/log/idf_py_stderr_output_22122 and /home/ganindu/Workspace/sandbox/expressif/workspace/esp-dev-kits/esp32-s3-lcd-ev-board/examples/lvgl_demos/build/log/idf_py_stdout_output_22122

```

Lzw655, 2024-05-07:
Yes, the component `lvgl__lvgl` in `managed_components` should be deleted. And the first line of file `lvgl_demos/main/CMakeLists.txt` should also be changed from `set(LV_DEMO_DIR ../managed_components/lvgl__lvgl/demos)` to `set(LV_DEMO_DIR ../components/lvgl__lvgl/demos)`.

ganindu7, 2024-05-07:
Hi Liu, 

That fixed part of the problem, Now it is buiulding the demo from the `componsnts` directory instead of the `managed_componsnts` directory. 

but the original problem (the graphic not changing) is still present. 

To remind what I have done to see this change is to just change the `avatar.png`  file on

the `components/lvgl__lvgl/demos/widgets/assets` folder and expect to see the updated file appear on the display. 

I wonder if this is pre built and optimised 

Cheers, 
Ganindu.

ganindu7, 2024-05-07:
> For LVGL, it's usually necessary to use tools
> <https://lvgl.io/tools/imageconverter> to convert images into arrays.
> What you need to replace is the img_demo_widgets_avatar.c file, not the
> image itself.

HI Liu,

Thanks a lot! That essentially fixed the original issue. There aere other
suff like the image seems different. I reckon that is a settings issue.
Thanks a lot for your time!

closed 2024-05-07 by ganindu7

Lzw655, 2024-05-08:
You are welcome!