espressif/esp-dev-kits #99: Using the ESP32-S3-USB-Bridge in a wireless mode at a remote site behind a firewall

ammaree, 2024-07-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We would like to use this dev board to access remote target devices located at customer site and behind a firewall, across the internet. For this we believe ESP-NOW need to be replaced with a normal IP stack and the remote host device need to be configured to act as STA to the local Wifi at the customer site.

To make installation easier the remote device should also enable AP mode if it cannot connect to the local SSID. The remote host device should also be able to be powered from the target device via 5V or 3V3

To fulfil our remote diagnostic and support requirements we need to be able to access and control the remote target (RxD, TxD, EN and IO0) through the remote host, allowing us to monitor (UART) reset, read and write flash of the remote target.

This configuration working across the internet brings complications of firewalls (NAT config we can handle) and latency / propagation delays. It also requires some enhanced telnet support to control the RST and BOOT signals between the central site and the remote host to target connection.

At the central support site we would like to use standard Espressif tools in idf.py and esptool.py, both with the socket://remote.host:port addressing scheme:

Any reason why this functionality is not demonstrated in an example?
Is there any reason why this is not (easily) possible to achieve, something we might be missing.

Ultimately we are looking for a reliable, higher performing version of ESP-Link focussed on ESP32xx target devices, but without the MQTT etc bells and whistles.

All advice appreciated.

ammaree, 2024-08-26:
@igrr 

A month later anger response from anybody. The current repo does not build with ESP-IDF master, comes up with 100's of errors.

Can we please get to a point ASAP where this can build with 5v3 or master, we would like to start a process to add normal IP support to access a device at a remote site.

Please !!!!

lijunru-hub, 2024-08-28:
Sorry for the late reply. The public network debugging solution is quite interesting, and we already have clients who have developed remote debuggers based on the public network. However, we currently don't have the resources to dedicate to this feature, so you might need to experiment on your own. We will work on fixing the compilation issue as soon as possible. In the meantime, you can revert to the IDF version before release/v5.2 to ensure correct compilation.

ammaree, 2024-08-28:
@lijunru-hub 

Thanks for the response, sorry about nagging.
In order for us to make a decision on possibly committing resources, can you help with some guidance.

A: [https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-usb-bridge/examples/usb_wireless_bridge](url)
B: [https://github.com/espressif/esp-usb-bridge](url)

1: Why are there 2 different repos, both focussing on very much the same functionality
2: Of the 2 repos above, which of them is where Espressif will be devoting resources?
3: If your focus is specifically on the repo A above, any plans to consolidate the SWD and similar functionality from repo B?
4: Any timeframe for getting repo A (if that is where you will focus) to a successful compile state, with updated IDF and components?

Thank

lijunru-hub, 2024-08-28:
1. A is a wireless debugging solution based on an older version of B.
2. More code maintenance will focus on B.
3. SWD has limited relevance for the debugging needs focused on ESP32 in A.
4. ASAP.

ammaree, 2024-08-28:
Thanks for the response.

So if I understand correctly you will be focussing on A (usb_wireless_bridge) ASAP.

Any plans or ideas for adding normal IP type communication as alternative to or replacement for ESP-NOW?

lijunru-hub, 2024-08-29:
It may be necessary for you to implement IP-based debugging on your own, but it is feasible, as some clients have already developed related products.

ammaree, 2024-09-03:
@lijunru-hub 
Any progress on updating the usb_wireless_bridge [example](https://github.com/espressif/esp-dev-kits/issues/url) to compile using IDF v5.3 or later, preferably master?

We would like to get started working on the LWIP functionality but unable to do so with current state of the repo.

Thanks for your help

lijunru-hub, 2024-09-05:
I'll look into it as soon as possible.

ammaree, 2024-09-27:
@lijunru-hub 

Any progress on updating for IDF v5.3 or master. It would be of great value to have some idea of when we can expect to start testing, even if just an early beta version?

In your post about a month ago you mentioned that some clients have already developed related products, any chance you can provide me with links to those so I can have a look at viability to purchase/use?

The requirement to debug and flash remote devices behind a firewall has high priority on our side...

Thanks

lijunru-hub, 2024-09-29:
related products https://item.taobao.com/item.htm?id=618820111522&pisk=fp8tMza3tvDGpz3T4nihmeTmgZlH6FdZpdR7otXgcpppGd6Gni61dk1G1o6flI1fkp9m_OAjQKLAdI-i_iDaMHpwNGvvgf5fhK9AnhmnrCRNuZMkD40k_Km69bvAfG1COB5RG5_iu-PcuZMoHk4fb4Q2EWv6dsII9sfVGl9fck1CLsE_CK61AJ1VgZ6Xrgzov-N4rUxd1VSO6D0CezCCXTOK4raYf1ylEC_gkrBOJGEM19UbleKUhZWfC0Ph3NXwL1pqzP71DnTGRFMIlatypEI5ymcNfIKp3NYIGl6WjdjD7Me_f_QOBgT1x4hBTitpJNYs3uRH9OsR-HmEIib9B3WPfDkeHBBMhFIQBgR2rUp3E8XRilGK9orVf67g3NL-tpMNS6Bo61E40i1F9TcK9orVc65dEfeT0oSfT&spm=a1z10.3-c-s.w4002-24057373988.11.46b81c44Mhvprd

ammaree, 2024-09-29:
Link does not work, tried 3 different browsers, also have Taobao app loaded on phone

ammaree, 2024-10-08:
@lijunru-hub, sorry to continue bugging you but...

1. Any chance of an indication when the updates to support v5.3/master can be expected?
2. Can you please check and resend links for any and all already developed similar/related products?

If we can find something already available with the required minimum functionality and at a justifiable price that can buy us (and yourself) some time.

Please, please, please, becoming critical this side..

ESP-Mars, 2024-10-09:
> related products [...]

That's odd. I can access the link normally from my side using Chrome, and my IP address is based in Hong Kong.
[image]

ammaree, 2024-10-09:
@ESP-Mars 
@lijunru-hub 

Thanks for the info, got to the link, translated, but the PW200 but cannot get more info, nor can I create a login from South Africa.

Found www.powerwriter.com, only product that MIGHT be able to do the trick is the PowerDebugger. None of the PW200/300/400/PWLINK2/PWX1 has wifi capabilities. Considering the list of functionality it seems the product is focussed on ARM and limited RISC-V with SWD/ITM/JTAG etc, none of which we are interested in.

All we need is remote access via public IP to ESP32xxx UART port for monitoring and flashing. Have looked for a price to determine feasibility but cannot find it anywhere, so maybe you can help with separate pricing for PowerDebugger TX and RX items.

But considering we need 2xTX and ~10xRX to start with, very unlikely to be justifiable.

ammaree, 2024-10-17:
@igrr 

Any chance you can assist to move this issue forward, please?

We are based in South Africa and the TaoBao link provided is of no help to us. The proposed alternative product does not look viable to us, not specifically ESp32 focussed and unlikely justifiable price.

If we can just get the existing example updated to successfully compile with v5.3 or preferably master, that will be a big step forward.

Although the ultimate goal still has to be full IP support to access remote devices behind a firewall.

Thanks

lijunru-hub, 2024-10-21:
@ammaree 
I completely understand your sense of urgency. We have already updated the example to support the latest IDF build, but the code release will take a bit more time. Please be patient.

ammaree, 2024-10-25:
@lijunru-hub 

Thanks for the update, have synced the repo, done full clean (couple of times) but still coming up with errors

```
Executing action: all (aliases: build)
Running ninja in directory /Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/build
Executing "ninja all"...
[4/120] Building C object esp-idf/espressif__esp-now/CMakeFiles/__idf_espressif__esp-now.dir/src/espnow/src/espnow_group.c.obj
FAILED: esp-idf/espressif__esp-now/CMakeFiles/__idf_espressif__esp-now.dir/src/espnow/src/espnow_group.c.obj 
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.4-dev-3602-ga97a7b0962-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/build/config -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/control/include -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/debug/include -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/debug/include/sdcard -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/debug/src -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/debug/src/commands/pcap -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/espnow/include -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/ota/include -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/provisioning/include -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/security/include -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/security/include/protocomm/security -I/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/utils/include -I/Users/andremaree/DevSpace/z-sdk/master/components/protocomm/proto-c -I/Users/andremaree/DevSpace/z-sdk/master/components/newlib/platform_include -I/Users/andremaree/DevSpace/z-sdk/master/components/freertos/config/include -I/Users/andremaree/DevSpace/z-sdk/master/components/freertos/config/include/freertos -I/Users/andremaree/DevSpace/z-sdk/master/components/freertos/config/xtensa/include -I/Users/andremaree/DevSpace/z-sdk/master/components/freertos/FreeRTOS-Kernel/include -I/Users/andremaree/DevSpace/z-sdk/master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/andremaree/DevSpace/z-sdk/master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/Users/andremaree/DevSpace/z-sdk/master/components/freertos/esp_additions/include -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_hw_support/include -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_hw_support/include/soc -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_hw_support/include/soc/esp32s3 -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_hw_support/dma/include -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_hw_support/ldo/include -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_hw_support/debug_probe/include -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_hw_support/port/esp32s3/. -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_hw_support/port/esp32s3/include -I/Users/andremaree/DevSpace/z-sdk/master/components/heap/include -I/Users/andremaree/DevSpace/z-sdk/master/components/heap/tlsf -I/Users/andremaree/DevSpace/z-sdk/master/components/log/include -I/Users/andremaree/DevSpace/z-sdk/master/components/soc/include -I/Users/andremaree/DevSpace/z-sdk/master/components/soc/esp32s3 -I/Users/andremaree/DevSpace/z-sdk/master/components/soc/esp32s3/include -I/Users/andremaree/DevSpace/z-sdk/master/components/soc/esp32s3/register -I/Users/andremaree/DevSpace/z-sdk/master/components/hal/platform_port/include -I/Users/andremaree/DevSpace/z-sdk/master/components/hal/esp32s3/include -I/Users/andremaree/DevSpace/z-sdk/master/components/hal/include -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_rom/include -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_rom/esp32s3/include -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_rom/esp32s3/include/esp32s3 -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_rom/esp32s3 -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_common/include -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_system/include -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_system/port/soc -I/Users/andremaree/DevSpace/z-sdk/master/components/esp_system/port/include/private -I/Users/andremaree/DevSpace/z-sdk/master/components/xtensa/esp32s3/include -I/Users/andremaree/DevSpace/z-sdk/master/components/xtensa/include -I/Users/andremaree/DevSpace/z-sdk/master/components/xtensa/deprecated_include -I/Users/andremaree/DevSpace/z-sdk/master/components/lwip/include -I/Users/andremaree/DevSpace/z-sdk/master/components/lwip/include/apps -I/Users/andremaree/DevSpace/z-sdk/master/components/lwip/include/apps/sntp -I/Users/a [... cut ...]
/Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/managed_components/espressif__esp-now/src/espnow/src/espnow_group.c:30:20: warning: 'TAG' defined but not used [-Wunused-variable]
   30 | static const char *TAG = "espnow_group";
      |                    ^~~
[21/120] Linking C static library esp-idf/unity/libunity.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/build/log/idf_py_stderr_output_51311 and /Users/andremaree/DevSpace/z-projects/usb_wireless_bridge/build/log/idf_py_stdout_output_51311
```

Any suggestions?

lijunru-hub, 2024-10-25:
I am unable to reproduce this error on my side; the compilation completes successfully.

ammaree, 2024-10-25:
Are you building using Master or a different version?

I am using absolute latest master on macOS.

ammaree, 2024-10-25:
@lijunru-hub 

Apologies, fault is my side. Cleaned everything, resync'd and tried again, all working.

Will now try to flash the new firmware and then start looking at adding the remote IP support.

Just so I know, is there a changelog available that I can look at to be aware of changes?

Thanks again.

lijunru-hub, 2024-10-28:
@ammaree 
Due to some changes in the dependent components, updating the component versions to resolve certain compilation errors will allow support for the master branch. You can check the specific changes in this [commit](https://github.com/espressif/esp-dev-kits/commit/f9e5aa66c284ab7de41895758922571fcf1735f2#diff-5516b89b71a301e0b768119523ba33f8182862845b884e8b035264e1796a8f93).

Lzw655, 2024-12-24:
Since there has been no feedback, we will temporarily close this issue.

closed 2024-12-24 by Lzw655

ammaree, 2025-01-22:
@lijunru-hub 
@Lzw655 

I have build tried the firmware numerous times, cannot get it to work. Initially the WS2812 LED did not function, fix error in sdkconfig.defaults specifying GPIO=48 ! After that the LED started working but no serial port shows up after flashing, unplugging and replugging.

So, although I can now change modes local<>host<>slavering the S3BOOT button, I cannot flash any attached device since the bridge does not show up as a USB port. I then had a look at the component manifest and started checking if newer versions of the managed components were available and found many.

I started upgrading the components 1 by 1 with newer versions it made no difference, same problem. NO serial port found after flashing and rebooting.

Any suggestion please, this is becoming embarrassing for me.

ammaree, 2025-01-22:
@lijunru-hub @Lzw655 @ESP-Mars 

Just something I forgot to mention.

I also spent some time trying to get the ESP NOW mode connected, hoping that the USB port will miraculously appear on my host MacBook Pro. But no luck, I was able to get the host & slave to sync (I think) since I was able to go through the process with the flashing LEDs, successfully got to solid green on the slave and blue on the host, but still no USB port showing up.

I also disconnected the host from my MacBook and shutdown/rebooted a number of times, checking if the USB port shows up after reboot, still no luck.

ammaree, 2025-01-23:
@lijunru-hub @ESP-Mars @Lzw655 

If we can please reopen this issue, much appreciated.

ammaree, 2025-02-05:
@lijunru-hub 
@ESP-Mars 
@Lzw655 
@igrr 

In the absence of any feedback on this issue we have spent a day testing and trying to use the previous version of the firmware that shipped with the ESP32-S3-USB-BRIDGE as of Sept/Oct'24.

The test environment was an empty (term starts next week) multi-storied student residence with all (40x) ESP32 motes mounted to the roof/ceiling in open passages. We tested connecting from a MacBook Pro to each mote using a pair of ESP32-S3-USB-BRIDGE devices, shipped firmware, configured and paired in ESP-NOW Master/Slave mode.

Connecting indoors with line of sight (LOS) at a distance (X meters) measuring time to flash 987KB app binary.
           Connecting              Flashing                Time               Comment
5m     100% success         80% success       200~220s
10m    33% success          0% success          Infinite            Only tried 6 devices, 2 connected but then failed to flash
7m      67% success          84% success       200~230s.      Same 6 devices as above

Using a USB cable we have 100% success connecting and flashing, time for same binary 22~23s, ~10x the throughput.

We believe the shipped firmware is only of use in a very close local environment for monitoring the UART on the remote device. For flashing, extracting core dumps and similar larger data volumes it is of zero or very limited use, primarily due to the unacceptably low throughput as well as the limited range.

I would guess the cause lies somewhere within ESP-NOW, possibly also within the application source code. Maybe some of the problems and bottlenecks have been solved in the updated firmware but, as mentioned is my post 2 weeks ago, the latest firmware does not work.

Ultimately, having this tool work reliably and at reasonably speed (~200Kbps or more) would make it a powerful tool to support the 1B+ of ESP32/8266 production devices.

Adding true remote, via the public internet and behind the firewall, capabilities would take it from powerful to awesome.

Our request is simple:

If Espressif sees value and/or has plans to fix the range, performance and reliability problems then a committed timeline towards achieving that would be appreciated.

If Espressif sees no value in or has no plans to fix the firmware then please let us know so we can move on and not waste any more time.

Looking forward to your soonest response.

PS: If a new issue is required to take this process further, please say so since I do not have permissions to reopen this issue.

lijunru-hub, 2025-02-05:
@ammaree 
Apologies for the delayed response—we were on holiday last week.

First, the wireless functionality in this firmware does not currently guarantee error-free data transmission.

Second, the data rate of ESP-NOW has been tested to reach a maximum of 200 Kbps in an open environment, but it decreases as the transmission distance increases. You can refer to the official documentation for more details: [ESP-NOW Data Rate](https://docs.espressif.com/projects/esp-faq/zh_CN/latest/application-solution/esp-now.html#esp32-esp-now).

Third, remote OTA updates are not planned at this time, so you will need to implement them on your own.

Additionally, there are no current plans to optimize the stability or speed of the flashing process.

ammaree, 2025-02-05:
@lijunru-hub 

Thanks for the response, I am aware of the Chinese new year, but numerous messages in almost 3 weeks have been ignored.

The situation of the current/latest "updated" example code not working is however being ignored.

We tried getting the firmware to work by selectively updating the component that the example relies one, still with no success.