espressif/esp-dev-kits #100: esp32-p4-function-ev-board/examples /lvgl_demo_v9 fails to build

dmartauz, 2024-07-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3 (fresh install by online installer)

### Espressif SoC revision.

ESP32-P4 ECO1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-P4-Function-EV-Board

### Steps to reproduce.

follow https://github.com/espressif/esp-dev-kits/blob/master/esp32-p4-function-ev-board/examples/lvgl_demo_v9/README.md
fails at idf.py menuconfig

### Build Logs.

```plain
C:\Espressif\frameworks\esp-idf-v5.3\workspace\esp32-p4-function-ev-board\examples\lvgl_demo_v9>idf.py menuconfig
Executing action: menuconfig
Running cmake in directory C:\Espressif\frameworks\esp-idf-v5.3\workspace\esp32-p4-function-ev-board\examples\lvgl_demo_v9\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Espressif\frameworks\esp-idf-v5.3\workspace\esp32-p4-function-ev-board\examples\lvgl_demo_v9"...
-- IDF_TARGET is not set, guessed 'esp32p4' from sdkconfig 'C:/Espressif/frameworks/esp-idf-v5.3/workspace/esp32-p4-function-ev-board/examples/lvgl_demo_v9/sdkconfig.defaults'
-- Found Git: C:/Espressif/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32p4
Dependencies lock doesn't exist, solving dependencies.
Using component placed at C:\Espressif\frameworks\esp-idf-v5.3\workspace\esp32-p4-function-ev-board\examples\common_components\esp32_p4_function_ev_board for dependency esp32_p4_function_ev_board(*), specified in C:/Espressif/frameworks/esp-idf-v5.3/workspace/esp32-p4-function-ev-board/examples/lvgl_demo_v9/main\idf_component.yml
.......................Using component placed at C:\Espressif\frameworks\esp-idf-v5.3\workspace\esp32-p4-function-ev-board\examples\common_components\esp32_p4_function_ev_board for dependency esp32_p4_function_ev_board(*), specified in C:/Espressif/frameworks/esp-idf-v5.3/workspace/esp32-p4-function-ev-board/examples/lvgl_demo_v9/main\idf_component.yml
.......................CMake Error at C:/Espressif/frameworks/esp-idf-v5.3/tools/cmake/build.cmake:552 (message):
  ERROR: Because no versions of esp32_p4_function_ev_board match
  >=0.0.0,<0.0.1 || >0.0.1

   and esp32_p4_function_ev_board (0.0.1) depends on lvgl/lvgl (>=8), esp32_p4_function_ev_board (>=0.0.0) requires lvgl/lvgl (>=8).

  So, because no versions of lvgl/lvgl match >=8

   and project depends on esp32_p4_function_ev_board (*), version solving failed.

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.3/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-idf-v5.3/workspace/esp32-p4-function-ev-board/examples/lvgl_demo_v9/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

dmartauz, 2024-08-01:
@Horion0415 it seems it is up to date
`
C:\Espressif\frameworks\esp-idf-v5.3\workspace\esp32-p4-function-ev-board\examples\lvgl_demo_v9>pip install --upgrade idf-component-manager
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Requirement already satisfied: idf-component-manager in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (1.5.3)
Requirement already satisfied: packaging in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (24.1)
Requirement already satisfied: requests<3 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (2.32.3)
Requirement already satisfied: urllib3<2 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (1.26.19)
Requirement already satisfied: requests-file<2 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (1.5.1)
Requirement already satisfied: requests-toolbelt in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (1.0.0)
Requirement already satisfied: schema<=0.7.5 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (0.7.5)
Requirement already satisfied: six in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (1.16.0)
Requirement already satisfied: tqdm<5 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (4.66.4)
Requirement already satisfied: pyparsing in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (3.1.2)
Requirement already satisfied: colorama in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (0.4.6)
Requirement already satisfied: pyyaml>5.2 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (6.0.1)
Requirement already satisfied: cachecontrol>0.12.6 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from cachecontrol[filecache]>0.12.6; python_version >= "3.6"->idf-component-manager) (0.14.0)
Requirement already satisfied: contextlib2>0.6.0 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (21.6.0)
Requirement already satisfied: click in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from idf-component-manager) (8.1.7)
Requirement already satisfied: msgpack<2.0.0,>=0.5.2 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from cachecontrol>0.12.6->cachecontrol[filecache]>0.12.6; python_version >= "3.6"->idf-component-manager) (1.0.8)
Requirement already satisfied: filelock>=3.8.0 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from cachecontrol[filecache]>0.12.6; python_version >= "3.6"->idf-component-manager) (3.15.4)
Requirement already satisfied: charset-normalizer<4,>=2 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from requests<3->idf-component-manager) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from requests<3->idf-component-manager) (3.7)
Requirement already satisfied: certifi>=2017.4.17 in c:\espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from requests<3->idf-component-manager) (2024.7.4)
`

dmartauz, 2024-08-07:
@Horion0415 I just tried on different machine with Windows 11 and I get exactly the same result. What do you recommend do make it work?

Horion0415, 2024-08-08:
It looks like the issue is with fetching the LVGL component. Have you made any modifications to this project? You can try cleaning the build directory and the dependencies.lock file and then rebuild the project. Alternatively, you can try manually fetching the corresponding version of LVGL to see if it builds correctly.

Daniel-Abdi, 2024-08-10:
Have encountered the same issue starting today. Problem persisted after full clean for me as well. My workaround is to replace items in idf_component.yaml to:

`dependencies:
  espressif/esp32_p4_function_ev_board: "^3.0.0"
  lvgl/lvgl:
    version: "^9"
    public: true
`
Then delete all build directories, remember to re-set the target to esp32P4, and try building again.

Horion0415, 2024-08-10:
> espressif/esp32_p4_function_ev_board: "^3.0.0"

The currently released version uses a temporary BSP (the official BSP hadn’t been released at the time). The official BSP was just released in the past few days, so when you change it to espressif/esp32_p4_function_ev_board: "^3.0.0", you will be pulling the official BSP. However, the project doesn’t support it yet. We will update the project to support the official BSP next week. But I don’t think this should affect pulling the LVGL component. May I ask if you’re also using a Windows system?

Horion0415, 2024-08-10:
I believe the issue is likely caused by the package manager on the Windows system. I will investigate this problem and work on an update and fix next week. Additionally, I will adapt the project to the officially released BSP.

Daniel-Abdi, 2024-08-10:
I've encountered the error on 2 different systems, one running windows and the other running Ubuntu 22.04 inside WSL. Cloned a fresh copy of the repo for both systems to check whether the issue was on my end or not. Changing the BSP to the new one (encountering a crash because it isn't supported yet) and then back to the old version managed to fix the issue for me.

Horion0415, 2024-08-10:
I’m really sorry about this situation. I’m puzzled by the issue since I can successfully compile on both Mac and Linux systems. Could you please check the version of the package manager with ‘$ pip list | grep idf-component-manager’? You might want to try downgrading to version 1.5.3 and then try fetching again. However, I believe this issue will be resolved once I update the project with the new BSP next week.

brookMD, 2024-08-16:
@Horion0415 I have an issue with P4 Function EV board where I can't run LVGLv9 demo. I got the LVGL v9 demo to compile on Ubuntu 22.04 using VScode IDE and ESP IDF extension. But LCD doesn't display anything. The factory example binary included in the examples seems to run fine. Is this something to do with BSP?

Horion0415, 2024-08-19:
Yes, this is related to the BSP. Since the example was released before the official BSP version, it uses a temporary version of the BSP. I will update to the new BSP this week.

Horion0415, 2024-08-20:
Hello, I have updated the BSP and released the relevant examples for esp-ui. You can try again to see if the issue is resolved.

dmartauz, 2024-08-20:
@Horion0415 esp_ui_demo, lvgl_demo_v8 and lvgl_demo_v9 build without any problems. I also flashed them on the devboard and they work well.

brookMD, 2024-08-20:
> Hello, I have updated the BSP and released the relevant examples for esp-ui. You can try again to see if the issue is resolved.

I tried building after the updates and it works now!

Horion0415, 2024-08-21:
> > Hello, I have updated the BSP and released the relevant examples for esp-ui. You can try again to see if the issue is resolved.
> 
> I tried building after the updates and it works now!

Great, I'm very glad that this issue can be resolved.