espressif/esp-dev-kits #102: ESP32P4-function-ev-board

liujkek22, 2024-08-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

so,why not give us some examples about how to use wifi with ESP32C6min and camera?

Horion0415, 2024-08-15:
Hello, we are currently optimizing the ISP driver and will gradually release the relevant examples in the future.

chegewara, 2024-08-21:
Maybe silly question, but, are asking for example to use wifi and camera together or just separate examples?