espressif/esp-dev-kits #103: why  cannt run "bsp_sdcard_mount() in the demo of "esp-dev-kits/tree/master/esp32-p4-function-ev-board/examples/esp_ui_demo"

liujkek22, 2024-08-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Espressif SoC revision.

ESP32P4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32P4-function-ev-board

### Steps to reproduce.

- build the project and run well'
- change the code with the fats file system with sdcard;
- remove all code in main function only leave"bsp_sdcard_mount()",
- build an run;
- always error :E (1226) vfs_fat_sdmmc: host init failed (0x103）

...

### More Information.

I (1201) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x480094e6] Rx[0x48014dfc]
0x480094e6: transport_drv_sta_tx at D:/examples/esp_ui_demo/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:211
0x48014dfc: esp_wifi_remote_channel_rx at D:/examples/esp_ui_demo/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19

I (1209) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4800942c] Rx[0x48014dfc]
0x4800942c: transport_drv_ap_tx at D:/examples/esp_ui_demo/managed_components/espressif__esp_hosted/host/drivers/transport/transport_drv.c:241
0x48014dfc: esp_wifi_remote_channel_rx at D:/examples/esp_ui_demo/managed_components/espressif__esp_wifi_remote/esp_wifi_remote_net.c:19

I (1218) main_task: Started on CPU0
I (1222) main_task: Calling app_main()
E (1226) vfs_fat_sdmmc: host init failed (0x103).
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x48014ece
0x48014ece: app_main at D:/examples/esp_ui_demo/main/main.cpp:33 (discriminator 1)

file: "./main/main.cpp" line 33
func: void app_main()
expression: bsp_sdcard_mount()

abort() was called at PC 0x4ff07da9 on core 0
0x4ff07da9: _esp_error_check_failed at D:/ESP53/v5.3/esp-idf/components/esp_system/esp_err.c:49

Core  0 register dump:
MEPC    : 0x4ff00c3c  RA      : 0x4ff07db4  SP      : 0x4ff21070  GP      : 0x4ff11000
0x4ff00c3c: panic_abort at D:/ESP53/v5.3/esp-idf/components/esp_system/panic.c:463
0x4ff07db4: flash_read at D:/ESP53/v5.3/esp-idf/components/esp_system/port/image_process.c:77

TP      : 0x4ff21150  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x4ff210ac  S1      : 0x4ff210ac  A0      : 0x4ff210ac  A1      : 0x4ff2108e
A2      : 0x00000000  A3      : 0x4ff210d9  A4      : 0x00000001  A5      : 0x4ff15000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x4ff21090  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x4ff00003: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4ff21070: 0x00000000 0x00000000 0x4ff210ac 0x4ff0f62e 0x00000000 0x00000000 0x00000000 0x00000030
0x4ff0f62e: __assert_func at D:/ESP53/v5.3/esp-idf/components/newlib/assert.c:33

4ff21090: 0x30666634 0x39616437 0x48043000 0x4ff1251c 0x4ff21090 0x4ff12538 0x4ff2108c 0x726f6261
4ff210b0: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x37306666 0x20396164
4ff210d0: 0x63206e6f 0x2065726f 0x00000030 0x48040000 0x00000000 0x48037000 0x48043000 0x4ff07dac
0x48037000: esp_startup_start_app at D:/ESP53/v5.3/esp-idf/components/freertos/app_startup.c:86 (discriminator 1)
0x4ff07dac: esp_system_abort at D:/ESP53/v5.3/esp-idf/components/esp_system/port/esp_system_chip.c:91

4ff210f0: 0x00000000 0x48037000 0x48043000 0x48014ed2 0x480434e8 0x000004c2 0x00000000 0x48036f4e
0x48037000: esp_startup_start_app at D:/ESP53/v5.3/esp-idf/components/freertos/app_startup.c:86 (discriminator 1)
0x48014ed2: bsp_sdcard_mount at D:/examples/esp_ui_demo/managed_components/espressif__esp32_p4_function_ev_board/esp32_p4_function_ev_board.c:75
0x48036f4e: main_task at D:/ESP53/v5.3/esp-idf/components/freertos/app_startup.c:209 (discriminator 13)

4ff21110: 0x00000000 0x00001388 0x00000003 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff21130: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff21150: 0x00000000 0x4ff20180 0x4ff20180 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5
4ff21170: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000160 0x4ff20f50 0x00000000 0x4ff13614
4ff21190: 0x4ff13614 0x4ff21184 0x4ff1360c 0x00000018 0x00000000 0x00000000 0x4ff21184 0x00000000
4ff211b0: 0x00000001 0x4ff20180 0x6e69616d 0x00000000 0x00000000 0x00000000 0x00000000 0x4ff21170
4ff211d0: 0x00000009 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x00004bdb 0x00000000
4ff211f0: 0x4ff154fc 0x4ff15564 0x4ff155cc 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
4ff21210: 0x00000000 0x48005528 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x48005528: esp_cleanup_r at D:/ESP53/v5.3/esp-idf/components/newlib/newlib_init.c:43

4ff21230: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff21250: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff21270: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff21290: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff212b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff212d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000600 0xa5a5a5a5 0xa5a5a5a5
4ff212f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff21310: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff21330: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff21350: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff21370: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff21390: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff213b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff213d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff213f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff21410: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff21430: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff21450: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: d2f53a113

CPU halted.
ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x7 (HP_SYS_HP_WDT_RESET),boot:0x30f (SPI_FAST_FLASH_BOOT)
Core0 Saved PC:0x48001056
0x48001056: esp_panic_handler at D:/ESP53/v5.3/esp-idf/components/esp_system/panic.c:444

Core1 Saved PC:0x4ff03f16
0x4ff03f16: esp_cpu_wait_for_intr at D:/ESP53/v5.3/esp-idf/components/esp_hw_support/cpu.c:57 (discriminator 1)

SoftwareOli, 2024-08-21:
I have tested the SD_card_example and it works on my ESP32 P4 development board. However, I had to adjust the pins in the menuconfig. My settings are:
(44) MOSI GPIO number
(39) MISO GPIO number
(43) CLK GPIO number
(42) CS GPIO number

Maybe it will help you with your project.

liujkek22, 2024-08-21:
> I have tested the SD_card_example and it works on my ESP32 P4 development board. However, I had to adjust the pins in the menuconfig. My settings are: (44) MOSI GPIO number (39) MISO GPIO number (43) CLK GPIO number (42) CS GPIO number
> 
> Maybe it will help you with your project.

Thank you for your help. I'm using the bsp_sdcard_mount() API from the managed components. Normally, this API works without any issues, but in this particular demo, it’s causing the error I mentioned earlier. Since the API is official and has been well-tested, I’m puzzled as to why it’s not working in this case. Could you provide some insight into this?

chegewara, 2024-08-21:
Normally yes, but i believe that all examples for P4 may have incorrectly assigned pins for peripherals.
I am strongly suggesting to check always with schematics.
The same situation is with some example (dont remember which one) and ethernet.

liujkek22, 2024-08-22:
Thank you for your suggestion. I followed your advice and checked the assigned pins. Interestingly, I conducted a test in another project, such as "lvgl_demo_v8/v9." In that test, I performed the following steps:

1. Called `bsp_sdcard_mount();`
2. Tested opening and reading a file

Everything worked fine. However, in the ESP_UI_DEMO project, the same steps don't seem to work as expected. This is quite puzzling. 

If the `bsp_sdcard_mount();` API were faulty, it wouldn't work in the other project either, so the issue might be related to some setting or an API conflict within the ESP_UI_DEMO project.

st3m0r, 2024-09-25:
same problem, if you are using the wifi than mmc doesn't work, only sd_spi

Horion0415, 2024-10-11:
Currently, the SD card initialization method conflicts with esp_hosted, but this issue has been resolved and is being merged into ESP-BSP. If you need to test it urgently, you can use the PR (https://github.com/espressif/esp-bsp/pull/401).