espressif/esp-dev-kits #105: USB between esp32 slave and the MCU

maaamcube, 2024-09-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello everyone,
I want to know in the example usb_wireless_bridge, can i use a usb between the esp32 slave and the MCU like between PC and esp32 ?
Thanks.

lijunru-hub, 2024-09-20:
The slave connection to the esp32 can only be made through the uart serial port, the USB connection is not supported.

maaamcube, 2024-09-20:
@lijunru-hub so what is the solution to connect my stm32 usb device to send through wifi ?

lijunru-hub, 2024-12-02:
use UART

maaamcube, 2024-12-02:
@lijunru-hub for the moment, i want to use USB, so i think i have to look for another solution.

lijunru-hub, 2024-12-09:
It is also possible to connect via USB. You will need to use the `usb_host_cdc_acm` library. However, you will need to port the relevant code yourself by modifying the UART host program into a USB host program.

https://components.espressif.com/components/espressif/usb_host_cdc_acm/versions/2.0.6