espressif/esp-dev-kits #106: ADC Continuous mode, ringbuffer time out continuously after 1 buffer read on ESP P4.

brookMD, 2024-09-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-P4 (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-P4 Function EV Board

### Power Supply used.

USB

### What is the expected behavior?

I tried to run the adc continuous mode example from esp-idf examples making necessary GPIO changes to suit P4. It should continuously read adc samples using the API function adc_continuous_read(). 

### What is the actual behavior?

The function adc_continuous_read() returns values for exactly 1 buffer length (whatever the frame size is. I'm using frame size = 256, so it reads 256 values) and then continuously returns ESP_ERR_TIMEOUT after that. Also, the adc raw values are not accurate. I expect the data to be '0' when the adc input pin is grounded. But it reads values around '280' - '290'.

### Steps to reproduce.

This is the code I'm using:
```
/* Initializtion Function */
void adc_init(adc_continuous_handle_t *out_handle)
{
    adc_unit_t adc_unit_num;
    adc_channel_t adc_channel;
    adc_continuous_handle_t adc_handle = NULL;

    adc_continuous_handle_cfg_t handle_config = {
        .max_store_buf_size = ADC_MAX_BUFFER_SIZE, // size = 2048
        .conv_frame_size = ADC_FRAME_SIZE, // 256
    };

    ESP_ERROR_CHECK(adc_continuous_new_handle(&handle_config, &adc_handle));

    adc_continuous_io_to_channel(ADC_GPIO_NUM, &adc_unit_num, &adc_channel);

/* Need to configure pattern struct as per below for each channel being used
Currently using only 1 channel */

    adc_digi_pattern_config_t adc_pattern_config = {
        .atten = ADC_ATTEN, // 0dB
        .bit_width = ADC_RESOLUTION, // 12 bits
        .channel = adc_channel,
        .unit = adc_unit_num,
[... 378 lines cut ...]
I (2414) adc_Test: ADC_Value: 296, Channel:5, i188:
I (2420) adc_Test: ADC_Value: 296, Channel:5, i192:
I (2425) adc_Test: ADC_Value: 296, Channel:5, i196:
I (2431) adc_Test: ADC_Value: 293, Channel:5, i200:
I (2436) adc_Test: ADC_Value: 295, Channel:5, i204:
I (2442) adc_Test: ADC_Value: 295, Channel:5, i208:
I (2447) adc_Test: ADC_Value: 295, Channel:5, i212:
I (2453) adc_Test: ADC_Value: 295, Channel:5, i216:
I (2459) adc_Test: ADC_Value: 290, Channel:5, i220:
I (2464) adc_Test: ADC_Value: 290, Channel:5, i224:
I (2470) adc_Test: ADC_Value: 290, Channel:5, i228:
I (2475) adc_Test: ADC_Value: 295, Channel:5, i232:
I (2481) adc_Test: ADC_Value: 295, Channel:5, i236:
I (2486) adc_Test: ADC_Value: 287, Channel:5, i240:
I (2492) adc_Test: ADC_Value: 287, Channel:5, i244:
I (2497) adc_Test: ADC_Value: 287, Channel:5, i248:
I (2503) adc_Test: ADC_Value: 289, Channel:5, i252:

Returns ESP_ERR_TIMEOUT continuously after reading 256 values.
```

### More Information.

I tried running the same code on ESP32-S3 LCD EV Board with ESP-IDF v5.2.2
The code ran successfully without any issues.

YanKE01, 2025-01-06:
Hi, @brookMD .  The ADC driver for P4 is not yet fully developed, and we will fix these issues as soon as possible.

YanKE01, 2025-02-18:
Hi, @brookMD . You can re-test this part of the code based on the latest master. I have tested your code on the current version and didn't encounter any issues.