espressif/esp-dev-kits #110: ESP32-P4 - Factory Demo / esp-ui-demo Camera App Missing

wezhunter, 2024-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
The ESP32-P4 factory demo original firmware build (p4_factory_v14_012.bin) includes a CSI camera app but the source hosted on here doesn't include it. The newer esp-ui-demo doesn't include it either.

Would it be possible to release the source for it please, even it its out of date, since it has LVGL, CSI and rendering all integrated?

Video of the original factory firmware is here:
https://youtu.be/sQvEbEkGLbs?t=98

Thanks

Horion0415, 2024-10-08:
Yes, the camera-related examples have not been released yet because the ISP driver and algorithms have not been fully optimized. Once they are ready, we will release the relevant code as soon as possible. If you urgently need to test the camera functions, you can refer to: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/camera/camera_dsi. In the future, we will officially release examples integrating the LCD, as well as the camera code in the demo.

wezhunter, 2024-10-08:
Thanks for the prompt and detailed reply!

I understand there is optimisations due and it's pre-release hardware and software. I have seen the ECO2 comments and YUV422 notes in the idf codebase.

Purely in the interest of supporting future engineering, I'm interested to see how the camera app was implemented as a back buffer with LVGL overlays in the factory demo. I can only assume it's using a dma transfer of the ISP YUV bitstream output into the LVGL display buffer with an LVGL transparent bottom and active background? Would be really interesting to know!

If you could please provide a concrete example as to how this was done it would be greatly appreciated. In return I can help support longer term testing/validation, especially around some 1080p DSI outputs that I am working on.

Thanks

Horion0415, 2024-10-08:
There are two approaches to implementing this. The simple approach is to use the LVGL canvas widget, fill the canvas with the camera buffer, and then overlay the required LVGL widgets on top. However, this might result in frame rate loss. The more complex approach is to use the ESP32-P4's PPA (https://github.com/espressif/esp-idf/tree/master/examples/peripherals/ppa/ppa_dsi), but this requires complex buffer management. We will release related examples later, but currently, these examples are blocked by ISP driver and algorithm optimizations.

It’s also important to note that the examples in IDF are recommended only for driver testing purposes. For application-level implementations, it’s suggested to use the esp_video component (https://github.com/espressif/esp-video-components/tree/master/esp_video), although examples combining it with an LCD are still being integrated and will be released later.

wezhunter, 2024-10-08:
Thanks once again for the immediate response and your support!

Yes we've been experimenting the PPA Blend and LVGL and optimising for maximum frame rate at 1920x1080 resolutions with RGB565 data.

Regardless it's always useful to see a functional concrete example, even if at different resolutions and sub optimised pre-release, purely as a concept.

Will continue experimenting here to find which works best.

Thanks for the pointer regarding esp-video component, it's something we've just refactored for that given it's near seamless integration. It's great.

Do you happen to know what kind of timescales we might be looking at regarding a future release of an example camera app? Be insightful to understand what particular areas may require improvement that may be an impediment, as such.

Thanks

Horion0415, 2024-10-08:
It is expected that in October, examples of using esp_video in combination with an LCD will be released. At the same time, the optimization of ISP algorithms and drivers will be completed soon. Once this is done, we will first release the camera part of the development board’s factory firmware (this part of the code is actually just LVGL widgets overlaid). Following that, we will release examples of Camera + LVGL using PPA management, as well as examples related to AI (such as facial recognition and pedestrian detection). All of this is in the plan.

wezhunter, 2024-10-08:
Fantastic to hear. I'm sure others out there will be excited to hear this too!

Keep up the amazing work!

Thanks

wezhunter, 2024-10-10:
Closing this issue as it's been answered

closed 2024-10-10 by wezhunter

Horion0415, 2024-10-11:
Hi, the ISP driver and algorithm for ESP32-P4 have been preliminarily completed. You can test it based on the esp_video v0.7.0 version (https://github.com/espressif/esp-video-components/tree/master/esp_video). In the future, we will gradually release examples that combine this component with LCD, as well as the factory firmware code for the development board.

wezhunter, 2024-10-24:
Thank you! I see the latest merge containing the camera app. Truly appreciated