espressif/esp-dev-kits #111: 无法编译ESP32-P4的示例工程“esp_brookesia_phone” Can't build ESP32-P4 factory example "esp_brookesia_phone"

PURESUN-OYLY, 2024-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-P4(rev0.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-P4-Function-EV-Board

### Steps to reproduce.

我像markdown文件中列举的那样，按步骤进行了仓库的克隆、项目配置修改（启用WiFi）和编译，并重新编译了几次，均无法完成。
I have read the markdown file and followed the steps in the file, like clone, change menuconfig(Enable WiFi) and build, but I still can't build the project.

### Build Logs.

```plain
· 部分致命错误日志：Part of core error:
esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp-brookesia/src/systems/phone/stylesheets/1024_600/assets/navigation_bar_image_recents_screen.c:190:1: fatal error: opening dependency file esp-idf\espressif__esp-brookesia\CMakeFiles\__idf_espressif__esp-brookesia.dir\076454434c07f2e85d6de121769c4ee1\navigation_bar_image_recents_screen.c.obj.d: No such file or directory

· 全部日志：Full build logs:

[2/18] Building C object esp-idf/espressif__esp-brookesia/CMakeFiles/__idf_espre...a.dir/4abc0ca85b93678db78b66e2c66c6655/navigation_bar_image_recents_screen.c.ob 
FAILED: esp-idf/espressif__esp-brookesia/CMakeFiles/__idf_espressif__esp-brookesia.dir/4abc0ca85b93678db78b66e2c66c6655/navigation_bar_image_recents_screen.c.obj  
ccache C:\Develop\ESP32\tools\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.3\" -DLV_ATTRIBUTE_FAST_MEM=IRAM_ATTR -DLV_CONF_INCLUDE_SIMPLE -DLV_LVGL_H_INCLUDE_SIMPLE -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -ID:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/build/config -ID:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp-brookesia/src -IC:/Develop/ESP32/v5.3/esp-idf/components/newlib/platform_include -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/config/include -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/config/include/freertos -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/config/riscv/include -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/esp_additions/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/include/soc -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/include/soc/esp32p4 -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/dma/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/ldo/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/port/esp32p4/. -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/port/esp32p4/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/port/esp32p4/private_include -IC:/Develop/ESP32/v5.3/esp-idf/components/heap/include -IC:/Develop/ESP32/v5.3/esp-idf/components/log/include -IC:/Develop/ESP32/v5.3/esp-idf/components/soc/include -IC:/Develop/ESP32/v5.3/esp-idf/components/soc/esp32p4 -IC:/Develop/ESP32/v5.3/esp-idf/components/soc/esp32p4/include -IC:/Develop/ESP32/v5.3/esp-idf/components/hal/platform_port/include -IC:/Develop/ESP32/v5.3/esp-idf/components/hal/esp32p4/include -IC:/Develop/ESP32/v5.3/esp-idf/components/hal/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_rom/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_rom/include/esp32p4 -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_rom/esp32p4 -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_common/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_system/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_system/port/soc -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_system/port/include/riscv -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_system/port/include/private -IC:/Develop/ESP32/v5.3/esp-idf/components/riscv/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_timer/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/include/apps -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/include/apps/sntp -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/lwip/src/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/port/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/port/freertos/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/port/esp32xx/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/port/esp32xx/include/sys -ID:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/lvgl__lvgl -ID:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/lvgl__lvgl/src -ID:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components -ID:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/lvgl__lvgl/examples -ID:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/lvgl__lvgl/demos -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f  -fdiagnostics-color=always -Wno-format -Wno-unused-variable -Wno-missing-field-initializers -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -O2 -fmacro-prefix-map=D:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone=. -fmacro-prefix-map=C:/Develop/ESP32/v5.3/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration "-DESP_BROOKESIA_MEMORY_CUSTOM_MALLOC(x)=heap_caps_aligned_alloc(1, x, MALLOC_CAP_SPIRAM)" -DESP_BROOKESIA_MEMORY_CUSTOM_FREE=free -MD -MT esp-idf/espressif__esp-brookesia/CMakeFiles/__idf_espressif__esp-brookesia.dir/4abc0ca85b93678db78b66e2c66c6655/navigation_bar_image_recents_screen.c.obj -MF esp-idf\espressif__esp-brookesia\CMakeFiles\__idf_espressif__esp-brookesia.dir\4abc0ca85b93678db78b66e2c66c6655\navigation_bar_image_recents_screen.c.obj.d -o esp-idf/espressif__esp-brookesia/CMakeFiles/__idf_espressif__esp-brookesia.dir/4abc0ca85b93678db78b66e2c66c6655/navigation_bar_image_recents_screen.c.obj -c D:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp-brookesia/src/systems/phone/stylesheets/480_480/assets/navigation_bar_image_recents_screen.c      
D:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp-brookesia/src/systems/phone/stylesheets/480_480/assets/navigation_bar_image_recents_screen.c:174:1: fatal error: opening dependency file esp-idf\espressif__esp-brookesia\CMakeFiles\__idf_espressif__esp-brookesia.dir\4abc0ca85b93678db78b66e2c66c6655\navigation_bar_image_recents_screen.c.obj.d: No such file or directory
  174 | };
      | ^
compilation terminated.
[3/18] Building C object esp-idf/espressif__esp-brookesia/CMakeFiles/__idf_espre...a.dir/076454434c07f2e85d6de121769c4ee1/navigation_bar_image_recents_screen.c.ob 
FAILED: esp-idf/espressif__esp-brookesia/CMakeFiles/__idf_espressif__esp-brookesia.dir/076454434c07f2e85d6de121769c4ee1/navigation_bar_image_recents_screen.c.obj  
[... cut ...] eck_sizes.py --offset 0x8000 bootloader 0x2000 D:/Develop/Project/ESP32-P4/Factory_examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/build/bootloader/bootloader.bin"
Bootloader binary size 0x5870 bytes. 0x790 bytes (8%) free.
ninja: build stopped: subcommand failed.

 *  The terminal process "c:\Develop\ESP32\tools\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1.
```

### More Information.

我的电脑是ARM架构的CPU，不知道是否有影响。
My computer's CPU framework is ARM, is it some qustion?

PURESUN-OYLY, 2024-10-05:
我把IDF更新到了5.3.1，但是依旧无法完成编译。
I have updated my IDF to 5.3.1, but it still can't accomplish build process.

Horion0415, 2024-10-08:
您好，我这边测试使用 Linux 系统是能正常编译的，后续我会测试一下在 windows 系统上编译是否存在问题。

PURESUN-OYLY, 2024-10-11:
> 您好，我这边测试使用 Linux 系统是能正常编译的，后续我会测试一下在 windows 系统上编译是否存在问题。

您好，您这边有测试windows系统是否可以编译吗，我把报错的部分截图进行补充，其显示的错误均是类似“navigation_bar_image_recents_screen.c.obj.d”，似乎“navigation_bar_image_recents_screen.c”文件没有被正常编译，或者是路径不正确。
[image]
[image]

就文件“navigation_bar_image_recents_screen.c”而言，我在工程中进行了搜索，其结果显示已经生成编译指令，但是没有搜索到对应的“navigation_bar_image_recents_screen.c.obj.d”文件，我也附上了“build.ninja”、“compile_commands.json”以及“project_description.json”
[compile_commands.json](https://github.com/user-attachments/files/17343717/compile_commands.json)
[project_description.json](https://github.com/user-attachments/files/17343782/project_description.json)
[build.ninja.txt](https://github.com/user-attachments/files/17343788/build.ninja.txt)
备注：ninja文件无法上传，我将其后缀名添加了“.txt”

Horion0415, 2024-10-14:
我这边在 windows 上编过了，只是我这边是正常英特尔的 CPU。你测试一下其他示例是否也会出现这个情况呢？

PURESUN-OYLY, 2024-10-14:
> 我这边在 windows 上编过了，只是我这边是正常英特尔的 CPU。你测试一下其他示例是否也会出现这个情况呢？

我编译了
[image]
这两个工程，他们都可以被成功编译，但是"esp_brookesia_phone"不可以编译

Horion0415, 2024-10-23:
您这边是使用 ESP-IDF 插件进行编译的吗?

PURESUN-OYLY, 2024-10-25:
> 您这边是使用 ESP-IDF 插件进行编译的吗?

是的，的版本是1.8.1
[image]

Horion0415, 2024-10-26:
您好，最近示例进行了更新，您可以再次尝试是否出现相同问题。最近测试发现确实插件会出现问题。

PURESUN-OYLY, 2024-10-26:
好的，我试一下

PURESUN-OYLY, 2024-10-26:
> 您好，最近示例进行了更新，您可以再次尝试是否出现相同问题。最近测试发现确实插件会出现问题。

您好，轻问需要使用哪个版本的idf进行编译，我使用了5.3和5.3.1，目前也是无法编译，我把错误贴到下面。
[image]
[5/1652] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/C_/Develop/ESP32/v5.3/esp-idf/components/mbedtls/port/crypto_shared_gdma/esp_crypto_shared_gdma.c.obj
FAILED: esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/C_/Develop/ESP32/v5.3/esp-idf/components/mbedtls/port/crypto_shared_gdma/esp_crypto_shared_gdma.c.obj
C:\Develop\ESP32\tools\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.3\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -ID:/Develop/Project/ESP32-P4/Factory_Examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/build/config -IC:/Develop/ESP32/v5.3/esp-idf/components/mbedtls/port/include -IC:/Develop/ESP32/v5.3/esp-idf/components/mbedtls/mbedtls/include -IC:/Develop/ESP32/v5.3/esp-idf/components/mbedtls/mbedtls/library -ID:/Develop/Project/ESP32-P4/Factory_Examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/build/esp-idf/mbedtls/mbedtls/library -IC:/Develop/ESP32/v5.3/esp-idf/components/mbedtls/port/sha/dma/include -IC:/Develop/ESP32/v5.3/esp-idf/components/mbedtls/port/aes/dma/include -IC:/Develop/ESP32/v5.3/esp-idf/components/mbedtls/port/aes/include -IC:/Develop/ESP32/v5.3/esp-idf/components/newlib/platform_include -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/config/include -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/config/include/freertos -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/config/riscv/include -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -IC:/Develop/ESP32/v5.3/esp-idf/components/freertos/esp_additions/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/include/soc -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/include/soc/esp32p4 -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/dma/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/ldo/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/port/esp32p4/. -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/port/esp32p4/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_hw_support/port/esp32p4/private_include -IC:/Develop/ESP32/v5.3/esp-idf/components/heap/include -IC:/Develop/ESP32/v5.3/esp-idf/components/log/include -IC:/Develop/ESP32/v5.3/esp-idf/components/soc/include -IC:/Develop/ESP32/v5.3/esp-idf/components/soc/esp32p4 -IC:/Develop/ESP32/v5.3/esp-idf/components/soc/esp32p4/include -IC:/Develop/ESP32/v5.3/esp-idf/components/hal/platform_port/include -IC:/Develop/ESP32/v5.3/esp-idf/components/hal/esp32p4/include -IC:/Develop/ESP32/v5.3/esp-idf/components/hal/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_rom/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_rom/include/esp32p4 -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_rom/esp32p4 -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_common/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_system/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_system/port/soc -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_system/port/include/riscv -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_system/port/include/private -IC:/Develop/ESP32/v5.3/esp-idf/components/riscv/include -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_timer/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/include/apps -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/include/apps/sntp -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/lwip/src/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/port/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/port/freertos/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/port/esp32xx/include -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/Develop/ESP32/v5.3/esp-idf/components/lwip/port/esp32xx/include/sys -IC:/Develop/ESP32/v5.3/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -IC:/Develop/ESP32/v5.3/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -IC:/Develop/ESP32/v5.3/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -IC:/Develop/ESP32/v5.3/esp-idf/components/esp_mm/include -IC:/Develop/ESP32/v5.3/esp-idf/components/bootloader_support/include -IC:/Develop/ESP32/v5.3/esp-idf/components/bootloader_support/bootloader_flash/include -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f  -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -Wmissing-declarations -Wmissing-prototypes -fdiagnostics-color=always -Wno-format -Wno-unused-variable -Wno-missing-field-initializers -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-co [... cut ...]
C:/Develop/ESP32/v5.3/esp-idf/components/mbedtls/port/crypto_shared_gdma/esp_crypto_shared_gdma.c:250:1: fatal error: opening dependency file esp-idf\mbedtls\mbedtls\library\CMakeFiles\mbedcrypto.dir\C_\Develop\ESP32\v5.3\esp-idf\components\mbedtls\port\crypto_shared_gdma\esp_crypto_shared_gdma.c.obj.d: No such file or directory
  250 | }
      | ^
compilation terminated.
[18/1652] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedx509.dir/x509_crt.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "c:\Develop\ESP32\tools\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1.

PURESUN-OYLY, 2024-10-26:
> 您好，最近示例进行了更新，您可以再次尝试是否出现相同问题。最近测试发现确实插件会出现问题。

您好，我已经使用Ubuntu24重新尝试，我已经确保我的idf已经安装正确并使用idf.py编译了idf所附带的示例项目，但是最新版的esp32p4示例项目仍然无法编译，错误集中在"esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/src/device/esp_video_isp_device.c"中
[image]
我分别在vscode中和terminal中进行了尝试，其结果是一致的

Horion0415, 2024-10-27:
您好 此次更新需要最低基于 release/v5.4 进行编译

PURESUN-OYLY, 2024-10-27:
> 您好 此次更新需要最低基于 release/v5.4 进行编译，这在 README 中有所注明。

好的，我马上基于5.4进行编译，谢谢

PURESUN-OYLY, 2024-10-27:
> 您好 此次更新需要最低基于 release/v5.4 进行编译，这在 README 中有所注明。

您好，我已在Linux平台基于idf master分支进行编译，所有的源文件以及spiffs已经编译通过，但是最后link的时候报错如下，这个错误我第一次遇到，请指教
/home/puresun/Develop/ESP32/tools/tools/riscv32-esp-elf/esp-14.2.0_20240906/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld:sections.ld:743 cannot move location counter backwards (from 4857cb5c to 48579bb0)

PURESUN-OYLY, 2024-10-27:
[image]

> 您好，我已在Linux平台基于idf master分支进行编译，所有的源文件以及spiffs已经编译通过，但是最后link的时候报错如下，这个错误我第一次遇到，请指教 [...]

Horion0415, 2024-10-28:
我这边测试发现并不会发生链接错误，建议检察一下环境是否有问题，或者尝试清空重新编译下。

PURESUN-OYLY, 2024-10-28:
> 我这边测试发现并不会发生链接错误，建议检察一下环境是否有问题，或者尝试清空重新编译下。

好的，我这边重新配置一下环境后再尝试，请问您这边的平台是什么

Horion0415, 2024-10-28:
我这边是 linux 平台，IDF 使用 release/v5.4 或者 master 都可以。

hujianfeng, 2024-10-29:
[image]
我编译esp_brookesia_phone项目也是cannot move location counter backwards (from 4857e01c to 4857b070)这种错误,
环境为：windows 10, Visual studio code IDF 5.4.0.

Horion0415, 2024-10-30:
这有些难处理，因为我这边测试了几台机器，其中 mac、linux、windows 都是可以的，所以这个问题的原因可能是环境问题，但因为我无法复现所以很难定位。

hujianfeng, 2024-10-30:
[image]
我换到Linux平台可以编译成功。

hujianfeng, 2024-10-30:
但linux平台git clone --recursive https://github.com/espressif/esp-idf.git大部分时间下载太慢了，比win10 vs code下慢很多。有没有像vs code的国内镜像地址？

Horion0415, 2024-10-30:
有的，可以看下这个：https://gitee.com/EspressifSystems

PURESUN-OYLY, 2024-10-30:
> 但linux平台git clone --recursive [https://github.com/espressif/esp-idf.git大部分时间下载太慢了，比win10](https://github.com/espressif/esp-idf.git%E5%A4%A7%E9%83%A8%E5%88%86%E6%97%B6%E9%97%B4%E4%B8%8B%E8%BD%BD%E5%A4%AA%E6%85%A2%E4%BA%86%EF%BC%8C%E6%AF%94win10) vs code下慢很多。有没有像vs code的国内镜像地址？

您那边的Linux是x86吗，我这边的Windows和Linux都是ARM64，Linux也不可以，报错信息是一样的

PURESUN-OYLY, 2024-11-01:
> 这有些难处理，因为我这边测试了几台机器，其中 mac、linux、windows 都是可以的，所以这个问题的原因可能是环境问题，但因为我无法复现所以很难定位。

您好，我这边测试发现编译出错的地方应该是在wifi驱动部分或者是VSCode插件某个工具的问题，今天测试的时候进记录了一下，这里[链接](https://www.wolai.com/puresun/ocw1YXnAb6ZQG8WPWfh8tR)，另外发现另一个问题，每次修改配置之后需要100-300秒的时间才开始编译，这段时间很长

Horion0415, 2024-11-03:
您在 Linux 系统下也是使用的 ESP-IDF 插件吗？

PURESUN-OYLY, 2024-11-04:
> 您在 Linux 系统下也是使用的 ESP-IDF 插件吗？

我在Linux上分别试了，命令行是可以正常编译和下载的，但是VSCode插件不可以

Horion0415, 2024-11-05:
是的，vscode 插件发现会影响一些环境变量，这个问题已经上报了。建议目前使用命令行编译下载。非常抱歉。