espressif/esp-dev-kits #112: esp32-p4-function-ev-board eth example error:[ GPIO 50 is already reserved]

yunyizhi, 2024-10-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I referred to the schematic diagram and used these pins. My sdkconfig as:
```ini
CONFIG_EXAMPLE_ETH_MDC_GPIO=50
CONFIG_EXAMPLE_ETH_MDIO_GPIO=52
CONFIG_EXAMPLE_ETH_PHY_RST_GPIO=51
```

[image]

but 
```tex
E (4428) esp.emac.gpio: emac_esp_iomux_init(86): GPIO 50 is already reserved
E (4429) esp.emac.gpio: emac_esp_iomux_rmii_clk_input(196): invalid RMII CLK input GPIO number
E (4440) esp.emac: emac_esp_config_data_interface(611): invalid EMAC RMII clock input GPIO
E (4452) esp.emac: esp_eth_mac_new_esp32(692): config emac interface failed
```

Horion0415, 2024-10-21:
[image]
Hi, the correct pin configuration for using Ethernet should be like this! You can test it, and feel free to raise any other issues if needed.

yunyizhi, 2024-10-21:
> ![image](https://private-user-images.githubusercontent.com/167844607/378250513-f78fc908-a7df-4e62-a29f-eacc5eadaacc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

thanks

closed 2024-10-21 by yunyizhi