espressif/esp-dev-kits #115: ESP32-P4-Function-Ev-Board Touch Not working

ManikandanVaradharaj, 2024-11-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4-dev-4076-gce6085349f

### Espressif SoC revision.

ESP32-P4 ECO1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-P4-Function-EV-Board

### Power Supply used.

USB

### What is the expected behavior?

After I flashed this code.I expected the UI and working functionality as it before..

### What is the actual behavior?

Display Touch is completely not working ,UI got slight changed(Lvgl is not there instead of that Squareline is there)

### Steps to reproduce.

To get the esp32-p4-function-ev-board sdk i tired to run this repository code..but after build and flashed Touch is not completely working.I didn't modified anything from the code.

### Debug Logs.

```plain
Executing action: monitor
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-P4
Running idf_monitor in directory /home/smartbuild/Downloads/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone
Executing "/home/smartbuild/.espressif/python_env/idf5.5_py3.10_env/bin/python /home/smartbuild/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32p4 --revision 1 --decode-panic backtrace /home/smartbuild/Downloads/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/build/esp_brookesia_demo.elf -m '/home/smartbuild/.espressif/python_env/idf5.5_py3.10_env/bin/python' '/home/smartbuild/esp/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32p4-eco1-20240205
Build:Feb  5 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x1780
load:0x4ff2abd0,len:0xf24
load:0x4ff2cbd0,len:0x341c
entry 0x4ff2abda
I (25) boot: ESP-IDF v5.4-dev-4076-gce6085349f 2nd stage bootloader
I (26) boot: compile time Nov 12 2024 12:51:14
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v0.1
I (30) boot: efuse block revision: v0.0
[... 209 lines cut ...]
E (3138) i2c.master: s_i2c_synchronous_transaction(913): I2C transaction failed
E (3145) i2c.master: i2c_master_transmit_receive(1196): I2C transaction failed
E (3152) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(145): i2c transaction failed
E (3159) GT911: esp_lcd_touch_gt911_read_data(220): I2C read error!
Touch workingE (3170) i2c.master: I2C transaction unexpected nack detected
E (3171) i2c.master: s_i2c_synchronous_transaction(913): I2C transaction failed
E (3178) i2c.master: i2c_master_transmit_receive(1196): I2C transaction failed
E (3185) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(145): i2c transaction failed
E (3192) GT911: esp_lcd_touch_gt911_read_data(220): I2C read error!
Touch workingE (3203) i2c.master: I2C transaction unexpected nack detected
E (3205) i2c.master: s_i2c_synchronous_transaction(913): I2C transaction failed
E (3212) i2c.master: i2c_master_transmit_receive(1196): I2C transaction failed
E (3219) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(145): i2c transaction failed
E (3226) GT911: esp_lcd_touch_gt911_read_data(220): I2C read error!
Touch workingE (3237) i2c.master: I2C transaction unexpected nack detected
E (3238) i2c.master: s_i2c_synchronous_transaction(913): I2C transaction failed
E (3245) i2c.master: i2c_master_transmit_receive(1196): I2C transaction failed
E (3252) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(145): i2c transaction failed
E (3259) GT911: esp_lcd_touch_gt911_read_data(220): I2C read error!
Touch workingE (3270) i2c.master: I2C transaction unexpected nack detected
E (3272) i2c.master: s_i2c_synchronous_transaction(913): I2C transaction failed
E (3279) i2c.master: i2c_master_transmit_receive(1196): I2C transaction failed
E (3286) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(145): i2c transaction failed
E (3293) GT911: esp_lcd_touch_gt911_read_data(220): I2C read error!
```

Horion0415, 2024-11-13:
Sorry, will this issue persist? Please check if the wiring is correct (https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/user_guide.html#required-hardware).

ManikandanVaradharaj, 2024-11-13:
Hai,Thanks for responding.Yeah the wiring connections are connected
properly. I checked out.when i brought the product that time i connected
wires and that time it was working,Once i build and flashed this code UI is
slight changed and Touch is not completely working.

Horion0415, 2024-11-13:
You could try providing additional power to the development board or directly to the display (via the power port on the display’s sub-board) to rule out any power insufficiency issues. Alternatively, you could try flashing an LVGL example to see if the touch functionality still has issues.

ManikandanVaradharaj, 2024-11-13:
yeah,the LVGL example has been working with touches..but not in esp-brookesia-phone one..it shows i2c errors.I checked the display to sub-board continuity also its fine but the touch is not working.

ManikandanVaradharaj, 2024-11-13:
without connecting the camera..touch wont work?

Horion0415, 2024-11-13:
This is an issue with the esp_video component. When the camera initializes, it needs to confirm the camera sensor model based on the I2C address. I suspect it couldn't find the sensor, which occupied the bus and caused the touch functionality to freeze.

ManikandanVaradharaj, 2024-11-13:
But when i purchased the product I connect and checked without camera connection.that time the touch functionality works fine.

Horion0415, 2024-11-13:
I apologize for the inconvenience. It is possible that the version of the esp_video component in the factory firmware is different from the current one. I am sorry for the trouble this has caused. This issue will be reported internally and will be fixed in the future.

ManikandanVaradharaj, 2024-11-13:
Can I get the older one to work one.

Horion0415, 2024-11-13:
Sorry, the older version no longer works because it lacks many features. We recommend that customers use this example in conjunction with a camera.

ManikandanVaradharaj, 2024-11-15:
In esp32-p4 what are the necessary components and libraries for rtsp client in esp brookesia phone

elgerg, 2024-11-19:
Hi,

I just got my new board and found that the touch does not work while the camera is unplugged.

I was surprised and thought I had a faulty board until I read this post.

You should fix the firmware so it can load without the camera.

Thanks

Horion0415, 2024-11-20:
Apologies for this issue. It is a problem within the esp_video component, which has already been reported and is expected to be fixed soon.

ManikandanVaradharaj, 2024-11-23:
@Horion0415 when we can get the update for this issue

Horion0415, 2024-11-23:
We have identified the issue, but since it involves the esp_video component and the I2C driver, it will take some time to resolve.

ManikandanVaradharaj, 2024-12-10:
@Horion0415 while I'm trying build the brokeesia_phone project it getting lots of error,what caused this and does any changes should i do(I didn't changed the code from project).     
In file included from /home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/src/esp_video_isp_pipeline.c:26:
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/include/esp_video_isp_ioctl.h:89:40: error: 'ISP_GAMMA_CURVE_POINTS_NUM' undeclared here (not in a function)
   89 |     esp_video_isp_gamma_point_t points[ISP_GAMMA_CURVE_POINTS_NUM];
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/include/esp_video_isp_ioctl.h:130:20: error: 'ISP_SHARPEN_TEMPLATE_X_NUMS' undeclared here (not in a function); did you mean 'ISP_BF_TEMPLATE_X_NUMS'?
  130 |     uint8_t matrix[ISP_SHARPEN_TEMPLATE_X_NUMS][ISP_SHARPEN_TEMPLATE_Y_NUMS];
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                    ISP_BF_TEMPLATE_X_NUMS
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/include/esp_video_isp_ioctl.h:130:49: error: 'ISP_SHARPEN_TEMPLATE_Y_NUMS' undeclared here (not in a function); did you mean 'ISP_BF_TEMPLATE_Y_NUMS'?
  130 |     uint8_t matrix[ISP_SHARPEN_TEMPLATE_X_NUMS][ISP_SHARPEN_TEMPLATE_Y_NUMS];
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 ISP_BF_TEMPLATE_Y_NUMS
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/include/esp_video_isp_ioctl.h:159:5: error: unknown type name 'esp_isp_ae_env_detector_evt_data_t'
  159 |     esp_isp_ae_env_detector_evt_data_t ae;  /*!< ISP exposure statistics */
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/include/esp_video_isp_ioctl.h:161:5: error: unknown type name 'esp_isp_hist_evt_data_t'
  161 |     esp_isp_hist_evt_data_t hist;           /*!< ISP histogram statistics */
      |     ^~~~~~~~~~~~~~~~~~~~~~~
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/include/esp_video_isp_ioctl.h:162:5: error: unknown type name 'esp_isp_sharpen_evt_data_t'
  162 |     esp_isp_sharpen_evt_data_t sharpen;     /*!< ISP sharpen statistics */
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_ipa/include/esp_ipa.h:11,
                 from /home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/src/esp_video_isp_pipeline.c:27:
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_ipa/include/esp_ipa_types.h:17:38: error: 'ISP_AE_BLOCK_X_NUM' undeclared here (not in a function)
   17 | #define ISP_AE_REGIONS              (ISP_AE_BLOCK_X_NUM * ISP_AE_BLOCK_Y_NUM)   /*!< Auto exposure regions */
      |                                      ^~~~~~~~~~~~~~~~~~
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_ipa/include/esp_ipa_types.h:110:33: note: in expansion of macro 'ISP_AE_REGIONS'
  110 |     esp_ipa_stats_ae_t ae_stats[ISP_AE_REGIONS];
      |                                 ^~~~~~~~~~~~~~
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_ipa/include/esp_ipa_types.h:17:59: error: 'ISP_AE_BLOCK_Y_NUM' undeclared here (not in a function)
   17 | #define ISP_AE_REGIONS              (ISP_AE_BLOCK_X_NUM * ISP_AE_BLOCK_Y_NUM)   /*!< Auto exposure regions */
      |                                                           ^~~~~~~~~~~~~~~~~~
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_ipa/include/esp_ipa_types.h:110:33: note: in expansion of macro 'ISP_AE_REGIONS'
  110 |     esp_ipa_stats_ae_t ae_stats[ISP_AE_REGIONS];
      |                                 ^~~~~~~~~~~~~~
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_ipa/include/esp_ipa_types.h:114:37: error: 'ISP_HIST_SEGMENT_NUMS' undeclared here (not in a function)
  114 |     esp_ipa_stats_hist_t hist_stats[ISP_HIST_SEGMENT_NUMS];
      |                                     ^~~~~~~~~~~~~~~~~~~~~
[... cut ...]
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/src/device/esp_video_isp_device.c: In function 'isp_stop_gamma':
/home/smartbuild/socket/esp-dev-kits-master/esp-dev-kits-master/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/src/device/esp_video_isp_device.c:843:25: error: implicit declaration of function 'esp_isp_gamma_disable'; did you mean 'esp_isp_ccm_disable'? [-Werror=implicit-function-declaration]
  843 |     ESP_RETURN_ON_ERROR(esp_isp_gamma_disable(isp_video->isp_proc), TAG, "failed to disab

Horion0415, 2024-12-11:
You may try compiling with the ESP-IDF master branch.